zephyrproject-rtos/zephyr #32157: Counter (HW timer) - Resetting counter to start

stoberblog, 2021-02-10:
**Is your enhancement proposal related to a problem? Please describe.**
I would like a function that allows reset/clearing of a counter (HW timer) to 0.

**Describe the solution you'd like**
A function such as 
    int counter_clear(const struct device *dev);

for the nordic nrf hal, I believe this would link to 
    nrf_timer_task_trigger(get_nrfx_config(dev)->timer, NRF_TIMER_TASK_CLEAR);

**Describe alternatives you've considered**
1) Directly link to the HAL and call the above directly,
2) OR use relative timers, however this has the issue (well how I read the documentation) in that you can't use relative to loop around the timer (ie if at counter say at 50% and you want a relative timer of 75%, it will fail).

**Additional context**
I am starting the counter, running it x number of cycles and then stopping the counter. When I next use the timer I want it to be starting at 0, not an arbitrary number (due to the delay between interrupt trigger and the calling the timer stop function)

LBleijendaal, 2022-07-29:
Any progress on this feature? I need this too. Is there a way to reset the counter in a temporary way before this is implemented?

ahmedwahdan, 2023-07-02:
Any progress on this feature?

MarshMilo100, 2024-10-07:
A workaround for those still waiting on this feature is to unset `COUNTER_TOP_CFG_DONT_RESET` in the configuration flags and call `counter_set_top_value()` but set the top value to the same as it was before. It appears that there is no check for setting the top value to the same number, but the act of setting the value will reset the counter if that flag is configured properly.

nandojve, 2025-03-01:
Hi,
Recently I face a similar dilemma.
Question: Can the start API provide an option to reset the counter? This could safe one API call maybe.
So, if you start an already started timer it could simple reset and continue.
Just an idea and I'm not sure if it fits well.

rruuaanng, 2025-03-02:
> Hi, Recently I face a similar dilemma. Question: Can the start API provide an option to reset the counter? This could safe one API call maybe. So, if you start an already started timer it could simple reset and continue. Just an idea and I'm not sure if it fits well.

Hi Gerson,
I understand your point. You seem to refer to it as a reentrant reset, but wouldn’t that make the API unclear? This seems to be the same dilemma as being forced to use `counter_set_top_value`. I believe calling it the new `counter_reset` would be more appropriate.

PR #86442 (2025-03-26): drivers: counter: Add counter_reset function
Add a counter_reset function to the counter subsystem
to reset the counter value to zero. Introduce
a reset interface at the driver layer to implement
the counter_reset function.

Closes: #32157
closed 2025-03-26 by kartben