zephyrproject-rtos/zephyr #32201: arch_switch() on ARM64 isn't quite right

andyross, 2021-02-11:
Along the lines of #32197, the ARM64 implementation of arch_switch() has a bug which was just unmasked by a race treatment submitted in #32089.

The code gets the basic scheme correct, and understands that the second argument is the address of the switch_handle field of the struct thread_base record of the output thread.  And it doesn't load that pointer and merely uses it via an offset to get a pointer to the thread struct.

But it never stores the new "switch handle" through that pointer into the output thread record and simply assumes that it will always be set to the address of hte thread struct.  New code, to address an SMP race, wants to NULL out this field as a synchronization signal.  But now that corrupts the thread struct for the next context switch into it.

This is worked around in current code, but that will only work for uniprocessor configurations.  See the docs in kernel_arch_interface.

carlocaione, 2021-02-11:
This is in the SMP PR currently open https://github.com/zephyrproject-rtos/zephyr/pull/30676/commits/02b267b69095662b79f03dde9068d18a3f0ae97d is that enough to address this problem?

andyross, 2021-02-11:
Ah, excellent.  Indeed that looks correct to me.

galak, 2021-05-27:
@carlocaione is this still an issue?

carlocaione, 2021-05-28:
@galak nope

closed 2021-05-28 by carlocaione