zephyrproject-rtos/zephyr #32203: Cannot set static address when using hci_usb or hci_uart on nRF5340 attached to Linux Host

furkanhtk, 2021-02-11:
Hi,
As you suggested in #31308, I tried it on Raspberry with new kernel and Bluez versions and encountered the same error.
`Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 arm7l GsNU/Linux`
with bluez version : `5.50-1.2~deb10u1+rpt2`
Thanks for help.

#3230
#11016

@carlescufi
@jfischer-no 
@dcpleung
Label:

jfischer-no, 2021-02-11:
I have tried it on RPi3 with _2021-01-11-raspios-buster-armhf-lite_ and Kernel version `Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux` (after apt upgrade). I successfully tested nRF5340 on nRF5340DK (0.11.0).
Unfortunately, only now I realized that nRF5340 can not work OOTB with hci_usb. The **cpunet** must also be flashed with appropriate application, hci_rpmsg.

```
west build --build-dir build_cpunet -b nrf5340dk_nrf5340_cpunet samples/bluetooth/hci_rpmsg
west build --build-dir build_cpuapp -b nrf5340dk_nrf5340_cpuapp samples/bluetooth/hci_usb
west flash --build-dir build_cpunet/
west flash --build-dir build_cpuapp/
```

carlescufi, 2021-02-11:
> I have tried it on RPi3 with _2021-01-11-raspios-buster-armhf-lite_ and Kernel version `Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux` (after apt upgrade). I [...]

Define "OOTB". For me, having to flash 2 binaries, one per core, is a reasonable requirement for a dual-core device that gets its images from building Zephyr twice.

jfischer-no, 2021-02-11:
> Define "OOTB". For me, having to flash 2 binaries, one per core, is a reasonable requirement for a dual-core device that gets its images from building Zephyr twice.

Well I think this requirement is not immediately clear to everyone, how is an inexperienced user supposed to know which application is required for network-cpu? I think it makes sense to update README.rst in hci_usb sample.

furkanhtk, 2021-02-12:
Thanks for help. Now I am able to scan through btmgmt.

closed 2021-02-12 by furkanhtk