zephyrproject-rtos/zephyr #32205: [misc] AArch64 improvements and fixes

carlocaione, 2021-02-11:
Moving here several points raised by @jharris-intel in #30676 and not really related to that PR.

carlocaione, 2021-02-11:
https://github.com/zephyrproject-rtos/zephyr/pull/30676#discussion_r573933433

carlocaione, 2021-02-11:
https://github.com/zephyrproject-rtos/zephyr/pull/30676#discussion_r573942855

carlocaione, 2021-02-11:
https://github.com/zephyrproject-rtos/zephyr/pull/30676#discussion_r573946927

carlocaione, 2021-02-11:
tagging the relevant people @MrVan @jharris-intel @npitre

carlocaione, 2021-02-11:
@jharris-intel about your comment:
```
We start in EL3, but EL2 is implemented.

In this mode we need to do some EL2 setup also. 
```
After you setup for EL2 do you drop into EL2 or directly into EL1?

carlescufi, 2021-02-11:
@carlocaione could you please move all links/references to the body/description of the issue instead of adding comments?

carlocaione, 2021-02-11:
> @carlocaione could you please move all links/references to the body/description of the issue instead of adding comments?

done

carlescufi, 2021-02-11:
> > @carlocaione could you please move all links/references to the body/description of the issue instead of adding comments?
> 
> done

Thank you! Feel free to delete the comments to clean up.

jharris-intel, 2021-02-11:
> After you setup for EL2 do you drop into EL2 or directly into EL1?

The current hacked-together local changes skip the existing EL2 setup entirely, and as such drop directly into EL1's setup code in __start from EL3's setup code. 

This was only because I was trying to avoid changing __start itself if at all possible, which meant I was unable to avoid this. As a result I had to replicate portions of the EL2 setup code in `z_arch_el3_plat_init`.

Ideally, EL3 setup would drop into the next implemented EL. That is, something along the lines of the following:

1. Assert that the current EL is implemented.
2. Check if the current EL is the highest implemented EL. If so, call a routine that does anything that must be done at the highest implemented EL.
2. Do any setup in particular for this EL.
3. Return to the next lower implemented EL's setup code.

carlocaione, 2021-02-11:
What about simply having a weak `z_arch_plat_init(current_el, is_highest)` to be called by the reset code at each descending Elx. The platform can then use that as it please.

carlocaione, 2021-03-01:
Closing this because this are being taken care of in #32394

closed 2021-03-01 by carlocaione