zephyrproject-rtos/zephyr #32206: CMSIS-DSP support seems broken on link

mkvenkit, 2021-02-11:
**Describe the bug**

I am trying to use CMSIS-DSP in a simple project. I added the following to prj.conf:

```
CONFIG_CMSIS_DSP=y
CONFIG_NEWLIB_LIBC=y
CONFIG_FPU=y
# # Set CONFIG_NEWLIB_LIBC_FLOAT_PRINTF if  printf should be able to print float
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
```
In my code I am using:

```
#include "arm_math.h"
#include "arm_const_structs.h"
...

arm_rfft_fast_f32(&S, testInput, testOutput, 0);
```
This results in a linker error:

```
undefined reference to `arm_rfft_fast_f32'
```
As a workaround, I downloaded CMSIS_5 repo and modified CMakeLists.txt to include the necessary files. 

**Environment (please complete the following information):**
- Linux
- Zephyr 2.5.0-rc3

galak, 2021-02-11:
Did you also enable `CONFIG_CMSIS_DSP_TRANSFORM=y` ?

mkvenkit, 2021-02-11:
That fixes the issue, thanks. But is this documented anywhere?

closed 2021-02-11 by mkvenkit

galak, 2021-02-11:
> That fixes the issue, thanks. But is this documented anywhere?

Kinda in the Kconfig help for CMSIS_DSP.