zephyrproject-rtos/zephyr #32212: Tx power levels are not similar on ADV and CONN modes when set manually (nRF52)

selman-nus, 2021-02-11:
Hi, 

I have been trying to reduce the Tx power level to reduce the total power consumption of the device. I have successfully modified the hci_pwr_ctrl example and tested with  4,  0, -4, -8,  16, -20, -40 dBm levels while on both advertisement and connected. 

```
// txp is the array of different dBm levels
set_tx_power(BT_HCI_VS_LL_HANDLE_TYPE_ADV, 0, txp[idx]);  // for advertisement
set_tx_power(BT_HCI_VS_LL_HANDLE_TYPE_CONN, 0, txp[idx]);  //for connection
```

Power consumption is heavily influenced by the Tx level on advertisement mode. However, the more important mode, connection mode didn't seem affected by this Tx level alteration on lower levels. In my case device will stay connected and constantly transmitting data so power consumption on connection mode is much more important to me. 

Please see below two images. (I tried to show which time window is which Tx level.)

Advertisement:
[image]

Coonection:
[image]

.
.

- +4dBm:   ADV: 9mA  CONN: 11mA
-  -40dBm: ADV: 4mA  CONN: 6.7mA

Current consumption on the **connection at -40dBm** is almost equaled to one on the **advertisement at -4dBm**.

Am I missing something here? Is connection mode transmissions supposed to be using higher current?

(Forgive me if I misinterpreted what I am asking on the title.)

cvinayak, 2021-02-11:
Depending on advertising and connection interval; Radio, CPU and other peripheral current consumption can overlap increasing the total current consumption.

selman-nus, 2021-02-14:
I understand but it is actually a very simple code, it doesn't do any calculation just sending dummy HR data and no other peripheral is in use. Is there a way to reduce the CPU frequency, so it won't contribute much to the total current consumption?

closed 2021-05-01 by github-actions[bot]