zephyrproject-rtos/zephyr #32213: Universal error code type

carlescufi, 2021-02-11:
The goal here would be to be able to pass around instances of an error type that is able to describe errors beyond the error number itself. In particular, describe which specification/domain the error code maps to.

In order to encode different error codes defined in multiple, scattered specifications in a single class of data, we could provide a new type that encapsulates both the domain and the error code itself.

Something along the lines of:

```
typedef struct {
    uint16_t domain;
    uint16_t error;
} k_error_t;

k_error_t foo(void)
{
   k_error_t error;

    error.domain = K_ERR_POSIX;
    error.code = ENODEV;

    return error;
}
```
or
```
typedef uint32_t k_error_t;

k_error_t foo(void)
{
   k_error_t error;

    K_ERR_SET_DOMAIN(error, K_ERR_POSIX);
    K_ERR_SET_CODE(ENODEV);

    return error;
}
```

de-nordic, 2021-02-12:
I think that the domains may be also useful in logging, where it could allow efficient  real-time log filtering when logs are passed to another device via uart/ethernet; I have been working in such project some time ago.

You may also add `K_ERR_DEFAULT_DOMAIN` or  something like that that would set the domain for entire file so that `K_ERR_SET_CODE` could be variable parameter macro that would pick the default domain if set or the given as a parameter.
This could also allow to automatically generate errno type codes for an unit, when header containing for e.g.
```
#define BAD_AT_PREFIXES_EIO (struct k_error_t){ K_ERR_DEFAULT_DOMAIN, EIO }
#define BAD_AP_PREFIXES_ENOTSUP (struct k_error_t){ K_ERR_DEFAULT_DOMAIN, ENOTUSP }
...
```
would be added after definition of `K_ERR_DEFAULT_DOMAIN`. Such Definitions would allow code like this:
```
k_error_t ret = BAD_AT_PREFIXES_EIO ;
return BAD_AT_PREFIXES_ENOTSUP;
```

Actually for errno there could be macros that would just take the domain parameter:
```
#define K_ERR_EIO(domain) (struct k_error_t){domain, EIO }
```

I am using similar approach for returning (error, data) tuples, although I have seen resistance fro applying that in my commercial work as it is often viewed as inefficient, even though it is widely used in e.g. python code.

jhedberg, 2021-05-20:
@de-nordic btw, reading again what you wrote, I think we have a slightly different idea of what "domain" means.

I see it as answering the question "how to interpret the error code?" whereas it seems you were more thinking of it as "who created the error?". I think the former would be more useful, and also in line with this kind of error abstractions I've seen elsewhere. The "default domain" concept makes sense in both cases since a c-file for the most part generates errors for a single domain:
```
#define K_ERR_FOR_DOMAIN(_domain, _code) ...
#define K_ERR(_code) K_ERR_FOR_DOMAIN(K_ERR_DEFAULT_DOMAIN, _code)
```
Then a submodule predominantly returning posix errors would:
```
#define K_ERR_DEFAULT_DOMAIN K_ERR_DOMAIN_POSIX

k_err_t myfunc(void)
{
        return K_ERR(ENOTSUP);
}
```

de-nordic, 2021-05-20:
@jhedberg Everything is up for discussion.
There are several examples here in there that could be evaluated to figure out how they have performed and what we can learn from them; mbedtls for example uses codes which dedicate bits to identify an area where the error comes from, and so on.
On the other side I have been working on software that used such error code system: 15bits of 32 bit code would be used for file ID, where you would generate and register ID of a file in a special systemso that you could easily get where the code came from, the meaning of the other 16bit was a little bit complicated.

> I see it as answering the question "how to interpret the error code?" whereas it seems you were more thinking of it as "who created the error?"

Yes, you have read me correctly. Maybe we can get both?
When I have seen the idea first time, I have considered this as a possible solution to that:
```
ret =  something(...);
if (ret < 0) {
   returrn ret
}
...
if (not_doing()) {
   return -ERR_THAT_MAY_BE_EQUAL_ret
}
```
which happens a lot and  it forces you to relay on logs to figure out where it actually failed.

And regarding:

> 
> ..or as a more lightweight alternative to the log string stuff above: forget about string translations of codes, but at least maintain a readable string for the domain and provide the code as just a number (and let the user translate it manually)

To aid developers could follow Microsoft (https://docs.microsoft.com/en-us/windows/win32/debug/system-error-code-lookup-tool) or errinfo from MacOS and provide script to help user figure out.

jhedberg, 2021-05-20:
@de-nordic sure, for the most part I agree with you. Btw, the only case I can think of where "who created the error?" and "what does the code mean?" are different is when using the ubiquitous POSIX error numbers. Perhaps we could gradually have subcomponents declare their own domains & codes and then gradually move away from this problem.

jhedberg, 2021-05-20:
...or then encode both domain, code and location-of-creation information into the value, but maybe that's too much overhead

de-nordic, 2021-05-24:
> 
> 
> ...or then encode both domain, code and location-of-creation information into the value, but maybe that's too much overhead

If the uin32_t would be the container then we could probably figure out how to fit in the info, of course there will be problems with libraries like mbdtls which also use 32bit error codes.

> 
> (...) different is when using the ubiquitous POSIX error numbers (...)

The question is whether internally Zephyr should use POSIX error codes or rather methods of translation to POSIX layer.

The POSIX has been put together by IEEE somewhere around 1986 (published in 1990), in parallel to SVID (System V Interface Definition) from ATT. as a response to the UNIX world starting to divide. That was an attempt to get some common API among all the variants of UNIXs.
I think it has been more aligned with time-share operating systems than embedded, real-time oses.

In case of "big machines" the error codes would be very often assisted by text logs, and would be short to fit into return value of program, which could be then interpreted by shell. Something that is hard to do when you have limited resources and want to have quite responsive little system that can, for example, reliably respond to Bluetooth messages.
Printing test strings, whether to console or log file also takes time.

Basically POSIX is compatibility layer, but it limits what you can do with the system to universal stuff, and so has to translate system responses in simplified way. If you wanted to have real-time scheduling of Solaris you would have to abandon some of compatibility; tasks/threads/ports of Mach 3 could not be used via POSIX; when you wanted to use advanced Windows API features you would relay on Win32API directly (yes, 90s), and loose compatibility with POSIX.

We want POSIX, because it gives as abundance of third party libraries that use POSIX API and return standard POSIX error code; on the other side are libraries like mbdtls. and file system drivers like LittleFS or ELM Fat that have been specifically designed for embedded, and which use own error codes requiring translation.

Thalley, 2022-04-04:
Returning a tuple of values is a powerful tool as suggested by @carlescufi here, and also touched upon by @de-nordic. 

Rust employs a similar error handling design (https://doc.rust-lang.org/std/result/enum.Result.html) where the return value is a tuple of <value, err>. Personally I think such a design is more powerful than the 

```c
typedef struct {
    uint16_t domain;
    uint16_t error;
} k_error_t;
```
suggested here, as that doesn't really handle return values unlike something like `ssize_t` that has been designed to have both an error value (-1) and a (>= 0) return value (that unfortunately is limited by the negative return range (-1). 

I don't know if it is possible to implement something like the Rust `Result` type which is generic, but we could have something like

```c
typedef struct {
    int32_t value
    int error;
} k_error_int32_t;

typedef struct {
    uint32_t value
    int error;
} k_error_uint32_t;

typedef struct {
    bool value
    int error;
} k_error_bool_t;
```

nashif, 2022-11-18:
not going to happen per @carlescufi

closed 2022-11-18 by nashif