zephyrproject-rtos/zephyr #32220: gpio: api should provide a way to release resources

erwango, 2021-02-11:
**Is your enhancement proposal related to a problem? Please describe.**
In power management context, we need a way to disable GPIO port clock when needed.
Currently, GPIO API prevent this from happening because there is no way to know when a pin is in use or not.

**Describe the solution you'd like**
Provide a GPIO API function to notify driver that pin is not in use anymore and the matching resource can be released.

pabigot, 2021-02-12:
I think we can just use `gpio_config(port, pin, GPIO_DISCONNECTED)` for this purpose rather than introduce a new `GPIO_DISABLED` flag.  @mnkp your thoughts?

Something needs to track whether the device is actively used, i.e. whether a pin is configured for something other than `GPIO_DISCONNECTED` or a pin interrupt configuration is something other than `GPIO_INT_DISABLED`.  I'd normally push that down to the driver, but the way so much is done in the GPIO API wrapper functions maybe it belongs there.

But it could get complicated to do it that high up, as it would require assumptions about the state of the pin when the driver returns an error from a configure operation.  So on the whole I'd rather see that done at the driver level in concert with updates to support power management.

erwango, 2021-02-12:
> So on the whole I'd rather see that done at the driver level in concert with updates to support power management.

Fine for me, each driver may have different concept of what a resource is (pin, port, ..)

mnkp, 2021-02-15:
> I think we can just use `gpio_config(port, pin, GPIO_DISCONNECTED)` for this purpose rather than introduce a new `GPIO_DISABLED` flag. @mnkp your thoughts?

I agree with the proposal. Which means we already provide a way to release resources.

> Something needs to track whether the device is actively used, i.e. whether a pin is configured for something other than `GPIO_DISCONNECTED` or a pin interrupt configuration is something other than [...]

I believe the power management code needs to be implemented by the driver. There are SoCs where the module clock is only required to detect edge interrupts and the basic functionality, reading/writing to a pin, level interrupts work without the clock.

pabigot, 2021-02-22:
Based on above the way to indicate that a GPIO is no longer used is to configure it as `GPIO_DISCONNECTED`.

erwango, 2021-03-03:
@nmkp, @pabigot, I'm reopening this point for a specific question on GPIO interrupt configuration.

It was mentioned above that GPIO resource should be considered as busy for other state as `GPIO_INT_DISABLED`.
Though, in case we want to configure a wakeup button (as done [here](https://github.com/zephyrproject-rtos/zephyr/pull/26002/files#diff-db26abd64c8d675d7b8e66ba7103e5803e43755e2d44ec87db549dbd28eb0327)), obviously we need to configure an interrupt on a GPIO pin which would then set GPIO resource to busy and would prevent entering expected sleep/standby mode.

Is there an alternate way to achieve this (not using gpio API to configure a wake up button), or should we add a dedicated flag to support this case ?

mnkp, 2021-03-04:
Sorry, I didn't look into the STM32 datasheet, however in the sample application you use `GPIO_INT_EDGE_TO_ACTIVE` interrupt. As far as I know edge interrupts always require GPIO module clock to be on. Otherwise the hardware module would not be able to detect state change. It needs to sample the pin. And that means the device is busy. But maybe STM32 is different?

Typically to configure a wake up source we use a level sensitive interrupt. In case of some SoCs there are dedicated registers that needs to be configured. For a level interrupt configured on an input pin the GPIO module clock can be off, the device is "armed" but can be marked as not busy.

That said the question of "resources" the GPIO module is controlling is indeed a bit more complex. 

I would say it should be GPIO module that decides if it is busy or not. If it requires module's clock to be on to support current configuration then it's busy, otherwise it can be marked as not busy. `GPIO_INT_DISABLED` can be used to disable a pin / free a resource.

In general, if GPIO module has some pins configured as inputs and some as outputs it sill should be possible to enter low power mode (at the system level). GPIO module should not be blocking. If there is an edge interrupt configured on an input pin and GPIO module's clock needs to be on then the device is busy and should prevent from automatically entering low power mode.

erwango, 2021-03-05:
@mnkp thanks for the feedback. I'm trying to investigate more in the direction of what/when resources are needed to sustain various GPIO operations, but given some tries I was doing yesterday, in STOP2 core is in WFI and can be waken up by an interrupt configured on a pin using `GPIO_INT_EDGE_TO_ACTIVE` while GPIO port clock is off.

pabigot, 2021-03-05:
Nordic has a similar capability but in that case the wakeup configuration cannot be expressed with the GPIO API; see https://github.com/zephyrproject-rtos/zephyr/blob/19e5a9c0616df6e89280e42b20a0ebf03dec9200/samples/boards/nrf/system_off/src/main.c#L59

erwango, 2021-03-19:
I've been a bit deeper in this topic of GPIO resources on STM32.

First point is that GPIO interrupts don't need GPIO clock, as this part is handled in another peripheral (EXTI).
So, to receive an interrupt line on a GPIO pin, only EXTI clock is required.
But I guess this part is highly SoC specific, so it is just a side note.

Regarding pure GPIO operations, it appears (on STM32 SoC at least) that the only reason to keep the GPIO clock on after a `gpio_pin_configure` is when GPIO is configured for future GPIO bank write access (GPIO_OUTPUT).
For any other cases, clock is required for configuration, but could be cut afterwards.

Last point showed up while trying to optimize a basic blinky use case:
- First step: `gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE);` -> Clock is kept on for next operation
- Second step: `gpio_pin_set(led.port, led.pin, (int)led_is_on);` -> Led is on after the write operation
- Third step: Before going to sleep, GPIO clock could be released.

At this last step, we can call `gpio_pin_configure(led.port, led.pin, GPIO_DISCONNECTED);` to inform driver that resource could be released. Though, this should be done while not clearing the existing configuration, otherwise we clear the pin OUTPUT configuration and LED is shut down.
Conclusion: I think that `GPIO_DISCONNECTED` ("Disables pin for both input and output") is not sufficient for resource handling and we need an alternative flag GPIO_RELEASED, that would allow to release resource while keeping configuration untouched.

mnkp, 2021-03-23:
> I think that `GPIO_DISCONNECTED` ("Disables pin for both input and output") is not sufficient for resource handling and we need an alternative flag GPIO_RELEASED

Maybe this is not necessary. In this issue we are talking about GPIO module blocking or not blocking automatic entry into one of the supported low power states. In all of those low power states the CPU clock stops -> the program is no longer executed -> application cannot change the state of the output pin (or read the state of the input pin for the matter). So for the purpose of the power management all pins can be considered released. GPIO module should be free to stop its clock when SoC is entering low power state.

erwango, 2021-03-24:
> In this issue this issue we are talking about GPIO module blocking or not blocking automatic entry into one of the supported low power states. In all of those low power states the CPU clock stops -> the program is no longer executed -> application cannot change the state of the output pin (or read the state of the input pin for the matter). So for the purpose of the power management all pins can be considered released

Well, there are actually two different things to consider here:
****Core clock:****
On that front I'm totally in line with you. 
GPIO calls are all synchronous, no background processing. Program dynamic is application (or caller) responsibility, so GPIO device should just do what is asked and not block system to enter lower power state (no `device_busy_set`, `pm_constraint_set`, ...)
****GPIO device clock:****
On this front the problematic is a bit different. 
Caller (let's assume it is application) first need to configure the device in a certain state. Some states will require to keep clock on (`GPIO_OUTPUT` in STM32) for the next operation (`pin_set` to switch on a LED). Once next operation is done (LED is on), there is no more need for device clock but configuration should be kept. Issue is we miss a way to inform device GPIO:
- using `GPIO_DISCONNECT` would reset pin configuration and switch off the LED.
- doing nothing will keep the GPIO clock on uselessly.
That's why I consider we need a new config `GPIO_RELEASED` to allow to release device clock while keeping existing config.