zephyrproject-rtos/zephyr #32221: Sporadic kernel panics on stm32g4 flash erase/writes

alexanderwachter, 2021-02-11:
**Describe the bug**
When writing big data chunks to the flash of an STM32G4 SoC, sporadic kernel panics occur.
It seems that a cache flush is needed after each erase. 

**To Reproduce**
Build mcuboot for the nucleo_g474re and update several times.
Sometimes you will see a panic.

**Expected behavior**
Flash write and erase should work without panic

**Impact**
Firmware updates on stm32g4 series do not work

**Logs and console output**
````
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0xfab1f1a4
E: r0/a1:  0x0000045d  r1/a2:  0x00000000  r2/a3:  0xfab1f1a4
E: r3/a4:  0x20000208 r12/ip:  0xa0000000 r14/lr:  0x080043e1
E:  xpsr:  0x6100000f
E: Faulting instruction address (r15/pc): 0x08003d6a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000208 (unknown)
E: Halting system
````

alexanderwachter, 2021-02-11:
Fix: #32218

alexanderwachter, 2021-02-11:
why low @erwango? this affects all stm32g4 users that use the flash driver.

erwango, 2021-02-11:
@alexanderwachter This is from zephyr point of view: one series, one driver, not a regression (from what I saw) and not systematic. But doesn't prevent to review quickly and push so that it's getting merged.

PR #32218 (2021-02-11): drivers: flash: stm32g4: Flush caches after erase and write
Flush the caches after erasing.
This commit fixes sporadic kernel panics when writing big data chunks
to the flash. (data bus errors)

Fixes #32221
closed 2021-02-11 by nashif