zephyrproject-rtos/zephyr #32233: Often disconnect timeouts when running the BLE peripheral HR sample on Nitrogen96

agherzan, 2021-02-11:
**Describe the bug**

Connection often fails when running [samples/bluetooth/peripheral_hr](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/peripheral_hr) on Nitrogen96 (nRF52) and trying to connect to it from another device.

**To Reproduce**

Steps to reproduce the behaviour:
1. build the `peripheral_hr`for Nitrogen96 (I used the Yocto build system for this but it was reproducible using the west build system too)
2. run the app and connect to board from a Linux host using `bluez5` (`bluetoothctl`)

**Expected behaviour**

The application should always connect logging:

```
Connected
[00:05:28.493,316] <inf> hrs: HRS notifications enab
```

**Impact**

Product instability. 

**Logs and console output**

Failing while connecting with bluez5 (5.55) from a Linux machine:

```
Connected
Disconnected (reason 0x08)
 
1	0.000000	host	controller	HCI_CMD	12	Sent LE Set Extended Scan Parameters
2	0.109440	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
3	0.109504	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
4	0.111440	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
5	0.115463	controller	host	HCI_EVT	40	Rcvd LE Meta (LE Extended Advertising Report)
6	0.115563	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
7	0.125439	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
8	0.125533	host	controller	HCI_CMD	30	Sent LE Extended Create Connection
9	0.127413	controller	host	HCI_EVT	7	Rcvd Command Status (LE Extended Create Connection)
10	0.437462	controller	host	HCI_EVT	34	Rcvd LE Meta (LE Enhanced Connection Complete)
11	0.437667	host	controller	HCI_CMD	6	Sent LE Read Remote Features
12	0.438406	controller	host	HCI_EVT	7	Rcvd LE Meta (LE Channel Selection Algorithm)
13	0.439409	controller	host	HCI_EVT	7	Rcvd Command Status (LE Read Remote Features)
14	0.677410	controller	host	HCI_EVT	15	Rcvd LE Meta (LE Read Remote Features Complete)
15	0.678399	controller	host	HCI_EVT	7	Rcvd Disconnect Complete
 ```
 
Failing when connecting from an Android (10) device:

```
Connected
Disconnected (reason 0x13)
 ```
 
On IOS (14) it always connects and the expected behaviour is consistent:
```
Connected
[00:05:28.493,316] <inf> hrs: HRS notifications enabled
 
1	0.000000	host	controller	HCI_CMD	12	Sent LE Set Extended Scan Parameters
2	0.110266	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
3	0.110331	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
4	0.112257	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
5	0.116285	controller	host	HCI_EVT	40	Rcvd LE Meta (LE Extended Advertising Report)
6	0.116384	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
7	0.117265	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
8	0.117365	host	controller	HCI_CMD	30	Sent LE Extended Create Connection
9	0.119259	controller	host	HCI_EVT	7	Rcvd Command Status (LE Extended Create Connection)
10	0.225203	controller	host	HCI_EVT	34	Rcvd LE Meta (LE Enhanced Connection Complete)
[... 56 lines cut ...]
11	0.244435	host	controller	HCI_CMD	6	Sent LE Read Remote Features
12	0.245136	controller	host	HCI_EVT	7	Rcvd LE Meta (LE Channel Selection Algorithm)
13	0.246172	controller	host	HCI_EVT	7	Rcvd Command Status (LE Read Remote Features)
14	20.305255	controller	host	HCI_EVT	15	Rcvd LE Meta (LE Read Remote Features Complete)
15	20.306272	controller	host	HCI_EVT	7	Rcvd Disconnect Complete
```

**Environment (please complete the following information):**

 - OS: Archilnux, Android 10, IOS 14
 - Toolchain: Yocto/OE
 - Reporduce on both 2.4.0 and 2.5.0-rc3.

**Additional context**

Capturing the transfer with Wireshark, I could see that it always fails on _Sub Event: LE Read Remote Features Complete (0x04)_ with one of the following reasons:
- Connection Failed to be Established (0x3e)
- Connection Timeout (0x08)

Even if the supervisor time is bumped to 20s (on the Linux side), there are situations where it fails with a timeout after 20s.
```
15	20.306272	controller	host	HCI_EVT	7	Rcvd Disconnect Complete
```

Every time it succeeds connecting, it does in 300-400ms. See an example of a successful connection above (iOS).

cvinayak, 2021-02-11:
Please use an nRF sniffer to capture the air traffic, and attach the logs for analysis. 

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_sniffer_ble%2FUG%2Fsniffer_ble%2Fintro.html&cp=9_4

agherzan, 2021-02-11:
I've ordered one now but I think @idlethread might be able to provide some logs in the meanwhile. Thanks, @cvinayak.

cvinayak, 2021-02-12:
@agherzan if you have a spare nRF51DK, nRF51 Dongle, nRF52DK, nRF52840DK, or nRF52840 Dongle, you can use them as nRF sniffer.

agherzan, 2021-02-12:
Sadly I don't at this point but I'll get one shipped soon. I've ordered it yesterday. That being said, I think we have one in the team so we might be able to provide logs soon. We will update as soon as we have something.

idlethread, 2021-02-18:
Setup is as follows:

```
                            nRF52832-DK
                              (sniffer)
                                   |
Nitrogen (v1.1) --------------------- Linux Laptop
(peripheral_hr)                    (bluetoothctl, 
                                              scan on,
                                              connect  <mac>)
```

Nitrogen zephyr log shows:

```
*** Booting Zephyr OS build v2.5.0-rc4  ***
Bluetooth initialized
Advertising successfully started
[00:00:00.394,897] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.394,927] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.394,927] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.5 Build 0
[00:00:00.395,538] <inf> bt_hci_core: Identity: C5:AE:65:1D:0B:45 (random)
[00:00:00.395,568] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.395,568] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
Connected
Disconnected (reason 0x08)
```

nRF sniffer logs are here (just remove .zip extension): 

[v1.1-board-zephyr-2.5-rc4-samsung-laptop.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/6003281/v1.1-board-zephyr-2.5-rc4-samsung-laptop.pcapng.zip)

cvinayak, 2021-03-23:
@idlethread The attached zip file is invalid, please re-upload.

cvinayak, 2021-03-23:
The supervision timeout happens at the central as it thinks it is not receiving responses from the peripheral. Clearly, the sniffer can see the responses. 
[image]

cvinayak, 2021-05-12:
@agherzan and @idlethread any updates ? is this still a zephyr issue?

carlescufi, 2021-05-27:
Closing this due to reporter inactivity. Feel free to reopen if this is still an issue.

closed 2021-05-27 by carlescufi

idlethread, 2021-05-31:
We tracked this down to intel bluetooth controllers (and firmware) on the host PC as being the common reason for failures. We haven't yet spent time trying to run the latest bluetooth stack on those hosts to see if it made the problem go away. That is still pending on us.

We saw no issues with a different controller.