zephyrproject-rtos/zephyr #32234: Documentation: How to update Zephyr itself (with west)

dueringa, 2021-02-11:
**Current state**
Currently, the "Getting started Guide" only mentions the initial fetch of Zephyr with west init / west update. The "Beyond the Getting Started Guide" briefly mentions that 

>The west update command fetches and keeps Modules (External projects) in the zephyrproject folder in sync with the code in the local zephyr repo.
>Warning
>You must run west update any time the zephyr/west.yml changes, caused, for example, when you pull the zephyr repository, switch branches in it, or perform a git bisect inside of it.

**Suggestion**
I'd actually like the documentation making it more explicit "How to update Zephyr". Something along the lines of

> Running west update doesn't actually update Zephyr. It only keeps the modules in a known state (as specified by the West manifest file.)
> To actually update Zephyr.
> `cd <base>/zephyrproject/zephyr`, followed by a `git pull`.
> After that, run `west update` [...]

carlescufi, 2021-02-12:
@dueringa perhaps you would consider sending a Pull Request against the corresponding `.rst` files in `doc/`?

dueringa, 2021-02-13:
@carlescufi I can do that. Just wanted to check if my understanding was correct, first

PR #32284 (2021-02-23): Add Zephyr update instructions to the Beyond Getting Started Guide
Add explicit instructions on how to update the Zephyr
source code. For a beginner, it might not be obvious
how to do that. "west update" sounds like it'd do that,
which is not the case, however.

Fixes #32234
closed 2021-02-23 by carlescufi