zephyrproject-rtos/zephyr #32237: twister failing locally

jharris-intel, 2021-02-11:
**Describe the bug**

While following the [contribution guide](https://docs.zephyrproject.org/latest/contribute/index.html#twister) to create a PR for [this](https://github.com/zephyrproject-rtos/zephyr/pull/30676#discussion_r574319395) comment, I realized that `twister` is failing on my local dev setup.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/zephyrproject-rtos/zephyr.git`
2. `cd zephyr`
2. `west init`
3. `west update`
4. `./scripts/twister` (as an aside, is there a way to get twister to exit after the first failing test? I can't seem to find an option for it.)
4. See errors on (seemingly) all posix tests, as well as a few other scattered failures.

**Expected behavior**

Testing that is documented as "please run [...] locally before submitting your pull request to GitHub." passes on trunk.

**Impact**

I have to guess as to if my PR changes actually break code or not, which may cause CI failures.

As per the documentation, "We highly recommend you run these tests locally to avoid any CI failures."

**Logs and console output**

```
Including boilerplate (Zephyr base): /home/jharri/r/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/jharri/r/zephyr/samples/philosophers
-- Zephyr version: 2.5.0-rc3 (/home/jharri/r/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.0") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: /home/jharri/.cache/zephyr
CMake Warning at /home/jharri/r/zephyr/cmake/host-tools.cmake:31 (message):
  Could NOT find dtc: Found unsuitable version "1.4.5", but required is at
  least "1.4.6" (found /usr/bin/dtc).  Optional devicetree error checking
  with dtc will not be performed.
Call Stack (most recent call first):
  /home/jharri/r/zephyr/cmake/app/boilerplate.cmake:516 (include)
  /home/jharri/r/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/jharri/r/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /home/jharri/r/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /home/jharri/r/zephyr/twister-out/native_posix/samples/philosophers/sample.kernel.philosopher/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/jharri/r/zephyr/twister-out/native_posix/samples/philosophers/sample.kernel.philosopher/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/jharri/r/zephyr/twister-out/native_posix/samples/philosophers/sample.kernel.philosopher/zephyr/include/generated/device_extern.h

warning: The choice symbol LOG_MODE_IMMEDIATE (defined at subsys/logging/Kconfig.mode:15) was
selected (set =y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LOG_MODE_IMMEDIATE.html and/or look up
LOG_MODE_IMMEDIATE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/jharri/r/zephyr/Kconfig
Loaded configuration '/home/jharri/r/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/home/jharri/r/zephyr/samples/philosophers/prj.conf'
Configuration saved to '/home/jharri/r/zephyr/twister-out/native_posix/samples/philosophers/sample.kernel.philosopher/zephyr/.config'
Kconfig header saved to '/home/jharri/r/zephyr/twister-out/native_posix/samples/philosophers/sample.kernel.philosopher/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Deprecation Warning at ../../modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
[... 132 lines cut ...]

Also see the attached [twister.zip](https://github.com/zephyrproject-rtos/zephyr/files/5967968/twister.zip) (zipped because it's a rather giant log file).

Overall:

```
INFO    - 5016 of 5421 test configurations passed (92.53%), 405 failed, 9558 skipped with 0 warnings in 1920.87 seconds
INFO    - In total 55296 test cases were executed, 124133 skipped on 323 out of total 329 platforms (98.18%)
INFO    - 2920 test configurations executed on platforms, 2501 test configurations were only built.
```

**Environment (please complete the following information):**
- OS: Ubuntu 18.04.5 LTS
- Toolchain (e.g Zephyr SDK, ...): how do I collect this information?
- Commit SHA or Version used: 
```
jharri@jharri-compilebox-01 ~/r/zephyr (master) $ git log -n 1 --oneline
9f858ac1b6 (HEAD -> master, origin/master, origin/HEAD) tests: drivers: can: timing: Fix potential div by zero
```

**Additional context**

I see the DTS warning in the above; I don't think it is related (The log message does say "Optional", but famous last words?). Unfortunately, 1.4.5. is the latest version available in Ubuntu 18.04 LTS.

I'm probably just missing something obvious here. E.g. I'm using a different default `ld` version or something that the tests aren't compatible with but don't override.

galak, 2021-02-11:
Can you see which version of `ld` you are running `ld --version`.

The docker image we use in CI is based on ubuntu 18.04.5:

```
root@030cbe6f9579:/# ld --version
GNU ld (GNU Binutils for Ubuntu) 2.30
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

jharris-intel, 2021-02-11:
```
jharri@jharri-compilebox-01 ~/r/zephyr (master) $ ld --version
LLD 10.0.0 (compatible with GNU linkers)
```
(which is apparently a lie :-) )

If Zephyr requires GNU ld in particular, perhaps it could explicitly use or specify `ld.bfd`? I have that version of GNU ld installed also:

```
jharri@jharri-compilebox-01 ~/r/zephyr (master) $ ld.bfd --version
GNU ld (GNU Binutils for Ubuntu) 2.30
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

galak, 2021-02-11:
@nashif has been working on LLVM support and @tejlmand can say how we specify LD and such.  The native posix target probably makes assumptions that the majority of host linux systems are setup with GNU tools at this point.

jharris-intel, 2021-02-11:
Yep, turns out that I have a short-term workaround of "setting ld to bfd before running Zephyr tests".

Longer-term, can those tests please explicitly use ld.bfd if that's what they require?

*****

There are a few other seemingly-unrelated failures still. In particular:

2 flaky tests (reran and they passed):

1. qemu_x86_64_nokpti samples/kernel/condition_variables/condvar/sample.kernel.cond_var
    - Annoyingly, all I've got is the snippit from `twister.log` here, not the full output:
```
2021-02-11 13:13:41,043 - twister - DEBUG - QEMU (71369): inc_count: thread 3, count = 9, unlocking mutex
2021-02-11 13:13:41,044 - twister - DEBUG - QEMU (71369): inc_count: thread 2, count = 10, unlocking mutex
2021-02-11 13:13:41,044 - twister - DEBUG - QEMU (71369): E: Page fault at address 0xb9 (error code 0x10)
2021-02-11 13:13:41,044 - twister - DEBUG - QEMU (71369): E: Linear address not present in page tables
2021-02-11 13:13:41,045 - twister - DEBUG - QEMU (71369): E: Access violation: supervisor thread not allowed to execute
2021-02-11 13:13:41,045 - twister - DEBUG - QEMU (71369): E: PTE: not present
2021-02-11 13:13:41,046 - twister - DEBUG - QEMU (71369): E: RAX: 0x000000000010b370 RBX: 0x000000000010b370 RCX: 0x0000000000000001 RDX: 0x0000000000000001
2021-02-11 13:13:41,046 - twister - DEBUG - QEMU (71369): E: RSI: 0x0000000000000246 RDI: 0x000000000010f800 RBP: 0x0000000000115380 RSP: 0x0000000000115358
2021-02-11 13:13:41,047 - twister - DEBUG - QEMU (71369): E:  R8: 0x0000000000000001  R9: 0x000000007fd8cf35 R10: 0x0000000000000000 R11: 0x0000000000000000
2021-02-11 13:13:41,047 - twister - DEBUG - QEMU (71369): E: R12: 0x000000000010b370 R13: 0x0000000000000246 R14: 0x0000000000000246 R15: 0x000000000000000a
2021-02-11 13:13:41,047 - twister - DEBUG - QEMU (71369): E: RSP: 0x0000000000115358 RFLAGS: 0x0000000000000247 CS: 0x0018 CR3: 0x0000000000131000
2021-02-11 13:13:41,048 - twister - DEBUG - QEMU (71369): E: RIP: 0x00000000000000b9
2021-02-11 13:13:41,048 - twister - DEBUG - QEMU (71369): E: Attempt to resume un-suspended thread object
2021-02-11 13:13:41,048 - twister - DEBUG - QEMU (71369): E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
2021-02-11 13:13:41,048 - twister - DEBUG - QEMU (71369): E: Current thread: 0x10b370 (unknown)
2021-02-11 13:13:41,048 - twister - DEBUG - QEMU (71369): E: Halting system
2021-02-11 13:13:41,052 - twister - DEBUG - QEMU (71369) complete (unexpected eof) after 2.793084144592285 seconds
```
    - Not reproducable, likely a race condition.
2. qemu_x86_64_nokpti tests/kernel/mbox/mbox_usage/kernel.mailbox.usage
    - Ditto, on both accounts.
```
2021-02-11 13:36:45,568 - twister - DEBUG - QEMU (68187): Booting from ROM..*** Booting Zephyr OS build v2.5.0-rc3-79-g9f858ac1b6bc  ***
2021-02-11 13:36:45,580 - twister - DEBUG - QEMU (68187): Running test suite test_mbox
2021-02-11 13:36:45,580 - twister - DEBUG - QEMU (68187): ===================================================================
2021-02-11 13:36:45,580 - twister - DEBUG - QEMU (68187): START - test_msg_receiver
2021-02-11 13:36:45,589 - twister - DEBUG - QEMU (68187): ASSERTION FAIL [thread->base.pended_on] @ WEST_TOPDIR/kernel/sched.c:516
2021-02-11 13:36:45,590 - twister - DEBUG - QEMU (68187): E: RAX: 0x0000000000000004 RBX: 0x000000000010b040 RCX: 0x0000000000000001 RDX: 0x0000000000000001
2021-02-11 13:36:45,592 - twister - DEBUG - QEMU (68187): E: RSI: 0x0000000000000204 RDI: 0x000000000010a2cf RBP: 0x0000000000115370 RSP: 0x0000000000115358
2021-02-11 13:36:45,593 - twister - DEBUG - QEMU (68187): E:  R8: 0x0000000000000001  R9: 0x0000000000115100 R10: 0x00000000ffffffff R11: 0x0000000000000035
2021-02-11 13:36:45,595 - twister - DEBUG - QEMU (68187): E: R12: 0x000000000010b040 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
2021-02-11 13:36:45,596 - twister - DEBUG - QEMU (68187): E: RSP: 0x0000000000115358 RFLAGS: 0x0000000000000002 CS: 0x0018 CR3: 0x000000000012e000
2021-02-11 13:36:45,596 - twister - DEBUG - QEMU (68187): E: RIP: 0x00000000001012c7
2021-02-11 13:36:45,597 - twister - DEBUG - QEMU (68187): E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
2021-02-11 13:36:45,597 - twister - DEBUG - QEMU (68187): E: Current thread: 0x10c240 (ztest_thread)
2021-02-11 13:36:45,599 - twister - DEBUG - QEMU (68187): E: Halting system
2021-02-11 13:36:45,604 - twister - DEBUG - QEMU (68187) complete (unexpected eof) after 0.6466786861419678 seconds
```

And a few deterministic failures (or seemingly so), in seemingly two buckets:

Bucket A:

1. `native_posix` `tests/subsys/openthread/openthread.radio` -> `fatal error: openthread/platform/radio.h: No such file or directory` [native_posix_tests_subsys_openthread_openthread_radio.log](https://github.com/zephyrproject-rtos/zephyr/files/5968710/native_posix_tests_subsys_openthread_openthread_radio.log)

Bucket B:

1. `nrf5340dk_nrf5340_cpuappns` `samples/synchronization/sample.kernel.synchronization` (and `tests/kernel/common/kernel.common`)-> `[Errno 2] No such file or directory: '/home/jharri/r/zephyr/../modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072.pem'`
2. `nrf5340pdk_nrf5340_cpuappns`, same examples and same error.

Both of these buckets look like path issues. Is there additional setup I need to do beyond what I mentioned at the start? If so, what, and if so, can that please be documented better?

Also, should I make separate tickets for these, or reuse this one?

galak, 2021-02-11:
> Yep, turns out that I have a short-term workaround of "setting ld to bfd before running Zephyr tests".
> 
> Longer-term, can those tests please explicitly use ld.bfd if that's what they require?

We'd need a survey of common linux distro and if the majority support ld.bfd.  Or possible try ld.bfd and fall back on ld.

galak, 2021-02-11:
> There are a few other seemingly-unrelated failures still. In particular:
> 
> 2 flaky tests (reran and they passed):
> 
>     1.

QEMU is know to have issues when run under load w/twister.

galak, 2021-02-11:
> And a few deterministic failures (or seemingly so), in seemingly two buckets:
>
> Bucket A:
>
>     1. `native_posix` `tests/subsys/openthread/openthread.radio` -> `fatal error: [...]

Did you do a `west update` to get all the modules pulled down and sync'd?

galak, 2021-02-11:
> Did you do a `west update` to get all the modules pulled down and sync'd?

Hmm, it feels like there is something else with your setup.  We do nightly builds of Zephyr and aren't seeing any similar issues.

How many tests are failing?  can you report the specific summary here.

For example for the `root-RSA-3072.pem` do you have that file?  If not something else is wrong w/your setup.

jharris-intel, 2021-02-11:
> We'd need a survey of common linux distro and if the majority support ld.bfd. Or possible try ld.bfd and fall back on ld.

Hm. I, for one, would prefer to know I was running in an unsupported configuration up front, rather than failures later.

> Did you do a west update to get all the modules pulled down and sync'd?

Hm. I can reproduce this from "scratch" with the steps at the start. That is, start with a shell in an empty folder, then do:

1. `git clone https://github.com/zephyrproject-rtos/zephyr.git`
2. `cd zephyr`
3. `west init`
4. `west update`
5. `./scripts/twister`

...which likely means that something in the above is subtly incorrect. (Or blatantly incorrect and I'm just missing something.)

> QEMU is know to have issues when run under load w/twister.

Hm. Is this QEMU itself? Or just exposing race conditions in Zephyr that are more prominent under system load?

> For example for the root-RSA-3072.pem do you have that file?

I do, but not in that directory (note the directory path ends up outside the repo!)

galak, 2021-02-11:
> > We'd need a survey of common linux distro and if the majority support ld.bfd. Or possible try ld.bfd and fall back on ld.
> 
> Hm. I, for one, would prefer to know I was running in an unsupported configuration up front, rather than failures later.
> 

You're the first person I'm aware of to run into this issue.  As I mentioned, I'm guessing the majority of distro installs probably still default to GNU tools.  We check version of other tools, so a version check could be added to catch such an issue.

> Hm. I can reproduce this from "scratch" with the steps at the start. That is, start with a shell in an empty folder, then do:
>
>     1. `git clone [...]

So I think the about is subtly incorrect.  As you did a `git clone`, I think step 3, should be something like `west init -l .`.  Based on what you've got you probably have <ROOT>/zephyr/{bunch of zephyr files like west.yml, arch/ boards/, etc} and <ROOT/zephyr/zephyr/{bunch of zephyr files}.  Effectively have 2 clones of zephyr
 
> > QEMU is know to have issues when run under load w/twister.
> 
> Hm. Is this QEMU itself? Or just exposing race conditions in Zephyr that are more prominent under system load?

Its believed to be an issue w/QEMU itself.  

> 
> > For example for the root-RSA-3072.pem do you have that file?
> 
> I do, but not in that directory (note the directory path ends up outside the repo!)

Sure, just confused by the specific error as its saying no such file or dir.

jharris-intel, 2021-02-11:
> So I think the about is subtly incorrect. As you did a git clone, I think step 3, should be something like west init -l .. Based on what you've got you probably have /zephyr/{bunch of zephyr files like west.yml, arch/ boards/, etc} and <ROOT/zephyr/zephyr/{bunch of zephyr files}. Effectively have 2 clones of zephyr

Yep, that's exactly the case. I'll retry with `west init -l .` and let you know how it goes.

jharris-intel, 2021-02-11:
> You're the first person I'm aware of to run into this issue. As I mentioned, I'm guessing the majority of distro installs probably still default to GNU tools. We check version of other tools, so a version check could be added to catch such an issue.

It's a little unfortunate, indeed. Linux has all of the infrastructure to have system default tools with people changing around for different implementations, but there's now too much software that assumes that e.g. `cc` is always `gcc` , which results in people not actually changing the configuration because it breaks too many things, which in turn means that more software makes those assumptions because the assumptions aren't challenged as often, and so on.

> Its believed to be an issue w/QEMU itself.

Oh interesting. Is there an upstream bug report for this?

jharris-intel, 2021-02-11:
Hm. With `west init -l .`, `west update` seems to be dumping a lot of stuff into the _parent_ directory, (`modules` folder, etc, etc) in this case my main toplevel repository folder.

I usually have a folder structure that's essentially:

```
~/r/ -> parent folder for all git repositories
~/r/foo/ -> some repository that I'm working on's git root folder
~/r/bar/ -> some other repository that I'm working on's git root folder
~/r/baz/ -> some other repository that I'm working on's git root folder
```
etc. This makes it (relatively) straightforward to do e.g. mass fetches of repositories and avoids repos in deeply nested folders (which can be a problem in some cases).

Is there no way to do this with zephyr?

For now I suppose I can make a folder `~/r/z/` and use that as the base `west` folder, but then I'm "always" working in `~/r/z/zephyr`, assuming I understand how this works correctly.

galak, 2021-02-11:
> Hm. With `west init -l .`, `west update` seems to be dumping a lot of stuff into the _parent_ directory, (`modules` folder, etc, etc) in this case my main toplevel repository folder.
>
> I usually have [...]

@mbolivar-nordic is probably the best person to answer this.

galak, 2021-02-11:
> > Its believed to be an issue w/QEMU itself.
> 
> Oh interesting. Is there an upstream bug report for this?

Take a look at https://github.com/zephyrproject-rtos/zephyr/issues/14173

jharris-intel, 2021-02-12:
@galak  - yep, rerunning with the workarounds from this thread and I only get a couple of QEMU timeouts/unexpected EOF. Good catch.

```
INFO    - Total complete: 5740/6670  86%  skipped: 9402, failed:    0
ERROR   - qemu_x86_64               tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls FAILED: unexpected eof
ERROR   - see: /home/jharri/r/z/zephyr/twister-out/qemu_x86_64/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/handler.log
INFO    - Total complete: 6427/6670  96%  skipped: 9498, failed:    1
ERROR   - qemu_x86_64               tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED: Timeout
ERROR   - see: /home/jharri/r/z/zephyr/twister-out/qemu_x86_64/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
INFO    - Total complete: 6437/6670  96%  skipped: 9498, failed:    2
ERROR   - qemu_x86_64_nokpti        tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED: Timeout
ERROR   - see: /home/jharri/r/z/zephyr/twister-out/qemu_x86_64_nokpti/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
INFO    - Total complete: 6670/6670  100%  skipped: 9558, failed:    3
```

And then rerunning with --only-failed

```
ERROR   - qemu_x86_64               tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls FAILED: unexpected eof
ERROR   - see: /home/jharri/r/z/zephyr/twister-out/qemu_x86_64/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/handler.log
INFO    - Total complete:    3/   3  100%  skipped:    0, failed:    1
```

Unfortunately, this one failed the same way 3/4 runs on my box. Not the end of the world, but unfortunate.

mbolivar-nordic, 2021-02-13:
> For now I suppose I can make a folder `~/r/z/` and use that as the base `west` folder, but then I'm "always" working in `~/r/z/zephyr`, assuming I understand how this works correctly.

You need to create a west workspace. `west init -l .` means `.` is your manifest repository directory, so `..` is the workspace topdir, and `../modules` is where the modules will go, etc.

Details in https://docs.zephyrproject.org/latest/guides/west/basics.html and related pages.

carlescufi, 2021-02-13:
> Hm. With `west init -l .`, `west update` seems to be dumping a lot of stuff into the _parent_ directory, (`modules` folder, etc, etc) in this case my main toplevel repository folder.

I was looking at your logs and though that your `r/` stood for "repos" so I immediately thought of that before seeing your comment. The Getting Started Guide does [ask you](https://docs.zephyrproject.org/latest/getting_started/index.html#get-zephyr-and-install-python-dependencies) to create an enclosing folder (using `west`, but you can do that manually too since `west init` defaults to `pwd`:

```
west init ~/zephyrproject
cd ~/zephyrproject
west update
```

For now, moving your `r/zephyr` folder into `r/z/zephyr` and then running `west init  -l` inside it will do the trick. Don't forget to delete all the modules in `r/`, and then `r/.west`, which will have been created by it and marks the top dir of the west workspace.

carlescufi, 2021-02-13:
FYI @jharris-intel, GitHub/CI will run `twister` for you when you post the PR, so running it locally is technically optional. That said, having a working `twister` setup on your local machine definitely helps enormously.

tejlmand, 2021-02-15:
>  instead of implicitly relying on the default system linker being ld?

well, expecting `ld` to be `ld` doesn't seem wrong to me.
The fact that `lld` can work as a drop-in replacement of `ld` in most cases doesn't make it 100% compatible with `ld`.

We still have some work to do before linker scripts are updated to support lld, but it is part of our plans.
Even when compiling using `clang` we still have to link with `ld`.
https://github.com/zephyrproject-rtos/zephyr/blob/5396350701cfd70834c61bccafe8bf94cb018c68/cmake/toolchain/llvm/generic.cmake#L12

Instead of trying to detect if users has `lld` as an `ld` replacement, I think it would be better to get proper lld support in Zephyr.
Thus i'm not sure Zephyr build system should try to handle the `lld` as `ld` replacement situation here.

In this case, users has the ability of telling CMake (gcc) to actually use `ld.bfd` instead.
This can be done as:
```
cmake -DCMAKE_C_FLAGS="-fuse-ld=bfd" .....
```
or when using `twister`:
```
./scripts/twister -p native_posix -T samples/philosophers/ -x=CMAKE_C_FLAGS="-fuse-ld=bfd"
```

this will tell the build system to specifically use `ld.bfd`, which I think is sufficient, and works.

We could try to make documentation clearer on the fact that Zephyr cannot link using `lld` at the moment, and how users that has `lld` as `ld` replacement still can link.

jharris-intel, 2021-02-16:
> The Getting Started Guide does ask you to create an enclosing folder (using west, but you can do that manually too since west init defaults to pwd:

FYI in terms of helping with documentation (and hopefully making so that the next person is less confused than I was): I had seen that but had assumed that that was just for the purposes of setting up the mentioned cmake and pip stuff, not something that had to be done for every local repo copy.

> FYI @jharris-intel, GitHub/CI will run twister for you when you post the PR, so running it locally is technically optional. That said, having a working twister setup on your local machine definitely helps enormously.

Good to know...

> well, expecting ld to be ld doesn't seem wrong to me.

Sure. It's more `expecting ld to be specifically ld.bfd` that's the issue here. (Akin to someone using `#!/bin/sh` for a `bash` script - sure, it works if your default shell is `bash` - but if your script relies on `bash` then please say so instead of having strange issues down the line when you run on a machine whose default shell is e.g. `dash`.)

tejlmand, 2021-02-16:
> Sure. It's more expecting ld to be specifically ld.bfd that's the issue here. 

I see your point.

nashif, 2021-02-17:
> Even when compiling using `clang` we still have to link with `ld`.

this is already moving forward, see https://github.com/zephyrproject-rtos/zephyr/pull/31706/commits/265532b26030e3cd6cf942babf8bd4525eedbd7d

```
i9:zephyr(oneAPI): west build -b native_posix samples/hello_world/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/nashif/Work/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/nashif/Work/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.5.99 (/home/nashif/Work/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.9 (found suitable exact version "3.9.1") found components: Interpreter
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: native_posix
-- Cache files will be written to: /home/nashif/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: host (clang/ld)
-- Found BOARD.dts: /home/nashif/Work/zephyrproject/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /home/nashif/Work/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/nashif/Work/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/nashif/Work/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h

warning: The choice symbol LOG_MODE_IMMEDIATE (defined at subsys/logging/Kconfig.mode:15) was
selected (set =y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LOG_MODE_IMMEDIATE.html and/or look up
LOG_MODE_IMMEDIATE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/nashif/Work/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/nashif/Work/zephyrproject/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/home/nashif/Work/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/nashif/Work/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/nashif/Work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is Clang 11.0.0
-- The CXX compiler identification is Clang 11.0.0
-- The ASM compiler identification is Clang
-- Found assembler: /usr/lib64/ccache/clang
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nashif/Work/zephyrproject/zephyr/build
-- west build: building application
[1/85] Preparing syscall dependency handling

[85/85] Linking C executable zephyr/zephyr.elf

i9:zephyr(oneAPI): grep lld build/CMakeCache.txt
CMAKE_LINKER:FILEPATH=/usr/bin/ld.lld
```

PR #35674 (2021-05-26): ld.bfd: ensure that ld.bfd is preferred over ld.
Fixes: #32237

When building for native_posix, then host tools are used.
This means that gcc will link using `/usr/bin/ld` per default.

If ld points to lld, then linking will fail.

This commit will first look for ld.bfd, and if found then use
-fuse-ld=bfd for linking. If ld.bfd is not found, then ld is used as
fallback as that will be assumed to be the best working candidate.
commit 39f06e01 (2021-05-26): ld.bfd: ensure that ld.bfd is preferred over ld.

Fixes: #32237

When building for native_posix, then host tools are used.
This means that gcc will link using `/usr/bin/ld` per default.

If ld points to lld, then linking will fail.

This commit will first look for ld.bfd, and if found then use
-fuse-ld=bfd for linking. If ld.bfd is not found, then ld is used as
fallback as that will be assumed to be the best working candidate.
closed 2021-05-26 by galak