zephyrproject-rtos/zephyr #32238: Request to upgrade BBC Microbit display driver to support Microbit v2 model

key1066, 2021-02-11:
**Is your enhancement proposal related to a problem? Please describe.**
The current version of the BBC Microbit display driver doesn't work with Microbit v2.

**Describe the solution you'd like**
In Microbit v2, the LED matrix display has changed internally from a logical 3x9 mapping to a 5x5. An upgrade would be required to support Microbit v2 with backward compatibility to the original Microbit model.

**Describe alternatives you've considered**
I have tested basic control of the new LED matrix layout but this has been limited to simple LED and PWM tests.

The BBC Microbit developers from Lancaster University offer display control but currently only for Microbit v1. It appears that so far, only Micropython from the BBC Microbit supports the new Microbit v2 display layout.

Also the Arduino core for the BBC Microbit currently only supports v1. 

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

carlescufi, 2021-02-12:
I have some patches for this, but the agreement was to introduce a proper generic led-matrix display drivers with correct Devicetree bindings instead of just updating the old driver. I don't currently have time to do that, maybe @LingaoM does.

key1066, 2021-02-12:
Thanks Carles.
Agreed, a new version of the display driver with devicetree bindings would be preferable.

key1066, 2021-02-15:
I've reached out to Lingao Meng for his comments and potential plans. Thanks.

LingaoM, 2021-02-22:
@key1066  CC https://github.com/zephyrproject-rtos/zephyr/pull/32532

key1066, 2021-03-02:
What’s the current plan for microbit v2 display support?
Is the plan to develop a display driver with devicetree bindings or go ahead with @LingaoM update? Thanks.

key1066, 2021-03-14:
Hi,

Any updates on where things are with getting microbit v2 display support in the next Zephyr  release?
 
Are we going ahead with your existing changes or are you going to develop a display driver with devicetree bindings as suggested by @carlescufi?

Thanks.

carlescufi, 2021-04-19:
@key1066 we need reviewers for #32532, specifically deciding which direction to take there: whether to merge it as-is or to turn it into a generic LED matrix driver.

peterniebert, 2021-06-11:
> @key1066 we need reviewers for #32532, specifically deciding which direction to take there: whether to merge it as-is or to turn it into a generic LED matrix driver.

IMHO, there is no such thing as a "typical" LED matrix in terms of pcb wiring.  In the case of the micro:bit, it is directly obtained by from GPIO, but it is much more frequent to find LED drivers that require an SPI like protocol to push the data. Some drivers implement PWM, others do not, etc. 

So it does make sense to have something generic in terms of contents and geometry, but the implementation will most likely vary from one pcb to another to be efficient. User supplied drawing and refresh functions could be the answer, but not something like a user defined wiring topology. Having the latter generic would mean an enormous complexity in the specification, whereas writing the update function is most likely not that difficult.

Regards
-- P.

peterniebert, 2021-06-15:
Submitted a related pullrequest, see here:
https://github.com/zephyrproject-rtos/zephyr/pull/36264

peterniebert, 2021-06-17:
OK, now I really think I got it right. With the exception of the right pin for sound output, all the microbit samples seem to work with the V2 and the new display driver and it has become much simpler.

https://github.com/zephyrproject-rtos/zephyr/pull/36280

zpm1066, 2021-06-17:
Very good! I look forward to trying out the new display driver. Thank you.

chudov, 2021-11-05:
Yet another attempt to support Micro:bit v2 board #40144

carlescufi, 2021-12-21:
This is now a duplicate of #40208

closed 2021-12-21 by carlescufi