zephyrproject-rtos/zephyr #32252: Building anything for nrf5340pdk_nrf5340_cpuappns pulls external git trees

nashif, 2021-02-12:
**Describe the bug**
Whenever you build for this platform, you end up with git trees that were pulled from outside the project. This should use modules and should not depend on external trees.

```
west build -b nrf5340pdk_nrf5340_cpuappns samples/synchronization/
..
..
[1/9] Creating directories for 'mbedcrypto-populate'
[1/9] Performing download step (git clone) for 'mbedcrypto-populate'
Cloning into 'mbedcrypto-src'...
remote: Enumerating objects: 18042, done.
remote: Counting objects: 100% (18042/18042), done.
remote: Compressing objects: 100% (8487/8487), done.
remote: Total 18042 (delta 14039), reused 12665 (delta 9429), pack-reused 0
Receiving objects: 100% (18042/18042), 17.40 MiB | 38.24 MiB/s, done.
Resolving deltas: 100% (14039/14039), done.
Note: switching to 'mbedtls-2.24.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 523f0554 Merge pull request #748 from ARMmbed/mbedtls-2.24.0r0-pr
[2/9] Performing update step for 'mbedcrypto-populate'
[4/9] Performing patch step for 'mbedcrypto-populate'
[5/9] No configure step for 'mbedcrypto-populate'
[6/9] No build step for 'mbedcrypto-populate'
[7/9] No install step for 'mbedcrypto-populate'
[8/9] No test step for 'mbedcrypto-populate'
[9/9] Completed 'mbedcrypto-populate'

...
..
```

ioannisg, 2021-02-22:
There's a possibility we can fix this just with the following upmerge. I'll get back to it soon.

galak, 2021-04-06:
Waiting on v1.3.0 TFM release which will have a fix for having a local mbed crypto/tls code base

galak, 2021-04-13:
@ioannisg can you take a look at this now that we merged the TFM 1.3.0 support.

ioannisg, 2021-04-13:
> @ioannisg can you take a look at this now that we merged the TFM 1.3.0 support.

It's on my task list for this week.

ioannisg, 2021-04-14:
Submitted a PR that addresses the issue.

PR #34984 (2021-05-09): mbedtls: update to 2.26 and use this for TF-M
- Update mbedtls to 2.26
- Moving glue code into the main tree
- Adding mbedcrypto path for tf-m

Fixes #32252