zephyrproject-rtos/zephyr #32256: Bluetooth mesh : Long friendship establishment after LPN reset

Ripitchip, 2021-02-12:
**Describe the bug**
If the low power node is resetted and added again to the network, depending of the scenario, the establishment of friendship is kinda long or not possible.

**To Reproduce**
The scenario are based on the [on-off ](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/nrf/mesh/onoff-app) sample applied to some nRF52840.

The **LPN** configuration :
```
CONFIG_BT_DEVICE_NAME="Zephyr LPN"
CONFIG_BT_MESH=y
CONFIG_BT_MESH_RELAY=n
CONFIG_BT_MESH_FRIEND=n
CONFIG_BT_MESH_LOW_POWER=y
CONFIG_BT_MESH_DEBUG=y
CONFIG_BT_MESH_DEBUG_LOW_POWER=y
# Other have default value
```

The **Friend** configuration :
```
CONFIG_BT_DEVICE_NAME="Zephyr Friend"
CONFIG_BT_MESH=y
CONFIG_BT_MESH_RELAY=n
CONFIG_BT_MESH_LOW_POWER=n
CONFIG_BT_MESH_FRIEND=y
CONFIG_BT_MESH_FRIEND_ENABLED=y
CONFIG_BT_MESH_FRIEND_QUEUE_SIZE=16
CONFIG_BT_MESH_DEBUG=y
CONFIG_BT_MESH_DEBUG_FRIEND=y
# Other have default value
```

***First scenario***
- Provision a friend
- Provision the LPN
- Reset the LPN. Wait for Friend to lose friendship and clear LPN.
- Provision the LPN again

Here, the friendship is established pretty quickly, like "only" after three 
```
[00:33:14.959,991] <wrn> bt_mesh_lpn: No acceptable Friend Offers received 
```

***Second scenario***
- Provision a friend
- Configure the friend : Generic on-off server suscribed to address 0xC000
- Provision the LPN
- Configure the LPN : Generic on-off client publishes to address 0xC000
- Reset the LPN. Wait for Friend to lose friendship and clear LPN.
- Provision the LPN again

The friendship establishement is longer. I had to wait for 12 
```
[00:33:14.959,991] <wrn> bt_mesh_lpn: No acceptable Friend Offers received 
```
until it was established again.

***Third scenario***
- Provision a friend
- Configure the friend : Generic on-off client suscribed to address 0xC000
- Provision the LPN
- Configure the LPN : Generic on-off server publishes to address 0xC000 with an interval of ca. 5 seconds.
- Reset the LPN. Wait for Friend to lose friendship and clear LPN.
- Provision the LPN again

The friendship establishement far longer. It's established after a while but I had to wait like ... 3-5 minutes ?

To make server publication possible, I added this code :
```
static int test_update(struct bt_mesh_model *mod);
BT_MESH_MODEL_PUB_DEFINE(gen_onoff_pub_srv, test_update, 2 + 2);

static int test_update(struct bt_mesh_model *mod){
        printk("pub_update called \n");
	struct net_buf_simple *msg = mod->pub->msg;
	bt_mesh_model_msg_init(msg, BT_MESH_MODEL_OP_GEN_ONOFF_STATUS);
	net_buf_simple_add_u8(msg, onoff_state->current);
	return 0;
}
```

**Expected behavior**
I expect the LPN be able to establish a friendship quickly with the friend upon provisioning (like the first time).

**Impact**
No real impact, as we can set up the network correctly the first time. Just a bit annoying when change are needed.

**Environment (please complete the following information):**
 - Linux Ubuntu 20.04.1 LTS
 - Toolchain : Zephyr SDK
 - `e7a335fb9d8b3dd8ee16976d4512946c9fa16b71` with this the addition of the PR correction [#32080 ](https://github.com/zephyrproject-rtos/zephyr/pull/32080)

LingaoM, 2021-02-20:
@Ripitchip Bluetooth Mesh has replay protection. The friend node will record the sequence number of the LPN. The sequence number of each message must be greater than the previous one, otherwise it will be considered a replay attack. If you reprovisioning the LPN, then the sequence number of the LPN will be cleared at this time. If the configuration node address is the same as the last time, the message will be considered a replay attack.

carlescufi, 2021-03-11:
@Ripitchip00 can we close this as per @LingaoM's comment above?

Ripitchip00, 2021-03-11:
Sorry, lots of things came up the last weeks.

Yes, this answer does satisfy me. I wanted to test it again with changing the node's address when provisioning but I hadn't time to do it. Nevertheless, thanks for the explanation and the general help, we were able to finish two projects related to Bluetooth mesh :).

Best regards.

closed 2021-03-11 by Ripitchip00