zephyrproject-rtos/zephyr #32258: power mgmt: pm_devices: Get rid of z_pm_core_devices array

erwango, 2021-02-12:
**Is your enhancement proposal related to a problem? Please describe.**
When using PM_DEVICE, a `z_pm_core_devices` table is defined and has to be populated per platform but undocumented and its use is unclear. 

```

/*
 * FIXME: Remove the conditional inclusion of
 * core_devices array once we enble the capability
 * to build the device list based on devices power
 * and clock domain dependencies.
 */

__weak const char *const z_pm_core_devices[] = {
#if defined(CONFIG_SOC_FAMILY_NRF)
	"CLOCK",
	"sys_clock",
	"UART_0",
#elif defined(CONFIG_SOC_SERIES_CC13X2_CC26X2)
	"sys_clock",
	"UART_0",
#elif defined(CONFIG_SOC_SERIES_KINETIS_K6X)
	DT_LABEL(DT_INST(0, nxp_kinetis_ethernet)),
#elif defined(CONFIG_NET_TEST)
	"",
#elif defined(CONFIG_SOC_SERIES_STM32L4X) || defined(CONFIG_SOC_SERIES_STM32WBX)
	"sys_clock",
	
#endif
	NULL
};
```
I think we should get rid of it in scope of PM_DEVICE rework

erwango, 2021-02-12:
^^ @ceolin

nashif, 2021-05-28:
should be resolved now

closed 2021-05-28 by nashif