zephyrproject-rtos/zephyr #32269: shield: cmake: Shield conf is not loaded during build

Zzzwierzak, 2021-02-12:
**Describe the bug**
Shield conf is not loaded during build

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz -DSHIELD=abc
3. make
4. See error - abc.proj is not taken to build. 

**Expected behavior**
abc.conf is used during cmake 

**Impact**
Missing configuration(in autoconf.h) for shield. 

**Workaround/Fix**
zephyr/cmake/app/boilerplate.cmake

Change line 350 from : 
if(EXISTS ${shield_dir}/${s_dir}/${s}.conf)
To:
if(EXISTS ${SHIELD_DIR_${s}}/${s}.conf)

erwango, 2021-02-12:
@tejlmand

carlescufi, 2021-02-12:
@jfischer-no and @erwango can you confirm?

jfischer-no, 2021-02-12:
> @jfischer-no and @erwango can you confirm?

No. Just tried it out, the shields we have in the tree work fine.
This issue description is weird. It is been a long time since we switched from *.conf to similar Kconfig structure as in boards.

tejlmand, 2021-02-15:
@Zzzwierzak thanks for reporting, fixed here: #32318

Zzzwierzak, 2021-02-15:
@tejlmand Thank you.

PR #32318 (2021-03-01): shield: cmake: adjust `if(EXISTS <shield>.conf)` to check for right file
Fixes: #32269

This commit is a followup to #31877 where the `${shield_dir}/${s_dir}`
was adjusted to `${SHIELD_DIR_${s}}` to ensure only including shields
and not boards in list of known shields.

An `if(EXISTS ${shield_dir}/${s_dir}...)` was missed in this process
which causes #32269. This is corrected in this commit.
commit 48321ef2 (2021-03-01): shield: cmake: adjust if(EXISTS <shield>.conf) to check for right file

Fixes: #32269

This commit is a followup to #31877 where the `${shield_dir}/${s_dir}`
was adjusted to `${SHIELD_DIR_${s}}` to ensure only including shields
and not boards in list of known shields.

An `if(EXISTS ${shield_dir}/${s_dir}...)` was missed in this process
which causes #32269. This is corrected in this commit.
closed 2021-03-01 by MaureenHelm