zephyrproject-rtos/zephyr #32279: Question about flasing Adafruit Feather Sense with Zephyr

jimakos96, 2021-02-12:
Hello,i am looking at the supported boards and i see  **adafruit_feather_nrf52840** and i was wandering if i can use this to flash my board the Adafruit Feather Sense.Both devices have the nrf52840 processor and i checked that the device tree has the same IO mapping .

dastarling, 2021-02-13:
I have this board too and was thinking the same thing.  However the Feather Sense only seems to have the programming signals SWCLK and SWDIO as test points on the back of the board.  So likely to use it, you would need to solder wires to the appropriate signals for an external programmer such as Segger, etc.  So even though I like this board, it is not really that good for general firmware development outside the built-in python.  In theory, you could replace the firmware via the USB, but in experience if you brick the unit, then the only recovery method would be an external programmer.
So in summary, I think it is possible, depending on your experience with interfacing the hardware to a programmer, otherwise it is designed for Arduino interface.

I would recommend getting a nrf52dk for the nrf52840 if you want to run zephyr since it has the built-in debugger/programmer.  It could also be used as an inexpensive programmer interface to the feather Sense if you want to work with it later.

Since I don't know your level of expertise, this is the best answer I can give at the moment.

jimakos96, 2021-02-13:
I do know that i need  an external programmer to flash the binary file to the board .Maybe my question wasn't so clear , for example if i build the blinky example by executing  **west build -b adafruit_feather_nrf52840**  and flash the hex or binary via the external programmer to the adafruit feather sense will it work ?Or it only works for  the adafruit feather ?

dastarling, 2021-02-13:
Ok, I think I understand the question better.  It looks like Adafruit has 2 boards the feather nrf52840 **express** and the **sense**.  From what I see the board that is supported is the nrf52840 express.  The sense likely has sensors that may not be supported yet.  I have had to write sensor drivers for all of the sensors that I am using on a custom board.
I have not yet tested it, but it appears it would work, but may likely not support the advanced features that are present on the sense version. 
To get the blinky example to work, it looks like you would have to create an overlay file to remap the LEDs, since in the current dts file the LEDs are on P1.15 and P1.10. On the Sense the LED_0 is on P1.09 instead of P1.15, so this would have to be modified, but with that change the blinky should work.  Alternately the LED_1 is the same pin on both boards, so modifying the blinky example to use the blue LED would also likely work.
I hope this answers your question. The only disclaimer is that I have not tested this combination, that is from reviewing the related files.

jimakos96, 2021-02-13:
Thanks for your answer , apart from the different mapping of the pins which is not difficult to configure .
Does the final hex (which is going to be flashed)  file do you think it will be compatible for the sense board ?

dastarling, 2021-02-13:
I believe the binary would be compatible, but if I were to change the FW on it, I would save a copy of original image so that you can restore it later if desired. Or at least make sure it is easily available somewhere.

pabigot, 2021-02-15:
#31066 coming in soon should make it easier to support more Adafruit boards without modifying hardware to support direct flashing.

jefffhaynes, 2021-03-23:
Sorry, I'm new at this but could you give an example of how to overlay the LEDs in this case?  I'm flashing the Sense via SWD on the bottom pads but haven't been able to get the overlay to work.

dastarling, 2021-03-23:
From checking the schematic, P1.09 for Red and P1.10 for Blue appear to be correct. So you are on the right track.

jefffhaynes, 2021-03-23:
So far I haven't gotten the board past clock initialization.  However, I am trying to build from Edge Impulse stuff, not just Zephyr, so it is possible there's something else going on.

jimakos96, 2021-03-23:
The Feather sense  uses a slightly different clock source so you need to add the following macros to the prj.conf .
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_500PPM=y

jefffhaynes, 2021-03-27:
That did it, thanks!

closed 2021-04-07 by jimakos96

ryangenz, 2021-05-03:
Is it the case that the Feather Sense should have its own board definition within the 'boards' directory to account for the small Pin and Clock differences from the basic nRF52840 Feather?

WilliamGFish, 2021-05-13:
> Is it the case that the Feather Sense should have its own board definition within the 'boards' directory to account for the small Pin and Clock differences from the basic nRF52840 Feather?

Yes, the Adafruit Arduino clones should have their own definitions and the definition should be configured to use the west BOSSA runner. There already are setting in west for this, I don't have any of these boards so not able to do this.

bkietz, 2023-10-11:
@jefffhaynes I tried modifying hello_world and blinky with the changes described here, as far as I could make out:

```diff
--- /dev/null
+++ b/samples/basic/blinky/app.overlay
@@ -0,0 +1,13 @@
+/ {
+    leds {
+        compatible = "gpio-leds";
+        led0: led_0 {
+            gpios = <&gpio1 9 0>;
+            label = "Red LED";
+        };
+        led1: led_1 {
+            gpios = <&gpio1 10 0>;
+            label = "Blue LED";
+        };
+    };
+};
--- a/samples/basic/blinky/prj.conf
+++ b/samples/basic/blinky/prj.conf
@@ -1 +1,4 @@
 CONFIG_GPIO=y
+CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
+CONFIG_CLOCK_CONTROL_NRF_K32SRC_500PPM=y
+CONFIG_BUILD_OUTPUT_UF2=y
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,4 @@
-# nothing here
+# add config options to indicate we're using the feather sense
+CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
+CONFIG_CLOCK_CONTROL_NRF_K32SRC_500PPM=y
+CONFIG_BUILD_OUTPUT_UF2=y
```

I haven't been able to successfully flash my board though. Did I miss anything? See also my PR to add the board definition

DMonitor, 2024-04-19:
> The Feather sense uses a slightly different clock source so you need to add the following macros to the prj.conf . CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y CONFIG_CLOCK_CONTROL_NRF_K32SRC_500PPM=y

Just wanted to thank you for this tip. I would've spent hours looking for this if it weren't for your post.