zephyrproject-rtos/zephyr #32286: ST7789 driver enhancement to support display orientation

key1066, 2021-02-13:
**Is your enhancement proposal related to a problem? Please describe.**
Unable to change the ST7789 display orientation. 

**Describe the solution you'd like**
The ability to set the ST7789 display orientation through st7789v_set_orientation(). The function exists
for driver but it is not implemented currently.

In reviewing the drivers for the other popular displays, like the ILI9340, it appears the display orientation function for those displays have not been implemented either. 

Is the absence of the display orientation due to the level of implementation difficulty or  just a lower priority? 
It certainly would be a very useful feature. 
I suspect if and when the functionality gets added, it'll be there for all the Zephyr supported display drivers.
 
Display orientation for the above displays is available under other frameworks/platforms, for example Arduino, Adafruit, and other ESP32/STM32 display drivers.

**Describe alternatives you've considered**
I have looked into seeing if the LVGL graphics library could somehow be used. They recommend that the functionality be handled by the display driver.  

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

key1066, 2021-03-21:
@vanwinkeljan Let me know if you need any assistance in testing out the ST7789 display orientation functionality once you’ve added the support. I assume you’ll be looking at adding the set orientation function for display controller. Thanks.

key1066, 2021-04-17:
@vanwinkeljan, Any progress on the ST7789 display orientation support? Thanks.

vanwinkeljan, 2021-04-18:
@key1066 sorry no progress on this. I currently have not the time or the HW to look into this.

As you have the HW feel free to send a PR introducing orientation support.

cvetaevvitaliy, 2021-12-03:
> 

Can you help with testing?
I checked on two displays st7789
240×320 and 240×135
https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/2inch-lcd-module.htm
https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/1.14inch-lcd-module.htm

What displays can you check?

zpm1066, 2021-12-03:
Hi! 
I have several ST7789 displays from Waveshare (240x240 1.3”, 240x320 2.4” and 2”, and 240x135 1.14”) and generic 240x240 1.3” without CS. 
I’d be glad to test these out with the ST7789 driver updates. Let me know when the code is ready for testing. Thanks.

cvetaevvitaliy, 2021-12-03:
@zpm1066 Thanks for reply 
I made a request, please see #40845 
You can pull this branch and test this feature

zpm1066, 2021-12-05:
@cvetaevvitaliy, I've tested the updated driver with the following five ST7789 displays: 

- Waveshare 135x240 Pico-LCD-1.14"
https://www.waveshare.com/pico-lcd-1.14.htm

- Generic 135x240 1.14"
https://www.aliexpress.com/item/4000571253836.html?spm=a2g0o.store_pc_allProduct.8148356.2.76ef4de3OsXkJN

- Waveshare 320x240, 2" LCD Display Module For Raspberry Pi Pico
https://www.waveshare.com/pico-lcd-2.htm

- Waveshare 240x240, 1.3" IPS LCD Display HAT For Raspberry Pi
https://www.waveshare.com/product/raspberry-pi/displays/lcd-oled/1.3inch-lcd-hat.htm

-Waveshare 240x240, 1.3" LCD Display Module For Raspberry Pi Pico
https://www.waveshare.com/pico-lcd-1.14.htm

Results:
I'm using ST7789 with LVGL under Zephyr 2.70 on a micro:bit v2 with the same code for all tests except changing the st7789 width/height in the microbit_v2_overlay for the display under test.

1. The updated drivers works fine for all rotations (0,90,180, and 270) for the 135x240 and 320x240 displays.
2. Mixed results for Waveshare 240x240 displays. 
microbit_v2_overlay: width = height = 240, x-offset = y-offset = 0
Rotations 0 & 270 degrees work fine.
Rotations of 90 & 180 have display area shifted off the display by 80 but rotation direction is correct. Setting y-offset = 80 corrects the shift but introduces shift off the display for rotations of 0 and 270 degrees.

cvetaevvitaliy, 2021-12-08:
@zpm1066 Thanks, good test and detailed test report
I think when offset is set, the recalculation goes in the wrong direction.
I will try to get a 240x240 display for debug

zpm1066, 2021-12-10:
@cvetaevvitaliy, I've made some additional changes to your code to get the display rotation to work for all three types of displays under test (240x240, 240x320, and 135x240) without making any changes in my applications or the `st7789v devicetree overlays`. 

As suspected, the problem with the code was the x_offset & y_offset for the 90/180 degree rotations.

In  `st7789v_set_orientation` , I've added a test for the display size. 

If the display is 135x240 (width = 135, height = 240), no adjustments of x_offset & y_offset are required as the correct offsets are already specified in the st7789v overlay, i.e. x-offset = 52, y-offset = 40. These values are derived from row_offset = (240 - width)/2 = (240 - 135)/2 = 52 (whole integer) and col_offset = (320 - height)/2 = (320 - 240)/2 = 40. So, no x_offset/y_offset adjustments are required in the code.

If the display size is `240x240 (width = height = 240) and with x-offset = y-offset = 0 in st7789v overlay`, then the required row_offset = (320 - height) = (320 - 240) = 80 and col_offset = (240 - width) = (240 - 240) = 0. This should be handled in the code, as shown below.

If the display size is `240x320 (width = 240, height = 320) and with x-offset = y-offset = 0 in st7789v overlay`, then row_offset = col_offset = 0. There are no x_offset/y_offset adjustments required in the code.

Actually, for all the above display sizes, we could adjust the offsets in the code but I wanted to minimize changes to your code and also not modify my `st7789v device overlays` for the displays.

Below is the tested modifications to your code. Only `st7789v.c` was changed.

When you get hold of a 240x240, please review review & test. It'll be easier for you to incorporate my changes into your code, make any other changes you see fit and submit for review.

```
static int st7789v_set_orientation(const struct device *dev,
			    const enum display_orientation orientation)
{
	struct st7789v_data *data = (struct st7789v_data *)dev->data;
	uint8_t tx_data = 0;
	uint16_t x_offset = 0;
	uint16_t y_offset = 0;

        uint16_t row_offset = 0;
        uint16_t col_offset = 0;

        if (data->width < 240) {
          // 135x240 display
          row_offset = data->y_offset;
          col_offset = data->x_offset;
        }
        else {
          // 240x320 and 240x240 displays
          row_offset = (320 - data->height);
          col_offset = (240 - data->width);
        }

	switch (orientation)
	{
	case DISPLAY_ORIENTATION_NORMAL:
		tx_data |= ST7789V_MADCTL_MV_NORMAL_MODE;
		x_offset = data->x_offset;
		y_offset = data->y_offset;
		break;

	case DISPLAY_ORIENTATION_ROTATED_90:
		tx_data |= (ST7789V_MADCTL_MY_BOTTOM_TO_TOP | ST7789V_MADCTL_MV_REVERSE_MODE);
		x_offset = row_offset;
		y_offset = col_offset;
		break;

	case DISPLAY_ORIENTATION_ROTATED_180:
		tx_data |= (ST7789V_MADCTL_MY_BOTTOM_TO_TOP | ST7789V_MADCTL_MX_RIGHT_TO_LEFT);
		x_offset = col_offset;
		y_offset = row_offset;
		break;

	case DISPLAY_ORIENTATION_ROTATED_270:
		tx_data |= (ST7789V_MADCTL_MX_RIGHT_TO_LEFT | ST7789V_MADCTL_MV_REVERSE_MODE);
		x_offset = data->y_offset;
		y_offset = data->x_offset;
		break;

	default:
		LOG_ERR("Error changing display orientation");
		return -ENOTSUP;
	}

	st7789v_set_lcd_margins(data, x_offset, y_offset);
	st7789v_transmit(data, ST7789V_CMD_MADCTL, &tx_data, 1U);
	data->orientation = orientation;
	LOG_INF("Changing orientation to: '%d'", data->orientation);

	return 0;
}

```

zpm1066, 2021-12-29:
@cvetaevvitaliy, Did you manage to get hold of a 240x240 display?

zpm1066, 2022-10-02:
@vanwinkeljan, @cvetaevvitaliy ,
I now have display rotation for all the above displays updated, working and fully tested with the latest Zephyr RTOS version.

I’ll look into how to doing a pull request to get the changes into Zephyr. I may need some assistant for the pull request. Thanks.

szczys, 2023-03-20:
@zpm1066 I am working with the st7789v driver and wondering about support for rotation. Were you able to submit a PR for the work you did back in October?

zpm1066, 2023-05-04:
@szczys, Sorry for the very late response. 
Unfortunately, I never got around to submit a PR. You're welcome to do so. The above code works fine.

LuskeyNoah, 2024-04-29:
@zpm1066 I needed this, so went ahead and created a PR with your (lightly modified) code: #72063

Thanks!

digiexchris, 2024-05-07:
whoah just as I discovered this issue myself! nice work! Do you have any test code or dt overlay I can see? I'm trying to hook one of the off brand ones up to an stm32f411ce and having limited success, getting the horizontal stripes thing but I wonder if it's just the rotation issue. Not intending to hijack the thread but I see the PR and would love to test it on my board somehow too.

LuskeyNoah, 2024-05-13:
@digiexchris I used the modification to the driver in the #72107 along with the sample located at [`samples/drivers/display`](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/display).

Instead of a DT overlay, I used a `SYS_INIT` function in the directory for my (not-yet-upstreamed) board that calls `display_set_orientation(..., DISPLAY_ORIENTATION_ROTATED_180)` at startup.

```c
int esp32s3_t_watch_display_on(void)
{
	// Set the orientation
	const struct device *display = DEVICE_DT_GET(DT_CHOSEN(zephyr_display));
	if (!device_is_ready(display))
	{
		return -EIO;
	}

	int ret = display_set_orientation(display, DISPLAY_ORIENTATION_ROTATED_180);
	if (ret < 0)
	{
		return ret;
	}

	return 0;
}

SYS_INIT(esp32s3_t_watch_display_on, APPLICATION, CONFIG_APPLICATION_INIT_PRIORITY);
```