zephyrproject-rtos/zephyr #32288: Enhance the ADC functionality on the STM32 devices to all available ADC channels

key1066, 2021-02-14:
**Is your enhancement proposal related to a problem? Please describe.**
Recently I’ve been testing out ADC & PWM functionality on the nRF52840 & STM32 (blackpill_f411ce) boards under Zephyr, as part of an effort to move projects over from Arduino framework to Zephyr.

I’m finding that the current ADC support in Zephyr for the STM32 boards is quite limited. From my debug sessions, it appears that only one or two ADC channels are currently supported on the STM32. On the blackpill_f411ce (stm32f411ce), only one ADC channel on pin PA1 is currently supported. 

Zephyr does support multiple ADC channels on other boards, for example, with the Particle Xenon (nRF52840), I’m able to access all 6 ADC channels without any issues.

As the STM32 devices are very popular and used extensively, it would be very useful to have full channel support.

**Describe the solution you'd like**
Extend ADC channel support on the STM32 (blackpill_f411ce and other) from the current one or two channels  to all the available ADC channels.

**Describe alternatives you've considered**
The current temporary fallback has been to use the Arduino Framework with the STM32 blackpill_f411ce to access the available ADC channels. I certainly would prefer to use Zephyr.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

alfonsotames, 2021-02-17:
I was just going to report a bug and stumble with this post. I have been testing with an STM32L432KC board different PWM output pins and found that only those that use Channel 1 are working. Is this the available implementation or is it a bug? 

The L4 dts declares these pins but only the ones in Channel 1 seem to be working:

`			/* TIM_CH / TIM_CHN */

			tim1_ch1n_pa7: tim1_ch1n_pa7 {
				pinmux = <STM32_PINMUX('A', 7, AF1)>;
			};

			tim1_ch1_pa8: tim1_ch1_pa8 {
				pinmux = <STM32_PINMUX('A', 8, AF1)>;
			};

			tim1_ch2_pa9: tim1_ch2_pa9 {
				pinmux = <STM32_PINMUX('A', 9, AF1)>;
			};

			tim1_ch3_pa10: tim1_ch3_pa10 {
				pinmux = <STM32_PINMUX('A', 10, AF1)>;
			};

			tim1_ch4_pa11: tim1_ch4_pa11 {
				pinmux = <STM32_PINMUX('A', 11, AF1)>;
			};

			tim1_ch2n_pb0: tim1_ch2n_pb0 {
				pinmux = <STM32_PINMUX('B', 0, AF1)>;
			};

			tim1_ch3n_pb1: tim1_ch3n_pb1 {
				pinmux = <STM32_PINMUX('B', 1, AF1)>;
			};

			tim2_ch1_pa0: tim2_ch1_pa0 {
				pinmux = <STM32_PINMUX('A', 0, AF1)>;
			};

			tim2_ch2_pa1: tim2_ch2_pa1 {
				pinmux = <STM32_PINMUX('A', 1, AF1)>;
			};

			tim2_ch3_pa2: tim2_ch3_pa2 {
				pinmux = <STM32_PINMUX('A', 2, AF1)>;
			};

			tim2_ch4_pa3: tim2_ch4_pa3 {
				pinmux = <STM32_PINMUX('A', 3, AF1)>;
			};

			tim2_ch1_pa5: tim2_ch1_pa5 {
				pinmux = <STM32_PINMUX('A', 5, AF1)>;
			};

			tim2_ch1_pa15: tim2_ch1_pa15 {
				pinmux = <STM32_PINMUX('A', 15, AF1)>;
			};

			tim2_ch2_pb3: tim2_ch2_pb3 {
				pinmux = <STM32_PINMUX('B', 3, AF1)>;
			};

			tim15_ch1n_pa1: tim15_ch1n_pa1 {
				pinmux = <STM32_PINMUX('A', 1, AF14)>;
			};

			tim15_ch1_pa2: tim15_ch1_pa2 {
				pinmux = <STM32_PINMUX('A', 2, AF14)>;
			};

			tim15_ch2_pa3: tim15_ch2_pa3 {
				pinmux = <STM32_PINMUX('A', 3, AF14)>;
			};

			tim16_ch1_pa6: tim16_ch1_pa6 {
				pinmux = <STM32_PINMUX('A', 6, AF14)>;
			};

			tim16_ch1n_pb6: tim16_ch1n_pb6 {
				pinmux = <STM32_PINMUX('B', 6, AF14)>;
			};
`

key1066, 2021-02-18:
From my debugging on the blackpill_f411ce (stm32f411ce), the STM32 code states that PWM works only for 1 or 2 channels.
Sounds like a work in progress. It's surprising since PWM would be used often by most folks and the STM32 is a very popular MCU.

Under the Arduino Framework, all ADC channels are accessible on the blackpill_f411ce, and also there is no channel usage limitation on nRF52840 under Zephyr.

StefJar, 2021-04-21:
hey guys
I needed to get some work around done
this is a work around for the internal channels
https://github.com/zephyrproject-rtos/zephyr/pull/34458

problem with the current stm32 adc drv:
- internal channels are overlapping with the external ones
- example: channel 17 at stm32f412  = external or VRef

Idea:
- there is no adc channel muxer yet but we might need some
- we don't like to rewrite the whole driver
- we change the dts binding, so the user can mux the channel
- the driver takes the #define generated by the dts compiler and do the channel muxing

example:
```
soc {
	pinctrl: pin-controller@40020000 {			
		// adc
		// PC1 ADC_VBAT -> ADC1_IN11
		// PC3 ADC_ICHARG -> ADC1_IN13
		adc_VBat: adc_VBat {
			pinmux = <STM32_PINMUX('C', 1, ANALOG)>;
		};
		adc_ICharg: adc_ICharg {
			pinmux = <STM32_PINMUX('C', 3, ANALOG)>;
		};
	};
};
zephyr,user {
	// PC1 ADC_VBAT -> ADC1_IN11
	// PC3 ADC_ICHARG -> ADC1_IN13
	// VREF -> adc channel 17
	io-channels = <&adc1 11>, <&adc1 13>, <&adc1 17>;
};

// adc
&adc1 {
	pinctrl-0 = <&adc_VBat &adc_ICharg>;
	ch17-internal;
	status = "okay";
};
```
I am sure that this principle can be easily extend over different stm32 MCUs.

mcscholtz, 2021-08-10:
This is critical for my project so I added the ability to use a shared IRQ on the stm32f4x, so that all 3 channels can be used.

Please see the PR here: https://github.com/zephyrproject-rtos/zephyr/pull/37585

This is my first contribution so feedback would be great. Thanks!

erwango, 2022-06-29:
> ere any progress of using multiple channels with adc_sequence struct. I am trying to use two channels but get -1 in the output result. I am using stm32l475 MCU with disco_l475 board.

This was closed by https://github.com/zephyrproject-rtos/zephyr/pull/41497

erwango, 2022-12-14:
Closing in favor of https://github.com/zephyrproject-rtos/zephyr/issues/39640.
For people interested in ADC+DMA:
https://github.com/zephyrproject-rtos/zephyr/pull/52965
Don't hesitate to test and provide feedback in directly in PR

closed 2022-12-14 by erwango