zephyrproject-rtos/zephyr #32289: USDHC: Fails after reset

rtalbott, 2021-02-14:
**Describe the bug**
The sd card sample fails after resetting board.

**To Reproduce**
Steps to reproduce the behavior:
1. build samples/subsys/fs/fat_fs  for imxrt1060_evk
2. west flash
3. *works ok*
4. press reset
5. fails with <err> usdhc: CMD17 Polling ERROR

**Expected behavior**
work the same after flash pressing reset

**Impact**
showstopper for use of the usdhc

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr v2.5.0-rc4
 - Commit SHA or Version used: Zephyr OS build v2.5.0-rc4-2-g769d91b922b7

**Logs and console output**
_west flash_
*** Booting Zephyr OS build v2.5.0-rc4-2-g769d91b922b7  ***
Sector size 512
Memory Size(MB) 15271
Disk mounted.

Listing dir /SD: ...
[DIR ] SPOTLI~1
[DIR ] FSEVEN~1
[DIR ] SYSTEM~1
[FILE] TEXTTX~1.TXT (size = 26)
[00:00:00.105,000] <inf> usdhc: SD inserted!

[00:00:00.112,000] <inf> main: Block count 31275008

_pressed reset_
*** Booting Zephyr OS build v2.5.0-rc4-2-g769d91b922b7  ***
Error mounting disk.
[00:00:00.105,000] <inf> usdhc: SD inserted!

[00:00:00.364,000] <err> usdhc: CMD17 Polling ERROR

[00:00:00.364,000] <err> usdhc: USDHC bus device initalization failed!

[00:00:00.364,000] <err> main: Storage init ERROR!
[00:00:00.365,000] <err> fs: fs mount error (-5)

_west flash_
*** Booting Zephyr OS build v2.5.0-rc4-2-g769d91b922b7  ***
Sector size 512
Memory Size(MB) 15271
Disk mounted.

Listing dir /SD: ...
[DIR ] SPOTLI~1
[DIR ] FSEVEN~1
[DIR ] SYSTEM~1
[FILE] TEXTTX~1.TXT (size = 26)
[00:00:00.105,000] <inf> usdhc: SD inserted!

[00:00:00.112,000] <inf> main: Block count 31275008

_pressed reset_
*** Booting Zephyr OS build v2.5.0-rc4-2-g769d91b922b7  ***
Error mounting disk.
[00:00:00.105,000] <inf> usdhc: SD inserted!

[00:00:00.364,000] <err> usdhc: CMD17 Polling ERROR

[00:00:00.364,000] <err> usdhc: USDHC bus device initalization failed!

[00:00:00.364,000] <err> main: Storage init ERROR!
[00:00:00.365,000] <err> fs: fs mount error (-5)

MaureenHelm, 2021-04-02:
I haven't been able to reproduce this issue. Have you tried a different SD card?

@lgl88911 have you seen this behavior?

lgl88911, 2021-04-03:
@MaureenHelm It may be that the hardware design does not support the high speed mode switching for UHS Card. This problem can also be solved from the software，please refer to https://github.com/zephyrproject-rtos/zephyr/issues/32289 for modification to see if it is valid.

lgl88911, 2021-04-11:
Since the usdhc code was moved to the driver, https://github.com/zephyrproject-rtos/zephyr/pull/29096 was abolished and create https://github.com/zephyrproject-rtos/zephyr/pull/34205

MaureenHelm, 2021-05-13:
Fixed by #34205. Please reopen if you still see the issue after pulling in this change.

nxf58843, 2021-08-17:
I still get this issue with some micro SD cards.
Kingston 16GB HC I fails after reset or power on, but works immediately after the firmware is loaded.
SanDisk Ultra 16GB HC I works after reset or power on. 
This is with latest: Zephyr OS build zephyr-v2.6.0-2021-ge78cb747bd85

lgl88911, 2021-09-01:
@nxf58843 Whether `no-1-8-v` has been added to the devicetree of usdhc, like this:
```
&usdhc1 {
	status = "okay";
	cd-gpios = <&gpio2 28 GPIO_ACTIVE_LOW>;
	no-1-8-v;
};
```

danieldegrasse, 2021-10-05:
Linked pull request adds `no-1-8-v` to the devicetree of all rt10xx evaluation boards that support SDMMC, since all of them encounter reliability issues when communicating with SD cards at 1.8V