zephyrproject-rtos/zephyr #32291: Zephyr don't build sample hello world for Particle Xenon

DuMaM, 2021-02-14:
**Describe the bug**
I'm unable to compile sample/hello_world for particle_xenon

**To Reproduce**
git checkout 769d91b92
west build -b particle_xenon samples/hello_world

**Expected behavior**
I expect to have output file with sample code.

**Impact**
I'm unable to build any app. This fails build also for mcuBoot. 

**Logs and console output**
```
 ~/A/z/z/zephyr  ➦ 769d91b92   west build -b particle_xenon samples/hello_world
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/dumam/Apps/zmk/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/dumam/Apps/zmk/zmk/zephyr/samples/hello_world
-- Zephyr version: 2.5.0-rc4 (/home/dumam/Apps/zmk/zmk/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.6") found components: Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: particle_xenon
-- Cache files will be written to: /home/dumam/.cache/zephyr
-- Using toolchain: zephyr 0.12.2 (/home/dumam/zephyr-sdk-0.12.2)
-- Found dtc: /home/dumam/zephyr-sdk-0.12.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/dumam/Apps/zmk/zmk/zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Generated zephyr.dts: /home/dumam/Apps/zmk/zmk/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/dumam/Apps/zmk/zmk/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/dumam/Apps/zmk/zmk/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/dumam/Apps/zmk/zmk/zephyr/Kconfig
Loaded configuration '/home/dumam/Apps/zmk/zmk/zephyr/boards/arm/particle_xenon/particle_xenon_defconfig'
Merged configuration '/home/dumam/Apps/zmk/zmk/zephyr/samples/hello_world/prj.conf'

warning: HAS_NORDIC_DRIVERS (defined at modules/hal_nordic/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF52X (defined at soc/arm/nordic_nrf/nrf52/Kconfig.series:6), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)

warning: HAS_NRFX (defined at modules/hal_nordic/nrfx/Kconfig:4) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF52X (defined at soc/arm/nordic_nrf/nrf52/Kconfig.series:6), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)

warning: NRFX_CLOCK (defined at modules/hal_nordic/nrfx/Kconfig:14) has direct dependencies HAS_HW_NRF_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF (defined at drivers/clock_control/Kconfig.nrf:13), with value y, direct dependencies SOC_COMPATIBLE_NRF && CLOCK_CONTROL (value: y), and select condition !CLOCK_CONTROL_NRF_FORCE_ALT && SOC_COMPATIBLE_NRF && CLOCK_CONTROL (value: y)

warning: NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED (defined at modules/hal_nordic/nrfx/Kconfig:18) has direct dependencies NRFX_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF_K32SRC_XTAL (defined at drivers/clock_control/Kconfig.nrf:36), with value y, direct dependencies <choice CLOCK_CONTROL_NRF_SOURCE> (value: y), and select condition !SOC_SERIES_BSIM_NRFXX && <choice CLOCK_CONTROL_NRF_SOURCE> (value: y)

warning: NRFX_GPIOTE (defined at modules/hal_nordic/nrfx/Kconfig:65) has direct dependencies HAS_HW_NRF_GPIOTE && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - GPIO_NRFX (defined at drivers/gpio/Kconfig.nrfx:4), with value y, direct dependencies SOC_FAMILY_NRF && GPIO (value: y), and select condition SOC_FAMILY_NRF && GPIO (value: y)

warning: NRFX_PPI (defined at modules/hal_nordic/nrfx/Kconfig:101, drivers/serial/Kconfig.nrfx:346) has direct dependencies (HAS_HW_NRF_PPI && HAS_NRFX && 0) || (HAS_HW_NRF_PPI && (UART_0_NRF_HW_ASYNC || UART_1_NRF_HW_ASYNC || UART_2_NRF_HW_ASYNC || UART_3_NRF_HW_ASYNC) && UART_NRFX && SERIAL) with value n, but is currently being y-selected by the following symbols:
 - UART_0_ENHANCED_POLL_OUT (defined at drivers/serial/Kconfig.nrfx:39), with value y, direct dependencies UART_0_NRF_UARTE && (UART_0_NRF_UART || UART_0_NRF_UARTE) && UART_NRFX && SERIAL (value: y), and select condition HAS_HW_NRF_PPI && UART_0_NRF_UARTE && (UART_0_NRF_UART || UART_0_NRF_UARTE) && UART_NRFX && SERIAL (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /home/dumam/Apps/zmk/zmk/zephyr/cmake/kconfig.cmake:265 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/dumam/Apps/zmk/zmk/zephyr/cmake/app/boilerplate.cmake:536 (include)
  /home/dumam/Apps/zmk/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/dumam/Apps/zmk/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/dumam/Apps/zmk/zmk/zephyr/build -S/home/dumam/Apps/zmk/zmk/zephyr/samples/hello_world -GNinja -DBOARD=particle_xenon
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK zephyr-sdk-0.12.2
 - Commit SHA: 769d91b92, master,  since first release of 2.5.0

**Additional context**
-

carlescufi, 2021-02-14:
Have you run `west update`?

pabigot, 2021-02-15:
Yes, it must be west update; the reproducer works fine for me.  Reopen if that doesn't solve this.

closed 2021-02-15 by pabigot

DuMaM, 2021-02-15:
Thanks for help. `west update` solved the issue :)