zephyrproject-rtos/zephyr #32293: Zephyr 2.6 Release Checlist

nashif, 2021-02-14:
- [x] Major enhancements:
   - See https://github.com/zephyrproject-rtos/zephyr/projects/9

- [x] Pre RC1 Steps
   - [x] Create v2.7.0 milestone so that features that don't make it to the release can be assigned a milestone
   - [x] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [x] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (Security working group).

See https://docs.zephyrproject.org/latest/development_process/release_process.html#tagging for tagging details

--------
- [x] Release Notes WIP
  - [x] Make sure to tell everybody to use the **past tense** (i.e. "X has been added" or "X is now Y")
  - [x] Create draft `doc/releases/release-notes-2.5.rst` with feature summary from https://www.zephyrproject.org/developers/#releases-overview #22923
  - [x] Update `doc/releases/release-notes-2.5.rst` detailed sections (relative to 2.4.0 release) as important issues are addressed (subsystem owners). See #25869
--------
- [x] Finalize Release Notes
  - [x] Add list of GitHub issues (spell check issue list and fix in GitHub as needed) using `scripts/release/list_issues.py`
  - [x] Review `doc/releases/release-notes-2.5.rst` overview summary and details sections (see Sign-off below)
    - [x] Request summary details from code owners
- [x] Update `doc/reference/overview.rst`
- [x] Update `doc/conf.py` (add version to pick list)
- [x] Update `doc/LICENSING.rst` for new components that do not have a license
--------
- [x]  Final Release Notes Sign-off
  - Subsystems
    - [x] Kernel - @andyross, @andrewboie
    - [x] Build and infrastructure - @tejlmand @nashif @galak  
    - [x] Devicetree - @galak, @mbolivar-nordic 
    - [x] Drivers and sensors - @MaureenHelm, @tbursztyka, @pabigot, @mnkp  
    - [x] Filesystems - @nvlsianpu, @pabigot  
    - [x] Networking - @jukkar 
    - [x] Bluetooth - @jhedberg , @joerchan, @carlescufi 
    - [x] USB - @jfischer-phytec-iot @emob-nordic 
    - [x] CAN - @alexanderwachter  
  - Architectures
    - [x] ARC - @ruuddw 
    - [x] ARM - @ioannisg , @galak, @MaureenHelm 
    - [x] X86 - @nashif, @andrewboie 
    - [x] RISCV - @pgielda @nategraff-sifive
    - [x] Xtensa, NIOS-II - @nashif, @andrewboie 
  - Board an Soc support - many
  - [x] Security - @d3zd3z 
--------
- [x] Update version in VERSION
- [x] Tag both v2.5.0 and zephyr-v2.5.0
- [x] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases. Make sure you use the `zephyr-v2.5.0` tag so that the `.zip` that GitHub offers has a reasonable name. Add the release notes to the GitHub release.
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [x] Update doc generation (in future should be automatic based on tag) ping @nashif
  - [ ] Update https://www.zephyrproject.org/developer-resources/#current-release (email Brett)
  - [x] Create `v2.5-branch` branch for 2.5.x releases (from v2.5.0 tag, don't include the switch to patchlevel 99)
  - [x] Create `backport v2.5-branch` label
  - [ ] Merge all post-release PRs in https://github.com/zephyrproject-rtos/zephyr/projects/27
  - [ ] Move all PRs/Issues in the v2.5.0 milestone to the next milestone
  - [ ] Close v2.5.0 milestone 
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Update https://www.zephyrproject.org with latest stats from GitHub (email Brett)
  - [ ] Update current release dates in https://github.com/zephyrproject-rtos/zephyr/wiki/Program-Management, move current release column to "past releases table", and add next release timeline if missing (v2.7.0)
  - [ ] Store the number of currently open PRs to keep track of post-release PR overhead over time

galak, 2021-06-04:
https://github.com/zephyrproject-rtos/zephyr/issues/35966

galak, 2021-06-04:
Bug status:
High: 0
Medium: 10
Low:

galak, 2021-06-04:
Git Status:
Num commits: 52908 (`git rev-list --count upstream/main`)
Contributors 976  (From main Zephyr Github Page, under about)
Repositories: 94 (on https://github.com/zephyrproject-rtos)
Num pull requests: 35968 (Look for latest pr number on https://github.com/zephyrproject-rtos/zephyr/pulls)

galak, 2021-06-04:
Closing this out.  2.6 is out the door.  Just need to work on the blog.

closed 2021-06-04 by galak