zephyrproject-rtos/zephyr #32314: 2021 GSoC Project Ideas - Enable Interactive Zephyr Test Suite

mengxianglinx, 2021-02-15:
Description:Zephyr project has a well-built test framework Ztest and a full-fledged test runner Twister for module testing and integration testing. There are thousands of test cases in zephyr project, contributed by community developers. Here is a typical way of executing test case: test case is detected by YAML configuration files and then Twister builds test for the platforms defined in the hardware map and subsequently flashes and runs the tests on the platform. However, this way is not providing users a straightforward view about executed test cases and subsequently flashing/running the test cases is less productive. This project is aiming to enable interactive Zephyr test suite(a collection of multiple test cases) based on Ztest and Twister which allows visualized test case config and test case execution through command line or shell script.     

Difficulty: Medium
Skill Required: Python/C
Hardware requirement: Computer with QEMU installed
Possible mentor: Meng, Xianglin (xianglinx.meng@intel.com]) Li Jocelyn(jocelyn.li@intel.com)

mengxianglinx, 2021-02-16:
fix #32000

punitvara, 2021-02-18:
I can help to co-mentor this project.

closed 2021-05-05 by github-actions[bot]