zephyrproject-rtos/zephyr #32321: /tmp/ccTlcyeD.s:242: Error: Unrecognized Symbol type  " "

saad-abdullah62, 2021-02-15:
Hi, I am trying to add support for zedboard and zynq7000 SOC. While compiling sample/hello_world for Cortex A processors using ARM Embedded GNU i get **assembler message Error:Unrecognized Symbol type " "** . I tried to used aarch64 toolchain in zephyr sdk and it works for cortex-a53 and cortex-a72. But since i need to use cortex-a9 which is not included for -mtune option in aarch64 so i am using ARM Embedded GNU to built application. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add ARM EMBEDDED toolchain as cross-compiler 
2. cd sample/hello_world, mkdir build; cd build
2. cmake -DBOARD=qemu_cortex_a53
3. make
4. See error

**Environment (please complete the following information):**
 - OS: ( Linux(Ubuntu 18.04))
 - Toolchain (ARM Embedded)
 
[image]

carlescufi, 2021-02-15:
This might be a crlf issue. What is your `core.autocrlf ` in `.gitconfig`?

saad-abdullah62, 2021-02-15:
@carlescufi I can't find any .gitconfig in zephyr directory. Can you mention its location? I tried using "grep -r" linux command also but no results.

saad-abdullah62, 2021-02-15:
@carlescufi  I can't find any .gitconfig. Can you mention its location? Google told me it should be in /etc. But there is no such file residing there.

saad-abdullah62, 2021-02-15:
I used git config --global core.autocrlf to set core.autrocrlf to input. But still getting same error

saad-abdullah62, 2021-02-21:
Another thing it builds application for Cortex M processors correctly.

carlescufi, 2021-02-23:
Try setting it to auto:

```
git config --global core.autocrlf true
```

and then run `west init` and west update` from scratch.

saad-abdullah62, 2021-02-24:
I tried above solution but there is no effect. I am getting same results as before.

galak, 2021-03-16:
As you are looking to support a 32-bit Cortex-A core you need to configure the compiler correctly for this.  @carlocaione or @npitre can probably comment on what needs to be done.

This is not a bug as we don't support any current 32-bit Cortex-A core or SoC and thus it would be new functionality.

closed 2021-03-16 by galak

npitre, 2021-03-17:
The Cortex-A9 is much more similar to a Cortex-M than Cortex-A53 or any
Aarch64 variations. Adjustments to the aarch32 code would also be needed
as this is currently unsupported.

The compiler to use from the Zephyr SDK would be arm-zephyr-eabi with
appropriate flags. The aarch64-zephyr-elf version won't work.

saad-abdullah62, 2021-03-24:
The error was because i kept my ARM v7 A configuration options in aarch 64 inside core. I added them to Kconfig file given for Cortex R processors and i was able to compile. But now i have other errors. 
I have to add support for Cortex A9 but i don't have much knowledge about zephyr. I went through documentation and understood board and SOC porting guides. But i can't find any guide to port zephyr on new CPU core. I see many assembly and header files for every core e.g. for Cortex M, for Cortex R and for ARM 64 but don't know where they come from or which one should i add fro Cortex A9.  So Any Suggestions or hints about where to start will be appreciated.