zephyrproject-rtos/zephyr #32331: Use of unitialized variables in can_set_bitrate()

talih0, 2021-02-15:
**Describe the bug**
The function can_set_bitrate() doesn't set timing.sjw and timing_data.sjw. 
The variables are uninitialized and parsed to can_set_timing(), which uses the uninitialized values to set sjw parameters.
Functions that calls can_set_bitrate() will also have this bug. (i.e. can_configure()).

The function was added in pull request #28345.

**To Reproduce**
west build -p auto -b ... samples/subsys/canbus/canopen
(I'm working on a board that hasn't yet been upstreamed)

alexanderwachter, 2021-02-20:
@henrikbrixandersen @karstenkoenig @martinjaeger
Any ideas how ti set a proper sjw. We could pass 0 and define in the API that zero does not change the value.

henrikbrixandersen, 2021-03-23:
> Any ideas how ti set a proper sjw. We could pass 0 and define in the API that zero does not change the value.

That sounds reasonable.

PR #34553 (2021-06-01): drivers: can: sjw == 0 in can_set_timing should not change sjw
If the supplied sjw in the timing parameters is zero,
the sjw parameter should not be changed.
This fixes the uninitialized swj in can_set_bitrate.

Fix: #32331
closed 2021-06-01 by galak