zephyrproject-rtos/zephyr #32343: openthread manual joiner hangs

krbvroc1, 2021-02-15:
```CONFIG_OPENTHREAD_MANUAL_START=y```

I am programmatically trying to call otJoinerStart(). The otJoinerStart() callback is never executed and looking at wireshark the expected 'discover' beacons are never sent.

Here is code that reproduces the issue:

```

#include <zephyr.h>
#include <logging/log.h>

#include <net/openthread.h>

#include <openthread/thread.h>
#include <openthread/server.h>
#include <openthread/sntp.h>
#include <openthread/joiner.h>

LOG_MODULE_REGISTER(app_sample);

struct k_poll_signal app_join_trigger_signal;

static struct k_poll_event join_trigger_events[1] = {
  K_POLL_EVENT_INITIALIZER(K_POLL_TYPE_SIGNAL,
                           K_POLL_MODE_NOTIFY_ONLY,
                           &app_join_trigger_signal),
};

void app_join_attempt_timer_callback(struct k_timer *dummy)
{
  k_poll_signal_raise(&app_join_trigger_signal, 1);
}

K_TIMER_DEFINE(app_join_attempt_timer, app_join_attempt_timer_callback, NULL);

void app_joiner_callback(otError join_err_code, void *passback)
{
  bool success = false;

  switch (join_err_code)
  {
    case OT_ERROR_NONE:
      LOG_INF("JOIN Success\r\n");
      success = true;
      break;
    case OT_ERROR_SECURITY:
      LOG_INF("JOIN Error - Security\r\n");
      break;
    case OT_ERROR_NOT_FOUND:
      LOG_INF("JOIN Error - Not found\r\n");
      break;
    default:
    case OT_ERROR_RESPONSE_TIMEOUT:
      LOG_INF("JOIN Error - Response timeout\r\n");
      break;
  }

  k_poll_signal_raise(&app_join_trigger_signal, success);

}

void main(void)
{
  otError error;
  
  LOG_INF("Waiting 30secs...");
  
  k_poll_signal_init(&app_join_trigger_signal);

  k_timer_start(&app_join_attempt_timer, K_SECONDS(30), K_NO_WAIT);

  k_poll(join_trigger_events, 1, K_FOREVER);
  join_trigger_events[0].signal->signaled = 0;
  join_trigger_events[0].state = K_POLL_STATE_NOT_READY;

  LOG_INF("Calling otJoinerStart\r\n");

  struct otInstance *ot_instance = openthread_get_default_instance();

  error = otIp6SetEnabled(ot_instance, true);
  __ASSERT_NO_MSG(error == OT_ERROR_NONE);
  
  // Must reset to 0xFFFF or it only tries to join default panid
  error = otLinkSetPanId(ot_instance, 0xFFFF);
  __ASSERT_NO_MSG(error == OT_ERROR_NONE);

  error = otJoinerStart(ot_instance, "TEST1234", NULL, "VENDOR", "MODEL", "SWVER", NULL, app_joiner_callback, NULL);
  if (error != OT_ERROR_NONE)
  {
    LOG_INF("otJoinerStart error\r\n");
  }
  else
  {
    // Wait for otJoinerStart callback to be invoked (***never happens***)
    k_poll(join_trigger_events, 1, K_FOREVER);
    join_trigger_events[0].signal->signaled = 0;
    join_trigger_events[0].state = K_POLL_STATE_NOT_READY;

  }
}

```
Enabling debug in openthread only shows the following... 

NOTE: The 'Failed to send IPv6' might be a clue? Not sure.

```
[00:00:30.636,505] <inf> app_sample: Calling otJoinerStart

[00:00:30.636,688] <dbg> net_openthread.-MAC-----: RadioState: Sleep -> Receive
[00:00:30.636,688] <dbg> net_openthread.-MAC-----: Idle mode: Radio receiving on channel 11
[00:00:30.636,779] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00001000) [Ip6Mult+] 
[00:00:30.636,810] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 0
[00:00:30.637,176] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00000001) [Ip6+] 
[00:00:30.637,207] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 0
[00:00:30.637,420] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00000008) [LLAddr] 
[00:00:30.637,420] <inf> net_l2_openthread: State changed! Flags: 0x00000008 Current role: 0
[00:00:30.637,573] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x01000000) [NetifState] 
[00:00:30.637,603] <inf> net_l2_openthread: State changed! Flags: 0x01000000 Current role: 0
[00:00:30.637,695] <inf> net_openthread: -MESH-CP-: Joiner starting
[00:00:30.638,061] <dbg> net_openthread.-MAC-----: RadioExtAddress: 56bbc40387a1d9fc
[00:00:30.638,183] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00000002) [Ip6-] 
[00:00:30.638,183] <inf> net_l2_openthread: State changed! Flags: 0x00000002 Current role: 0
[00:00:30.638,397] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00000001) [Ip6+] 
[00:00:30.638,397] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 0
[00:00:30.638,610] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00000008) [LLAddr] 
[00:00:30.638,610] <inf> net_l2_openthread: State changed! Flags: 0x00000008 Current role: 0
[00:00:30.639,007] <dbg> net_openthread.-MAC-----: Request to start operation "TransmitDataDirect"
[00:00:30.639,007] <dbg> net_openthread.-MAC-----: Starting operation "TransmitDataDirect"
[00:00:30.639,343] <dbg> net_openthread.==============================[TX len=000]===~
[00:00:30.639,648] <dbg> net_openthread.---------------------------------------------~
[00:00:30.639,678] <dbg> net_openthread.-MAC-----: Finishing operation "TransmitDataDirect"
[00:00:30.639,801] <inf> net_openthread: -MAC-----: Failed to send IPv6 UDP msg, len:56, chksum:2c3b, to:None, sec:no, error:Abort, prio:net
[00:00:30.639,923] <inf> net_openthread: -MAC-----:     src:[fe80:0:0:0:54bb:c403:87a1:d9fc]:19788
[00:00:30.640,045] <inf> net_openthread: -MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
[00:00:30.640,106] <dbg> net_openthread.-MAC-----: Idle mode: Radio receiving on channel 11
[00:00:30.640,258] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x00008000) [PanId] 
[00:00:30.640,258] <inf> net_l2_openthread: State changed! Flags: 0x00008000 Current role: 0
[00:00:30.640,319] <dbg> net_openthread.-MAC-----: RadioPanId: 0x3942
[00:00:30.640,441] <inf> net_openthread: -MLE-----: Send Discovery Request (ff02:0:0:0:0:0:0:2)
[00:00:30.640,533] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x08000000) [JoinerState] 
[00:00:30.640,563] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:00:30.640,594] <inf> net_openthread: -MESH-CP-: JoinerState: Idle -> Discover

```
In the reproduction code, if I simply replace the
error = otJoinerStart(...) with error = OT_ERROR_NONE;
and then manually invoke the 'ot joiner start TEST1234' from the shell/cli, it works fine, I see proper wireshark discover packets. The shell invoked command show a very different debug log sequence...

```
uart:~$ ot joiner start TEST1234
Done
[00:00:36.899,780] <inf> net_openthread: -MESH-CP-: Joiner starting
[00:00:36.900,177] <dbg> net_openthread.-MAC-----: RadioExtAddress: a671cbf48620e21b
[00:00:36.900,451] <dbg> net_openthread.-MAC-----: RadioPanId: 0xc64b
[00:00:36.900,604] <inf> net_openthread: -MLE-----: Send Discovery Request (ff02:0:0:0:0:0:0:2)
[00:00:36.900,604] <inf> net_openthread: -MESH-CP-: JoinerState: Idle -> Discover
[00:00:36.900,909] <inf> net_openthread: -CORE----: Notifier: StateChanged (0x0800800b) [Ip6+ Ip6- LLAddr PanId JoinerState] 
[00:00:36.900,939] <inf> net_l2_openthread: State changed! Flags: 0x0800800b Current role: 0
[00:00:36.901,092] <dbg> net_openthread.-MAC-----: Request to start operation "TransmitDataDirect"
[00:00:36.901,123] <dbg> net_openthread.-MAC-----: Starting operation "TransmitDataDirect"
[00:00:36.901,214] <dbg> net_openthread.-MAC-----: RadioState: Receive -> CsmaBackoff
[00:00:36.901,245] <dbg> net_openthread.-MAC-----: RadioState: CsmaBackoff -> Transmit
[00:00:36.905,395] <dbg> net_openthread.-MAC-----: RadioState: Transmit -> Receive
[00:00:36.905,761] <dbg> net_openthread.==============================[TX len=037]===~
[00:00:36.906,280] <dbg> net_openthread.| 01 D8 49 FF FF FF FF 4B | C6 1B E2 20 86 F4~
[00:00:36.906,738] <dbg> net_openthread.| A6 7F 3B 02 F0 4D 4C 4D | 4C CD F5 FF 10 1A~
[00:00:36.907,012] <dbg> net_openthread.| 02 28 00 20 D0 .. .. .. | .. .. .. .. .. ..~
[00:00:36.907,348] <dbg> net_openthread.---------------------------------------------~
[00:00:36.907,348] <dbg> net_openthread.-MAC-----: Finishing operation "TransmitDataDirect"
[00:00:36.907,501] <inf> net_openthread: -MAC-----: Sent IPv6 UDP msg, len:56, chksum:cdf5, to:0xffff, sec:no, prio:net
[00:00:36.907,623] <inf> net_openthread: -MAC-----:     src:[fe80:0:0:0:a471:cbf4:8620:e21b]:19788
[00:00:36.907,745] <inf> net_openthread: -MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
[00:00:36.907,806] <dbg> net_openthread.-MAC-----: Idle mode: Radio receiving on channel 11
[00:00:37.207,855] <dbg> net_openthread.-MAC-----: Request to start operation "TransmitDataDirect"
[00:00:37.207,885] <dbg> net_openthread.-MAC-----: Starting operation "TransmitDataDirect"
[00:00:37.207,977] <dbg> net_openthread.-MAC-----: RadioState: Receive -> CsmaBackoff
[00:00:37.208,007] <dbg> net_openthread.-MAC-----: RadioState: CsmaBackoff -> Transmit
[00:00:37.211,425] <dbg> net_openthread.-MAC-----: RadioState: Transmit -> Receive

```
I started with Nordic Connect SDK version 1.4.2 of Zephyr and tried various versions up to the 1.4.99 / master.
I am using nrf52840dk_nrf52840

rlubos, 2021-02-16:
Hi @krbvroc1,

OpenThread API is not thread safe, therefore when calling OT functions from other thread (like main) you need to protect the calls with a mutex, there's a dedicated function for that. Just add:
```
  struct openthread_context *ctx = openthread_get_default_context();

  openthread_api_mutex_lock(ctx);
```
before you call the OT functions, and then 
```
openthread_api_mutex_unlock(ctx);
```
after.

In your case, `otJoinerStart()` likely cause the scheduler to run and since OT thread has higher priority than main, it could've process the OT engine in the middle of `otJoinerStart()` call. I've observed similar behaviour with commissioner. I've reproduced your issue locally and adding a proper mutex protection fixes it.

krbvroc1, 2021-02-16:
@rlubos Thank you very much... I missed that detail! 

Is this for the entire API (every single call)? For example,  otCoapNewMessage() and all the individual functions such as appending URI, etc? Or if you retrieve a request from otCoapNewMessage() is it safe to build the message without locking?

rlubos, 2021-02-17:
There likely are APIs that are safe to call w/o mutex protection, but each case would need to be considered individually, I think it's safer to assume that you need to lock the OT thread in general.

closed 2021-02-17 by krbvroc1