zephyrproject-rtos/zephyr #32344: sporadic failure in tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace

nashif, 2021-02-15:
**Describe the bug**

sporadic failure in tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace 

```
[0mSeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.5.0-14-g1440279fe060  ***
Running test suite userspace
===================================================================
START - test_is_usermode
 PASS - test_is_usermode
===================================================================
START - test_write_control
E: General protection fault (code 0x0)
E: RAX: 0x00000000ffffffff RBX: 0x0000000000118028 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x000000000014efa0 RSP: 0x000000000014efa0
E:  R8: 0x0000000000105829  R9: 0x800000000014ea63 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014efa0 RFLAGS: 0x0000000000000246 CS: 0x003b CR3: 0x000000000015a000
E: RIP: 0x00000000001004aa
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x11c7c0 (ztest_thread)
Caught system error -- reason 0
System error was expected
 PASS - test_write_control
===================================================================
START - test_disable_mmu_mpu
E: General protection fault (code 0x0)
E: RAX: 0x00000000000c4023 RBX: 0x0000000000118050 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x000000000014efa0 RSP: 0x000000000014efa0
E:  R8: 0x0000000000105829  R9: 0x800000000014ea63 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014efa0 RFLAGS: 0x0000000000000246 CS: 0x003b CR3: 0x000000000015a000
E: RIP: 0x00000000001004db
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x11c7c0 (unknown)
Caught system error -- reason 0
System error was expected
 PASS - test_disable_mmu_mpu
===================================================================
START - test_read_kernram
E: Page fault at address 0x11c870 (error code 0x5)
E: Access violation: user thread not allowed to read
E: PTE: 0x11c000 -> 0x000000000011c000: RW A D XD 
E: RAX: 0x000000000011c7c0 RBX: 0x0000000000118078 RCX: 0x0000000000100151 RDX: 0x0000000000000000
E: RSI: 0x0000000000000000 RDI: 0x000000000011501c RBP: 0x000000000014efa0 RSP: 0x000000000014efa0
E:  R8: 0x0000000000105829  R9: 0x800000000014ea63 R10: 0x0000000000000000 R11: 0x0000000000000202
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014efa0 RFLAGS: 0x0000000000000202 CS: 0x003b CR3: 0x000000000015a000
E: RIP: 0x00000000001003ee
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x11c7c0 (ztest_thread)
Caught system error -- reason 0
System error was expected
 PASS - test_read_kernram
===================================================================
START - test_write_kernram
E: Page fault at address 0x11c870 (error code 0x7)
E: Access violation: user thread not allowed to write
E: PTE: 0x11c000 -> 0x000000000011c000: RW A D XD 
E: RAX: 0x000000000011c7c0 RBX: 0x00000000001180a0 RCX: 0x00000000000000f1 RDX: 0x0000000000114b7d
E: RSI: 0x0000000000114c09 RDI: 0x0000000000000000 RBP: 0x000000000014efa0 RSP: 0x000000000014efa0
E:  R8: 0x0000000000113e90  R9: 0x0000000000114c24 R10: 0x0000000000000000 R11: 0x0000000000000202
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014efa0 RFLAGS: 0x0000000000000246 CS: 0x003b CR3: 0x000000000015a000
E: RIP: 0x000000000010044e
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x11c7c0 (unknown)
Caught system error -- reason 0
System error was expected
[... 139 lines cut ...]
E: R12: 0x000000000011c7c0 R13: 0x0000000000000246 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014cd38 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x000000000015a000
E: RIP: 0x0000000000102d4f
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
E: Current thread: 0x11c7c0 (unknown)
Caught system error -- reason 4
Unexpected fault during test
```

**To Reproduce**
Look at CI results here for example:

https://buildkite.com/zephyr/zephyr/builds/20929#f7baab01-0129-467d-b945-1031f6dfe629

**Expected behavior**
test should pass

**Impact**
Block CI

**Logs and console output**
see above

**Environment (please complete the following information):**
- CI environment

pabigot, 2021-02-16:
This is causing failures in #32127 and can be reproduced with:

    twister -p qemu_x86_64_nokpti -s tests/kernel/mem_protect/userspace/kernel.memory_protection.userspace

git bisect shows this introduced at 4ff457113e90a3e51de5b9f792f4ae3608e92382  @andyross

andyross, 2021-02-16:
OK, the root cause seems to be a set of races in the abort implementation.  The simplest case to explain (though I'm not clear if this is exactly the failure we're seeing) is:

1. Thread A tries to abort a different thread B
2. Thread A enters `z_thread_single_abort()` and flags the thread ABORTING and continues, but...
3. Thread B (or anyone else) then aborts Thread A before it is complete (and before it has removed B from the run queue)
4.  Thread A is running with interrupts unmasked in multiple places in `z_thread_single_abort()`, so it gets interrupted by the IPI and is successfully aborted, never to run again.
5. Thread B is now a zombie.  Being already flagged ABORTING, any attempt to abort it will early exit as a noop
6. Thread B now tries to abort itself (i.e. by running to completion)
7. Thread B enters the self-abort path, switches into the idle thread, which tries to abort it, which doesn't work, and which then context switches back to the highest priority remaining thread, which is... Thread B.

The seemingly obvious way to fix this is by extending a local interrupt lock around all the work in step 2, but that is a **big** function and almost everything I try exposes a deadlock condition.  I'll need to think on this some.

andyross, 2021-02-16:
And FWIW: I don't see what this has to do with the SMP patch either.  But one effect of that fix was to make a bunch of operations in SMP much more synchronous with context switching on the other CPU, so I guess I'd believe we just stumbled on more reliably timed thread aborts.

andyross, 2021-02-17:
Woke up realizing that step 3/4 actually doesn't require a weird cross-abort, you can exercise it with a priority inversion too: all that's required is that thread A be preempted by any means (or on SMP, just get unlucky and hit the window from the other CPU) at the moment where it has flagged B "ABORTING" and then have B reach a self-abort on its own.

jharris-intel, 2021-02-18:
Conceptual question here.

You're in thread A. Thread A tries to abort thread B. While in the middle of the abort we context-switch to thread B. An interrupt happens while we're in thread B. The ISR tries to abort thread A.

What is allowed to happen?

In particular:

Are we allowed to defer the deletion from the ISR? What if the ISR was doing e.g. `delete_thread_a(); cleanup_thread_a_resources();`?

*****

A related question:

Thread A tries to abort thread C.
Thread B tries to abort thread C.

When does the lifespan of thread C end? (As in: what guarantee for application code do we have as to when they are allowed to clean up thread C's resources?) After the first abort call has returned? Or after the second? This is particularly relevant given Zephyr has the user manage thread resources for the most part.

jharris-intel, 2021-02-18:
One potential (rather heavy-handed) approach to this is roughly as follows:

1. Split out THREAD_ABORTING into multiple stages, one for each potential yield point.
2. Change thread abort to check the target thread's stage every yield point (so essentially state-machine driven.)
3. Have e.g. a pointer in the standard TCB indicating "I'm aborting thread X".
4. Any time you go to delete a thread (including resuming after a yield!), if its pointer indicates that it's aborting another thread, first follow the pointer and abort _that_ thread. (Note: you have to follow the pointer _first_ and _then_ state "I'm aborting thread X", as this prevents cycles. Otherwise you can get into infinite loops if you have cycles of threads all trying to delete each other.)

(Atomicity is rather required for some of these... which will likely be rather complex and subtle.)

(By 1 I mean, essentially, treat each spinlock-protected section inside k_thread_abort as its own stage.)

This essentially just breaks thread aborting into multiple atomic segments, and then handles an aborted thread's pending aborts for it before aborting the thread in question.

So, all told, with your latter example:

1. Thread A tries to abort thread B, gets preempted by thread B, thread B tries to self-abort. Some other thread C aborts thread A somewhere in here.

What actually happens would be something along the lines of the following:

1. Thread A calls `k_thread_abort` on thread B.
2. Thread A enters z_thread_single_abort().
3. Thread A checks thread B's aborting status, sees that it is not aborting anything and is not being aborted, and continues, flagging thread B as ABORTING.
4. Thread A hits a potential yield point, and setting thread A's "I'm aborting" pointer to thread B and flagging thread B as ABORTING_STAGE_A
5. Context switch to thread B.
6. Thread B tries to self-abort, telling the idle thread to abort thread B.
7. Thread C calls `k_thread_abort` on thread A.
8. Thread C enters z_thread_single_abort().
9. Thread C checks thread A's aborting status, sees that it is aborting thread B.
12. Thread C checks thread B's aborting status, sees that it is in ABORTING_STAGE_A and not aborting anything.
13. Thread C processes thread B's deletion to completion, clearing thread A's "I'm aborting" pointer.
14. Thread C checks thread A's aborting status, sees that it is not aborting anything.
15. Thread C processes thread A's deletion to completion.
14. Context switch to idle thread.
15. Idle thread picks up "please abort thread B".
16. Idle thread looks at thread B's status, sees it is fully aborted and not aborting anything, and drops the request.

...which works, mostly. There is a resource leak here in many cases (if e.g. thread A was going to do `abort_thread_b(); cleanup_thread_b_resources()`)

Unfortunately, I think you do have to go all the way from the root of the linked-list cleanup chain every time you enter a yield point, which means this is O(n^2) in the number of threads trying to abort a single thread.

And this adds a thread pointer / id to the generic TCB, which isn't ideal.

This has a _bunch_ of issues... but I think it's a valid approach. (I suggest modelling it in e.g. TLA+ or somesuch if you do go down this path.)

andyross, 2021-02-18:
See workaround in #32461 -- I needed to do that as part of the abort rework (it's a simple change to the lock ordering, really).  And when trying to isolate whether it had caused some new weirdness I was seeing, I tried it alone on top of upstream head and realized that it actually works to address the race in practice.  So we might want to merge that alone just to clean up the spurious failures that are blocking other work.  Please don't close this bug though.

andyross, 2021-02-18:
For those curious: the reason for extending the lock is to get the assignment to the CPU-specific current thread pointer, which was outside the lock and late, into the same lock as the selection of the new thread so they could be seen atomically from other CPUs and used for decisions (for example "Is this thread currently running on another CPU?")

andyross, 2021-02-18:
And @jharris-intel I think that's the natural way to fix it given our architecture, but I'm pretty sure we can do this without the extra non-atomic stages entirely, nor even the idle thread trick.  Still working out kinks (including ARM32's non-atomic PendSV z_swap() implementation coming back to bite me again...) but hope to have something up today.

andyross, 2021-02-18:
Oh, and a final note: I understand now the reason why the SMP deadlock fix made this so visible.  The action of that was to move the addition of the outgoing/interrupted _current thread to the run queue much later (to after the spin step that needs to wait for the inbound thread's context switch on its own CPU).   But that takes the original small hole (which again was pre-existing and not strictly in the swap code at all) and opens a whole new gulf: now another thread can come by and abort the outgoing thread in the unsynchronized space between the selection of the new thread and the new step where it adds it back (it's already aborted!) to the run queue, so it runs again unexpectedly.

andyross, 2021-02-26:
Heh, I realize I never put a Fixed # tag in any of the commits in the abort rewrite.  Close.

closed 2021-02-26 by andyross