zephyrproject-rtos/zephyr #32365: samples: hci_rpmsg build fail for nrf5340

AyturkDuzen, 2021-02-16:
**Describe the bug**
Compile failure during zephyr/samples/hci_rpmsg building for nrf5340, related to Kconfig warning NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED.

**To Reproduce**
Steps to reproduce the behavior:
1. clone zephyr and point to master
2. cd zephyr/samples/bluetooth/hci_rpmsg
3. west build -b nrf5340dk_nrf5340_cpunet
4. See error

**Expected behavior**
successful build.  ~2weeks ago, there was no problem.

**Impact**
Not able to progress on zephyr/tests/bluetooth/tester for nrf5340 since network core builds on hci_rpmsg sample.

**Logs and console output**
```
[n:hci_rpmsg] rm -rf build/
[n:hci_rpmsg] west build -b nrf5340dk_nrf5340_cpunet -p always
-- west build: generating a build system
Including boilerplate (Zephyr base): /work/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /work/ncs/zephyr/samples/bluetooth/hci_rpmsg
-- Zephyr version: 2.5.99 (/work/ncs/zephyr)
-- Found Python3: /usr/local/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: nrf5340dk_nrf5340_cpunet
-- Cache files will be written to: /home/.cache/zephyr
-- Using toolchain: zephyr 0.11.2 (/opt/zephyr-sdk)
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /work/ncs/zephyr/boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpunet.dts
-- Generated zephyr.dts: /work/ncs/zephyr/samples/bluetooth/hci_rpmsg/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /work/ncs/zephyr/samples/bluetooth/hci_rpmsg/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /work/ncs/zephyr/samples/bluetooth/hci_rpmsg/build/zephyr/include/generated/device_extern.h
Parsing /work/ncs/zephyr/Kconfig
Loaded configuration '/work/ncs/zephyr/boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpunet_defconfig'
Merged configuration '/work/ncs/zephyr/samples/bluetooth/hci_rpmsg/prj.conf'

warning: NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED (defined at modules/hal_nordic/nrfx/Kconfig:18, modules/hal_nordic/nrfx/Kconfig:18) has direct dependencies (NRFX_CLOCK && HAS_NRFX) || (NRFX_CLOCK && HAS_NRFX && 0) with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF_K32SRC_XTAL (defined at drivers/clock_control/Kconfig.nrf:36), with value y, direct dependencies <choice CLOCK_CONTROL_NRF_SOURCE> (value: y), and select condition !SOC_SERIES_BSIM_NRFXX && <choice CLOCK_CONTROL_NRF_SOURCE> (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /work/ncs/zephyr/cmake/kconfig.cmake:265 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /work/ncs/zephyr/cmake/app/boilerplate.cmake:536 (include)
  /work/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /work/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/local/bin/python3 -B/work/ncs/zephyr/samples/bluetooth/hci_rpmsg/build -S/work/ncs/zephyr/samples/bluetooth/hci_rpmsg -GNinja -DBOARD=nrf5340dk_nrf5340_cpunet
```

**Environment (please complete the following information):**
Linux
west version 0.9.0
Zephyr SDK 0.12.2
commit: dfa52a3 (master)

jfischer-no, 2021-02-16:
I can not confirm it. Please perform `west update` and recheck.

carlescufi, 2021-02-16:
Agreed, this is wrong, the sample builds fine. Probably `west update` required.

closed 2021-02-16 by carlescufi