zephyrproject-rtos/zephyr #32376: samples: driver: watchdog: Sample fails on disco_l475_iot1

hwilmers, 2021-02-16:
**Describe the bug**
The sample app samples/driver/watchdog fails at runtime, sending an error message.
Tested using Zephyr 2.5.0 with disco_l475_iot1.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/driver/watchdog
2. west build -b disco_l475_iot1
3. west flash
4. See error

**Expected behavior**
The expected behaviour of the app is to repeat following sequence:
- initialise watchdog
- feed the watchdog a few times
- stop feeding, and wait for a watchdog reset

**Impact**
The watchdog is not functional.

**Logs and console output**
Console log:

>    *** Booting Zephyr OS build zephyr-v2.5.0  ***
>   Watchdog sample application
>   Attempting to test pre-reset callback
>   Watchdog install error

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr 0.11.2
 - Zephyr Release 2.5.0

hwilmers, 2021-02-16:
Apparently, wdt_install_timeout() returns negative channel id here:
https://github.com/zephyrproject-rtos/zephyr/blob/b9fdab44747f732ffa6cb605f8d9595583430c3f/samples/drivers/watchdog/src/main.c#L112-L120

hwilmers, 2021-02-16:
At the attempt of setting timeout to 100ms, here the calculated timeout is set to 26214us:
https://github.com/zephyrproject-rtos/zephyr/blob/b275fec8c4b5a9ff4a3595a5659ad88625326150/drivers/watchdog/wdt_wwdg_stm32.c#L192
Is this the max timeout achievable with this MCU? Or is there a calculation error somewhere?

hwilmers, 2021-02-16:
When using the IWDG, it works with the timeout values used in the sample.
Enabling the WWDG is done in the overlay file boards/disco_l475_iot1.overlay, and a simple solution is to remove this file.
Alternatively, if WWDG shall be used, the value of WDT_MAX_WINDOW must be set to <= 26.

erwango, 2021-02-17:
@hwilmers Thanks for reporting. Actually the wwdg test should run with APB bu clock tunning: `CONFIG_CLOCK_STM32_APB1_PRESCALER=16`
Problem is that tunning is provided in sample.yaml file and then only applied when running twister.

I'll modify so this is available even when run with west.

PR #32387 (2021-02-19): samples/driver: watchdog: wwdg sample should run with west
A different configuration is applied when running this test
using west or using twister.
Both will use the disco_l475_iot1 overlay, but the extra config
for clock bus tunning is applied only when using twister and
samples is failed when run with west.
Move this extra config from sample.yaml to board .conf file so
it is applied in both cases.

Fixes #32376
commit 2179e068 (2021-02-19): samples/driver: watchdog: wwdg sample should run with west

A different configuration is applied when running this test
using west or using twister.
Both will use the disco_l475_iot1 overlay, but the extra config
for clock bus tunning is applied only when using twister and
samples is failed when run with west.
Move this extra config from sample.yaml to board .conf file so
it is applied in both cases.

Fixes #32376
commit 6cf2fdc5 (2021-02-24): samples/driver: watchdog: wwdg sample should run with west

A different configuration is applied when running this test
using west or using twister.
Both will use the disco_l475_iot1 overlay, but the extra config
for clock bus tunning is applied only when using twister and
samples is failed when run with west.
Move this extra config from sample.yaml to board .conf file so
it is applied in both cases.

Fixes #32376
closed 2021-02-19 by galak