zephyrproject-rtos/zephyr #32404: [Coverity CID :218734] Out-of-bounds access in subsys/bluetooth/host/smp.c

zephyrbot, 2021-02-17:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fe7c2efca800a0cf1bccf23aefe08b3c4beb88bf/subsys/bluetooth/host/smp.c#L3257

Category: Memory - corruptions
Function: `smp_pairing_rsp`
Component: Bluetooth
CID: [218734](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=218734)

Details:

```
3251         if (!sc_public_key) {
3252             atomic_set_bit(smp->flags, SMP_FLAG_PKEY_SEND);
3253             return 0;
3254         }
3255    
3256         atomic_set_bit(&smp->allowed_cmds, BT_SMP_CMD_PUBLIC_KEY);
>>>     CID 218734:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Passing "&smp->allowed_cmds" to function "atomic_clear_bit" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
3257         atomic_clear_bit(&smp->allowed_cmds, BT_SMP_CMD_SECURITY_REQUEST);
3258    
3259         return sc_send_public_key(smp);
3260     }
3261     #else
3262     static uint8_t smp_pairing_rsp(struct bt_smp *smp, struct net_buf *buf)
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

joerchan, 2021-02-23:
This one is a false positive, all SMP allowed commands are within index 0, (Keypress being the highest at 0x0e).

@jhedberg Do you think we should convert the definition of allowed commands from `atomic_t allowed_cmds` to `ATOMIC_DEFINE(allowed_commands, SMP_LAST_VALID_OPCODE)`?

jhedberg, 2021-02-24:
> @jhedberg Do you think we should convert the definition of allowed commands from atomic_t allowed_cmds to ATOMIC_DEFINE(allowed_commands, SMP_NUM_OPCODES)?

Yes, that sounds like a good idea, though I'm not 100% sure if it'll suppress the Coverity warning

joerchan, 2021-02-24:
Closing as false positive: SMP allowed commands are within index 0, Keypress being the highest at 0x0e, i.e bit 14 out of 32.

closed 2021-02-24 by joerchan