zephyrproject-rtos/zephyr #32420: [Error] bbc_microbit_v2 building error for lis2dh

Liviu112, 2021-02-17:
**Describe the bug**
Error while building the example lis2dh, it works for bbc_microbit but not for bbc_microbit_v2.
I have tried to find the issue, but I have no luck, one thing which might cause some problems is that the `sm303agr-magn@1e` and `lsm303agr-accel@19` from the `bbc_microbit_v2.dts` file are on the same GPIO pin.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. zephyr 2.4
2. cd zephyr/samples
3. west build -b bbc_microbit_v2 samples/sensor/lis2dh
4. See the error

**Expected behavior**
Bbc_microbit (v1.5 and v2) share the same accelerometer **lsm303agr** it is expected for the same code to work, or a solution to access the accelerometer for v2. 

**Impact**
Not able to progress on my FYP (Final year project) because I can not access the accelerometer information.

**Logs and console output**
 ```
In file included from ../include/sys/util.h:27,
                 from ../include/sched_priq.h:9,
                 from ../include/kernel_includes.h:23,
                 from ../include/kernel.h:17,
                 from ../include/init.h:11,
                 from /home/liviu/Part2/zephyrproject/zephyr/drivers/sensor/lis2dh/lis2dh.c:10:

zephyr/include/generated/devicetree_unfixed.h:2926:40: error: 'DT_N_S_soc_S_i2c_40003000_S_lsm303agr_accel_19_BUS_P_label' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_i2c_40003000_S_lsm303agr_accel_19_P_label'?
 2926 | #define DT_N_INST_0_st_lis2dh          DT_N_S_soc_S_i2c_40003000_S_lsm303agr_accel_19
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/sys/util_internal.h:65:26: note: in definition of macro '__DEBRACKET'
   65 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
../include/sys/util_internal.h:57:2: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   57 |  __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |  ^~~~~~~~~~~~~~~~~~~~
../include/sys/util_internal.h:52:2: note: in expansion of macro '__COND_CODE'
   52 |  __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |  ^~~~~~~~~~~
../include/sys/util.h:414:2: note: in expansion of macro 'Z_COND_CODE_1'
  414 |  Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |  ^~~~~~~~~~~~~
```
There are 300 lines, I have attached the file lis2dh_output.txt for the full output.

**Environment (please complete the following information):**
 - OS: Linux 20.04
 - Toolchain: Zephyr version: 2.4.0, west 0.9

**Additional context**
The bbc_microbit (v1.5) has the Cortex-M0 processor while v2 has the Cortex-M4 processor, I am not sure if that should make any difference in the ability to use the example lis2dh.

carlescufi, 2021-02-17:
Assigning to @mbolivar-nordic since this seems to be a Devicetree issue.

carlescufi, 2021-02-17:
>zephyr 2.4

Do you mean 2.5? Zephyr 2.4 did not have support for the bbc_microbit_v2 board

Liviu112, 2021-02-17:
> > zephyr 2.4
> 
> Do you mean 2.5? Zephyr 2.4 did not have support for the bbc_microbit_v2 board

I have brought the bbc_microbit_v2 Zephyr 2.4, I think it still should work, the Bluetooth works with bbc_microbit_v2 in the 2.4. I basically added the board from 2.5 to 2.4. 
The reason is what I thought 2.4 is more stable because 2.5 is a draft yet.
I am sorry I should have specified that.

carlescufi, 2021-02-17:
>The reason is what I thought 2.4 is more stable because 2.5 is a draft yet.

No, 2.5 was released a few days ago, please do test 2.5.0.

Liviu112, 2021-02-17:
> No, 2.5 was released a few days ago, please do test 2.5.0.

I have tried Zephyr 2.5 now, the exact same error appeared. I have attached the new file which contains the output. 
[lis2dh_output_ZephyrV2.5.txt](https://github.com/zephyrproject-rtos/zephyr/files/5998264/lis2dh_output_ZephyrV2.5.txt)

mbolivar-nordic, 2021-02-23:
Hi, thanks for the detailed report. I've posted a fix in #32597. I can only build test this, since I don't have hardware. @Liviu112 can you please confirm this fixes your issue?

closed 2021-02-24 by MaureenHelm

Liviu112, 2021-02-24:
> Hi, thanks for the detailed report. I've posted a fix in #32597. I can only build test this, since I don't have hardware. @Liviu112 can you please confirm this fixes your issue?

Hi, thank you, yes it works as expected on bbc_microbit_v2, thank you :)

mbolivar-nordic, 2021-02-24:
> yes it works as expected on bbc_microbit_v2

Great :). Thanks for confirming.

PR #32597 (2021-02-24): boards: bbc_microbit_v2: add missing i2c0 compatible
The board's main I2C bus controller doesn't have a compatible set, so
it's not detected as an I2C bus at all.

This breaks the build when trying to build the samples/sensor/lis2dh
application with the lis2dh sensor on that bus.

Fixes: #32420
commit d2280197 (2021-02-24): boards: bbc_microbit_v2: add missing i2c0 compatible

The board's main I2C bus controller doesn't have a compatible set, so
it's not detected as an I2C bus at all.

This breaks the build when trying to build the samples/sensor/lis2dh
application with the lis2dh sensor on that bus.

Fixes: #32420