zephyrproject-rtos/zephyr #32431: RTC driver support on STM32F1 series

ycsin, 2021-02-18:
I was trying to get the 32.768 kHz LSE work for my STM32F103RE custom board to achieve higher timing accuracy for `k_msleep()` (not sure if this is the correct direction), but then I found out that the STM32F1 series doesn't support the `CONFIG_COUNTER_RTC_STM32` flag.

I've added:

```C
#if defined(CONFIG_COUNTER_RTC_STM32)
#include <stm32f1xx_ll_rtc.h>
#include <stm32f1xx_ll_exti.h>
#include <stm32f1xx_ll_pwr.h>
#endif
```

to `soc.h` and relevant flags:

```
CONFIG_COUNTER=y
CONFIG_COUNTER_RTC_STM32=y
CONFIG_CLOCK_STM32_LSE=y
CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y
```

to `*_defconfig`. But the compilation failed due to `stm32f1xx_ll_rtc.h` lacks the `LL_RTC_Date_Get(RTC_TypeDef *RTCx)` function required by `drivers\counter\counter_ll_stm32_rtc.c` and a lots of other related things.

Is there a way of getting the RTC driver supported? The 32bit RTC register on the STM32F1 should be able to hold up to ~136 years. Or any possible workaround for the `stm32f1xx_ll_rtc.h` so that the COUNTER aspect of the LSE can work? (ie: Higher `k_msleep()` precision)

[zephyr_compile_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/6000598/zephyr_compile_log.txt)

erwango, 2021-02-18:
@ycsin Just as a clarification, adding counter driver for F1 won't have any impact on `k_sleep`.
`k_sleep` (and ms derivatives) are clocked by ARM Cortex Systick (cf `drivers/timer/cortex_m_systick.c`).

This doesn't mean it has no interest, but won't it provide the expected improvement.

I think what you rather look for is an implementation of timer api (to clock kernel ticks) using RTC.
A L1 implementation is tempted here: https://github.com/zephyrproject-rtos/zephyr/pull/30790.
Though, I'm not sure implementation can provide a better lock accuracy in ms, as the implementation of a ms ticker using RTC hardware is not easy (taking into account various cases expected by kernel).

ycsin, 2021-02-18:
> @ycsin Just as a clarification, adding counter driver for F1 won't have any impact on `k_sleep`.
> `k_sleep` (and ms derivatives) are clocked by ARM Cortex Systick (cf [...]

I see, thanks for the clarification!

erwango, 2021-02-18:
@ycsin, I'll close the current point as it doesn't answer your need.

closed 2021-02-18 by erwango