zephyrproject-rtos/zephyr #32448: C++ exceptions do not work in ZTEST under QEMU x86

r2r0, 2021-02-18:
**Describe the bug**
C++ exception (with matching catch statement) generated in ZTEST application and executed under QEMU x86 causes immediate test (application) exit.
The same test executed under QEMU for coretex_m3 works fine.

**To Reproduce**
See: attached ztest project.

**Expected behavior**
C++ exceptions should be handled properly in ZTEST environment.

**Impact**
showstopper

**Logs and console output**
See attached logs.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.3
 - Commit SHA: e287fd7ab2fb203d5683eab42ccbd5a3f5c07f62

**Additional context**
[logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/6002993/logs.zip)
[CPP_TEST.zip](https://github.com/zephyrproject-rtos/zephyr/files/6002994/CPP_TEST.zip)

r2r0, 2021-03-02:
It looks that it is a little wider problem - C++ exceptions do not work for Zephyr on QEMU x86(_64).
I modified sample cpp_synchronization by adding simple try/catch in main (I also changed kconfig: CONFIG_NEWLIB_LIBC=y CONFIG_LIB_CPLUSPLUS=y CONFIG_EXCEPTIONS=y).
The effect is the same as reported for ZTEST - throw statement causes "exit" printed on console (but application stays in QEMU).
And as previously, this works well on QEMU for CortexM3.

laurenmurphyx64, 2021-03-03:
I had a coworker confirm that adding a try/catch for throwing an exception in the cpp_synchronization example also doesn't work for the `up_squared`. 

Result:
```
*** Booting Zephyr OS build zephyr-v2.5.0-611-g48321ef2525c  ***
Create semaphore 0x123060
Create semaphore 0x123040
exit
```

prj.conf:
```
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_STD_CPP17=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_EXCEPTIONS=y
```

main.cpp:
```
...
int main(void)
{
	try {
		throw 42;
	}
	catch (int i) {
		printk("No failure!");
		return 0;
	}
...
```

The above example worked fine for `qemu_cortex_m3`.

r2r0, 2021-03-03:
Maybe title of this issue should be changed.

nashif, 2021-03-11:
I suspect the issue is newlib, we can reproduce this also on qemu_cortex_a53 and qemu_leon3. Given that _exit is being called leads me to believe this is something in newlib, so exceptions might not be enabled? 

@galak  any clues re newlib?

laurenmurphyx64, 2021-03-12:
Some links I found about compiler / linker issues with newlib and C++ exceptions:
- 
- https://miscsolutions.wordpress.com/2019/12/27/using-exceptions-in-c-embedded-software/

From stepping through on `qemu_x86` and checking `zephyr.lst`, I noticed that `__cxa_allocate_exception` and  `__cxa_throw` were being called but instead of checking and catching the exception the app just stepped on through to `abort()`.

Using the modified sample mentioned in previous comment:

#### QEMU platforms that successfully catch the exception:
- qemu_cortex_m3
- qemu_cortex_m0
- qemu_cortex_r5
- qemu_arc_em

#### QEMU platforms that fail with exit:
- qemu_x86
- qemu_x86_64
- qemu_cortex_a53
- qemu_riscv64

```
*** Booting Zephyr OS build zephyr-v2.5.0-623-gf9ea76636283  ***
Create semaphore 0x600171ec
Create semaphore 0x600171d8
exit
```

#### QEMU platforms that fail with terminate and exit message
- qemu_xtensa

```
*** Booting Zephyr OS build zephyr-v2.5.0-623-gf9ea76636283  ***
Create semaphore 0x600171ec
Create semaphore 0x600171d8
terminate called after throwing an instance of 'int'
terminate called recursively
exit
```

#### QEMU platforms that hang:
- qemu_riscv32

```
*** Booting Zephyr OS build zephyr-v2.5.0-623-gf9ea76636283  ***
Create semaphore 0x80012b94
Create semaphore 0x80012b80
```

#### QEMU platforms that exit without catching exception or printing catch / exit message
- qemu_leon3

```
*** Booting Zephyr OS build zephyr-v2.5.0-623-gf9ea76636283  ***
Create semaphore 0x400218b4
Create semaphore 0x400218a0
```

/home/herbert/zephyr-external/zephyr/samples/mod_cpp_sync.zip

andyross, 2021-03-29:
Yeah, we just don't seem to be doing this right at all.   The platforms that are working are presumably ones where the native ABI contains enough information to unwind a stack without help from the linker.  I stumbled over this architecture a few weeks ago and read through it, but recognize that I'm absolutely not an expert:

GCC emits a bunch of .eh_frame sections for every output section that has code that implements the exception frame handling (which is optional and managed via assembler primitives, not all functions do this) which contains enough DWARF-like info to extract a return address and a pointer to cleanup hooks.  We're supposed to be collecting these into a single block and bounding them with EH_FRAME_BEGIN/END (sp? names from vague memory!) symbols.  Then we need to link in the crt1.o files from the toolchain, which include .init sections which we need to include in system startup somewhere (IIRC these are actually raw code, not function pointers!) which then call into a registration function ("eh_frame_add()" I think?  Something like that) in libgcc that links them together into a runtime structure that can be searched by exception dispatch.

And we don't do that.  Some of our platforms do bits and pieces of that, but nothing contains the full toolkit.  And some of this is non-trivial: the initialization hooks may need to assume things like a cleared BSS, but otherwise need to be called extremely early; z_cstart() might be too late.  But we absolutely need to be running C code somewhere before it happens, which means that the toolchain crt* code may not be a good fit anyway, we might need to implement our own hook by reverse engineering libgcc.   And if we do that, then maybe we don't need the .eh_frames in a contiguous section and can get away with folding them into .rodata...  Basically there are lots of decisions to be made here.

And then you might ask whether or not clang's implementation of C++ exceptions is compatible, and I'd tell you I don't know and that it probably depends on whether the architecture has a specified C++ ABI or not.

galak, 2021-04-14:
Seeing this issue on qemu_riscv32 as well.

@bob2oneil / @r2r0 can you guys look at working up a proper testcase for this.  It seems like extending tests/application_development/cpp to have a test for catch/throw would be ideal.

galak, 2021-04-14:
So there are a few things to get this working.  I've debugged this some and got a hackish working solution.  I'll try to figure out how to post that somewhere:

* Get a proper testcase for this functionality
* We need to pull in crtbegin.o and crtend.o when we link.  These have various code bits we need for the C++ exception handling for these platforms.  @tejlmand to hopefully help with this.
* Cleanup linker scripts to save .ehframe sections and place them at the end.

galak, 2021-04-14:
Here's my working tree for that 
https://github.com/galak/zephyr/tree/cpp-exception

It in theory this works for x86 and riscv32/64

nashif, 2021-04-16:
@galak  do you want to submit a PR, this is not really broken in the tree after merging with #34291

galak, 2021-04-16:
Note: need to fixup tests/application_development/libcxx/testcase.yaml  to run on all arch's in the future with this is fixed.

galak, 2021-04-16:
#34357 is starts of fix for this issue.

carlescufi, 2021-05-25:
Adding @stephanosio in case this is related to newlib.

galak, 2021-05-25:
> Adding @stephanosio in case this is related to newlib.

I think we know the issues, its mostly a linker script updates and build system changes to pull in crtbegin.o and crtend.o

galak, 2021-05-27:
@tejlmand what's left to fix here?

stephanosio, 2021-05-27:
> @tejlmand what's left to fix here?

@galak the architectures that are not x86(-32), riscv32, xtensa (i.e. arm64 sparc nios2 riscv32 riscv64 x86_64).

tejlmand, 2021-05-27:
With #35647 and #35722 in place should we consider this an enhancement issue ?

stephanosio, 2021-05-27:
> With #35647 and #35722 in place should we consider this an enhancement issue ?

If C++ exception handling never worked for these architectures to begin with (we should really make sure that feature PRs have associated tests with them ...), it should indeed be considered an enhancement issue.

galak, 2021-05-27:
@tejlmand on riscv64/x86_64/arm64 CONFIG_LINKER_SORT_BY_ALIGNMENT=y is causing issues with crtbegin.o not getting linked at the right order.

stephanosio, 2021-05-27:
64-bit archs to be fixed by #35734.
SPARC to be fixed by #35733.
NIOS2 needs verification (`qemu-nios2` cannot run the test due to flash size limitation).

tejlmand, 2021-05-27:
> `qemu-nios2` cannot run the test due to flash size limitation

Increasing the flash can be done with this diff:
```diff
index b1b498b266..7f28a22002 100644
--- a/dts/nios2/nios2-qemu.dtsi
+++ b/dts/nios2/nios2-qemu.dtsi
@@ -18,7 +18,7 @@
 
        flash0: flash@420000 {
                compatible = "soc-nv-flash";
-               reg = <0x420000 0x20000>;
+               reg = <0x420000 0x30000>;
        };
 
        sram0: memory@400000 {
```
after all, flash is cheap for a qemu board :wink: 

But it doesn't solve the problem, as we still are getting this:
```
[114/115] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: nios2
        cpu->env.reset_addr:            0
        cpu->env.exception_addr:        400020
*** Booting Zephyr OS build v2.6.0-rc2-24-g726577132a97  ***
version 201703
Running test suite libcxx_tests
===================================================================
START - test_array
 PASS - test_array in 0.0 seconds
===================================================================
START - test_vector
 PASS - test_vector in 0.0 seconds
===================================================================
START - test_make_unique
 PASS - test_make_unique in 0.0 seconds
===================================================================
START - test_exception
terminate called after throwing an instance of 'int'
terminate called recursively
exit
```

stephanosio, 2021-05-27:
> Increasing the flash can be done with this diff:

@tejlmand IIRC, qemu hanged when I did that, so I assumed it had the flash size hardcoded somewhere in the code -- good to know that it isn't :)

galak, 2021-05-28:
Going to close this issue and open a new one for C++ on NIOS2

galak, 2021-05-28:
Opened #35772 for the NIOS2 C++ support.

closed 2021-05-28 by galak