zephyrproject-rtos/zephyr #32459: the cbprintf unit tests don't actually test all variations

npitre, 2021-02-18:
**Describe the bug**

The cbprintf unit tests are configured using DVIA_TWISTER in testcase.yaml.
This preprocessor define never makes it to the compiler.

**To Reproduce**

1. Apply this change:

```
diff --git a/tests/unit/cbprintf/main.c b/tests/unit/cbprintf/main.c
index 3c562206d2..7e756e992c 100644
--- a/tests/unit/cbprintf/main.c
+++ b/tests/unit/cbprintf/main.c
@@ -21,6 +21,7 @@
  * determine which features are desired.  Yes, this is a mess.
  */
 #ifndef VIA_TWISTER
+#error "VIA_TWISTER is undefined"
 /* Set this to truthy to use libc's snprintf as external validation.
  * This should be used with all options enabled.
  */
```

2. Run `./scripts/twister  -T tests/unit/cbprintf/`

3. Watch all tests fail.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version used: v2.5.0

pabigot, 2021-02-18:
That does seem to be true; it used to work, though.  I'll have to see when it broke.

npitre, 2021-02-23:
Fixed with commit 99425de38774

closed 2021-02-23 by npitre