zephyrproject-rtos/zephyr #32467: up_squared: tests/kernel/device failed.

chen-png, 2021-02-19:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/device
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
 Booting Zephyr OS build zephyr-v2.5.0-239-ge287fd7ab2fb  
Running test suite device

START - test_dummy_device_pm
SKIP - test_dummy_device_pm

START - test_build_suspend_device_list
SKIP - test_build_suspend_device_list

START - test_dummy_device
E: Page fault at address 0x158 (error code 0x0)
PASS - test_dummy_device
E: Linear address not present in page tables

**E: Access violation: supervisor thread not allowed to read**
START - test_enable_and_disable_automatic_idle_pm
E: PTE: not present
E: Page fault at address (nil) (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000000000 RBX: 0x000000000011a280 RCX: 0x0000000000000001 RDX: 0x00000000000000ff
E: RSI: 0x000000000011ada0 RDI: 0x000000000011a280 RBP: 0x00000000017c4ea0 RSP: 0x00000000017c4e90
E:  R8: 0x0000000000000001  R9: 0x8000000000000006 R10: 0x0000000000105bb9 R11: 0x0000000000000000
E: R12: 0x000000000011a280 R13: 0x00000000017bd000 R14: 0x0000000000002000 R15: 0x0000000000105bb9
E: RSP: 0x00000000017c4e90 RFLAGS: 0x0000000000010086 CS: 0x0018 CR3: 0x0000000001dca000
E: RIP: 0x0000000000109676
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x11ada0 (main)
E: Halting system

**Environment (please complete the following information):**

 - Toolchain: zephyr-sdk-0.12.2
 - Commit ID: e287fd7ab2f

chen-png, 2021-02-22:
for testcases "kernel.common.tls", "kernel.message_queue", failed with the same error "E: Access violation: supervisor thread not allowed to read".

dcpleung, 2021-02-23:
Tried with #32557 and seems to fix the issue. So wait for that to be merged...

andyross, 2021-02-23:
Indeed, this is likely a dupe of the abort menagerie of troubles.  I'm doing a ton of iterated qemu_x86_64 runs right now and not seeing this.

closed 2021-02-24 by nashif

alexpaschoaletto, 2025-02-20:
FWIW, I'm having a very similar error when trying to run a simple code involving condvars in `qemu_x86_64`. The error message points that _"supervisor thread not allowed to execute"_.

The code in question is the following, which implements simple thread barriers using the condvars API:

```c
#include <zephyr/kernel.h>

#define PRIORITY 10
#define NUM_THREADS 5
#define STACK_SIZE  1024

/** each test thread stack will be stored here */
K_THREAD_STACK_ARRAY_DEFINE(_stacks, NUM_THREADS, STACK_SIZE);

/** each test thread control block will be stored here */
struct k_thread _threads[NUM_THREADS];

// BARRIERS //////////////////////////////////////////////////////////////////////

typedef struct {
    struct k_mutex mutex;
    struct k_condvar condvar;
} barrier_t;

void barrier_init(barrier_t *barrier)
{
    k_mutex_init(&(barrier->mutex));
    k_condvar_init(&(barrier->condvar));
}

void barrier_wait(barrier_t *barrier, k_timeout_t timeout)
{
    k_mutex_lock(&(barrier->mutex), K_FOREVER);
    k_condvar_wait(&(barrier->condvar), &(barrier->mutex), timeout);
    k_mutex_unlock(&(barrier->mutex));
}

void barrier_release(barrier_t *barrier)
{
    k_mutex_lock(&(barrier->mutex), K_FOREVER);
    k_condvar_broadcast(&(barrier->condvar));
    k_mutex_unlock(&(barrier->mutex));
}

////////////////////////////////////////////////////////////////////////////////

int main(void)
{
    barrier_t barrier;
    barrier_init(&barrier);

    /* thread that will attempt to use a barrier */
    void waiter_thread(void *thread_id, void *ignored2, void *ignored3)
    {
        int id = POINTER_TO_INT(thread_id);

        /* wait on barriers here */
        printk("thread %d: waiting on barrier\n", id);
        barrier_wait(&barrier, K_FOREVER);
        printk("thread %d: released from barrier\n", id);
    }

    printk("\ncreating threads...\n\n");

    for(int i = 0; i < NUM_THREADS; i++)
    {
        k_thread_create(&_threads[i], _stacks[i], STACK_SIZE, waiter_thread,
			INT_TO_POINTER(i), NULL, NULL,
            PRIORITY, 0, K_MSEC(500*i)
        );

        printk("thread %d created.\n", i);
    }

    k_sleep(K_MSEC(5000));
    printk("\nreleasing barrier...\n\n");
    barrier_release(&barrier);

    /* Wait for all threads to complete */
    for (int i = 0; i < NUM_THREADS; i++)
    {
        k_thread_join(&_threads[i], K_FOREVER);
    }

	return 0;
}
```

prj.conf:
```c
CONFIG_LOG=y
CONFIG_QEMU_ICOUNT=n
```

expected output:
```c
creating threads...

thread 0 created.
thread 1 created.
thread 2 created.
thread 3 created.
thread 4 created.
thread 0: waiting on barrier
thread 1: waiting on barrier
thread 2: waiting on barrier
thread 3: waiting on barrier
thread 4: waiting on barrier

releasing barrier...

thread 0: released from barrier
thread 1: released from barrier
thread 2: released from barrier
thread 4: released from barrier
thread 3: released from barrier
```

observed error:
```c
creating threads...

thread 0 created.
thread 1 created.
thread 2 created.
thread 3 created.
thread 4 created.
[00:00:00.010,000] <err> os: Page fault at address 0x130968 (error code 0x11)
[00:00:00.010,000] <err> os: Access violation: supervisor thread not allowed to execute
[00:00:00.030,000] <err> os: PTE: 0x130000 -> 0x0000000000130000: RW A D XD 
[00:00:00.030,000] <err> os: RAX: 0x000000000010b000 RBX: 0x0000000000130968 RCX: 0x0000000000000000 RDX: 0x0000000000000000
[00:00:00.030,000] <err> os: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x0000000000122a20 RSP: 0x00000000001229e8
[00:00:00.030,000] <err> os:  R8: 0x0000000000000000  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000000
[00:00:00.030,000] <err> os: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000122a38
[00:00:00.030,000] <err> os: RSP: 0x00000000001229e8 RFLAGS: 0x0000000000000203 CS: 0x0018 CR3: 0x000000000013b000
[00:00:00.030,000] <err> os: RIP: 0x0000000000130968
[00:00:00.030,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
[00:00:00.030,000] <err> os: Current thread: 0x10b000 (unknown)
[00:00:00.070,000] <err> os: Halting system
```

The problem seems to be that `barrier` is declared within `main`. If altered to be a global variable, everything seems to work alright. This happens in both `qemu_x86` and `qemu_x86_64`, but it DOESN'T happen with `qemu_x86_tiny`. Also doen't happen with `qemu_riscv32` or `qemu_xtensa`.