zephyrproject-rtos/zephyr #32469: Twister: potential race conditions.

LeiW000, 2021-02-19:
Flashing errors show up when twister runs with multi-threads feature enabled. Please see following. The shell script attached to this issue is what I used to reproduce the issue.

leiw@leiw-NUC8i7BEH:~/Documents/src/zephyrproject/zephyr$ ./scripts/twister --device-testing --device-serial /dev/ttyACM0 -p reel_board -T tests/kernel
Renaming output directory to /home/leiw/Documents/src/zephyrproject/zephyr/twister-out.10
INFO    - Zephyr version: v2.5.0-rc2-85-gbda751f360b5
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 112 test suites (112 configurations) selected, 19 configurations discarded due to filters.

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
| reel_board |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   75/  93  80%  skipped:   31, failed:    0
ERROR   - reel_board                tests/kernel/tickless/tickless_concept/kernel.tickless.concept FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/tickless/tickless_concept/kernel.tickless.concept/device.log
INFO    - Total complete:   77/  93  82%  skipped:   31, failed:    1
ERROR   - reel_board                tests/kernel/msgq/msgq_api/kernel.message_queue    FAILED: Unknown
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/msgq/msgq_api/kernel.message_queue/handler.log
INFO    - Total complete:   78/  93  83%  skipped:   31, failed:    2
ERROR   - reel_board                tests/kernel/early_sleep/kernel.common.sleep       FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/early_sleep/kernel.common.sleep/device.log
INFO    - Total complete:   79/  93  84%  skipped:   31, failed:    3
ERROR   - reel_board                tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap/device.log
INFO    - Total complete:   81/  93  87%  skipped:   31, failed:    4
ERROR   - reel_board                tests/kernel/sleep/kernel.common.timing            FAILED: Failed
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/sleep/kernel.common.timing/handler.log
INFO    - Total complete:   82/  93  88%  skipped:   31, failed:    5
ERROR   - reel_board                tests/kernel/threads/thread_error_case/kernel.threads.error.case FAILED: Unknown
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/threads/thread_error_case/kernel.threads.error.case/device.log
INFO    - Total complete:   83/  93  89%  skipped:   31, failed:    6
ERROR   - reel_board                tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/device.log
INFO    - Total complete:   84/  93  90%  skipped:   31, failed:    7
ERROR   - reel_board                tests/kernel/context/kernel.common                 FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/context/kernel.common/device.log
INFO    - Total complete:   86/  93  92%  skipped:   32, failed:    8
ERROR   - reel_board                tests/kernel/threads/thread_init/kernel.threads.init FAILED: Failed
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/threads/thread_init/kernel.threads.init/handler.log
INFO    - Total complete:   87/  93  93%  skipped:   32, failed:    9
ERROR   - reel_board                tests/kernel/threads/tls/kernel.threads.tls        FAILED: Unknown
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/threads/tls/kernel.threads.tls/device.log
INFO    - Total complete:   88/  93  94%  skipped:   32, failed:   10
ERROR   - reel_board                tests/kernel/threads/thread_apis/kernel.threads.apis FAILED: Device issue (Flash?)
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/threads/thread_apis/kernel.threads.apis/handler.log
INFO    - Total complete:   91/  93  97%  skipped:   32, failed:   11
ERROR   - reel_board                tests/kernel/threads/no-multithreading/kernel.threads.no-multithreading FAILED: Unknown
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/threads/no-multithreading/kernel.threads.no-multithreading/device.log
INFO    - Total complete:   93/  93  100%  skipped:   32, failed:   12
INFO    - 68 of 80 test configurations passed (85.00%), 12 failed, 32 skipped with 0 warnings in 521.34 seconds
INFO    - In total 799 test cases were executed, 292 skipped on 1 out of total 329 platforms (0.30%)
INFO    - 80 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board      | ID   |   Counter |
|------------|------|-----------|
| reel_board |      |        80 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

laurenmurphyx64, 2021-02-22:
If it helps while investigating, I just tried Steven's script and had the same issue on my FRDM-K64F with a different error:

```
INFO    - Zephyr version: zephyr-v2.5.0-192-g570063138c0d
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 114 test scenarios (114 configurations) selected, 21 configurations discarded due to filters.

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
| frdm_k64f  |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   64/  93  68%  skipped:   31, failed:    0
ERROR   - frdm_k64f                 tests/kernel/timer/timer_error_case/kernel.timer   FAILED: Device issue (Flash?)
ERROR   - see: /home/herbert/zephyr-external/zephyr/twister-out/frdm_k64f/tests/kernel/timer/timer_error_case/kernel.timer/device.log
INFO    - Total complete:   65/  93  69%  skipped:   31, failed:    1ERROR   - Serial device error: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

ERROR   - frdm_k64f                 tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing FAILED: Failed
ERROR   - see: /home/herbert/zephyr-external/zephyr/twister-out/frdm_k64f/tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing/build.log
INFO    - Total complete:   66/  93  70%  skipped:   31, failed:    2ERROR   - Serial device error: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

...

(Every test case in between fails)

...

ERROR   - frdm_k64f                 tests/kernel/xip/arch.common.xip                   FAILED: Failed
ERROR   - see: /home/herbert/zephyr-external/zephyr/twister-out/frdm_k64f/tests/kernel/xip/arch.common.xip/build.log
INFO    - Total complete:   92/  93  98%  skipped:   32, failed:   27ERROR   - Serial device error: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

ERROR   - frdm_k64f                 tests/kernel/queue/kernel.queue                    FAILED: Failed
ERROR   - see: /home/herbert/zephyr-external/zephyr/twister-out/frdm_k64f/tests/kernel/queue/kernel.queue/build.log
INFO    - Total complete:   93/  93  100%  skipped:   32, failed:   28
INFO    - 54 of 82 test configurations passed (65.85%), 28 failed, 32 skipped with 0 warnings in 1167.15 seconds
INFO    - In total 826 test cases were executed, 247 skipped on 1 out of total 324 platforms (0.31%)
INFO    - 55 test configurations executed on platforms, 27 test configurations were only built.
```

The flash failure caused the FRDM board to switch to the bootloader for some reason:

[image]

My attempts to disconnect, reconnect and flash the board with the Hello World sample also fail after this. I think I'm going to have to [reinstall the bootloader](https://community.nxp.com/t5/Kinetis-Microcontrollers/My-FRDM-k64-board-has-a-an-issue-when-I-boot-loaded-red-and/m-p/419680).

**Environment:**
 - OS: Ubuntu 18.04 (also on a NUC)
 - Python: 3.7.5
 - Toolchain: Zephyr SDK 0.12.3
 - Commit SHA: 28897d

LeiW000, 2021-02-23:
Because of this issue, I'm wondering if we should disable multi-threads by default in Twister.

nashif, 2021-03-03:
please try https://github.com/zephyrproject-rtos/zephyr/pull/32784 and see if it fixes the issue for you.

LeiW000, 2021-03-04:
Yes, Give me some time. I will test it.

LeiW000, 2021-03-05:
Need more time. It seems that there is something wrong with my development environment.

laurenmurphyx64, 2021-03-09:
It seems to be working for me at least. :+1:

LeiW000, 2021-03-17:
I still can see the issue with the latest codebase.
Commit id: 35f0ed23c9
Board: reel_board

leiw@leiw-NUC8i7BEH:~/Documents/src/zephyrproject/zephyr$ ./loop-testing.sh
Testing: 1
Renaming output directory to /home/leiw/Documents/src/zephyrproject/zephyr/twister-out.1
INFO    - Zephyr version: zephyr-v2.5.0-1210-g35f0ed23c9b6
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 116 test scenarios (116 configurations) selected, 21 configurations discarded due to filters.

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
| reel_board |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    6/  95   6%  skipped:   24, failed:    0
ERROR   - reel_board                tests/kernel/pending/kernel.objects                FAILED: Flash error
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/pending/kernel.objects/handler.log
INFO    - Total complete:   13/  95  13%  skipped:   26, failed:    1
ERROR   - reel_board                tests/kernel/workq/work_queue/kernel.workqueue     FAILED: Flash error
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/workq/work_queue/kernel.workqueue/handler.log
INFO    - Total complete:   24/  95  25%  skipped:   27, failed:    2
ERROR   - reel_board                tests/kernel/sched/schedule_api/kernel.scheduler.dumb_timeslicing FAILED: Flash error
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/sched/schedule_api/kernel.scheduler.dumb_timeslicing/handler.log
INFO    - Total complete:   30/  95  31%  skipped:   27, failed:    3
ERROR   - reel_board                tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Flash error
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/handler.log

closed 2021-05-31 by github-actions[bot]