zephyrproject-rtos/zephyr #32509: west build -p auto -b nrf52840dk_nrf52840 samples/bluetooth/hci_uart FAILED with  zephyr-v2.5.0

taotaodong527, 2021-02-20:
**Describe the bug**
west build -p auto -b nrf52840dk_nrf52840 samples/bluetooth/hci_uart FAILED with  zephyr-v2.5.0
**14: Could not open '/home/dongtaotao/zephyrproject/zephyr/' (in 'osource "$(ZEPHYR_TRUSTED_FIRMWARE_M_KCONFIG)"') (EISDIR: Is a directory)**

**To Reproduce**
dongtaotao@dongtaotao-OptiPlex-5070:~/zephyrproject/zephyr$ git checkout zephyr-v2.5.0

dongtaotao@dongtaotao-OptiPlex-5070:~/zephyrproject/zephyr$ **west** build -p auto -b nrf52840dk_nrf52840 samples/bluetooth/hci_uart

**Expected behavior**
Build should be successfully

**Impact**

**Logs and console output**
OptiPlex-5070:~/zephyrproject/zephyr$ git checkout zephyr-v2.5.0
Note: checking out 'zephyr-v2.5.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at fe7c2efca8 release: Bump version to 2.5.0
OptiPlex-5070:~/zephyrproject/zephyr$ west build -p auto -b nrf52840dk_nrf52840 samples/bluetooth/hci_uart
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): /home/dongtaotao/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/dongtaotao/zephyrproject/zephyr/samples/bluetooth/hci_uart
-- Zephyr version: 2.5.0 (/home/dongtaotao/zephyrproject/zephyr)
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/dongtaotao/.cache/zephyr
-- Using toolchain: zephyr 0.11.4 (/home/dongtaotao/zephyr-sdk-0.11.4)
-- Found dtc: /home/dongtaotao/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/dongtaotao/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /home/dongtaotao/zephyrproject/zephyr/samples/bluetooth/hci_uart/boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: /home/dongtaotao/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/dongtaotao/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/dongtaotao/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/dongtaotao/zephyrproject/zephyr/Kconfig
/home/dongtaotao/zephyrproject/zephyr/scripts/kconfig/kconfig.py: build/Kconfig/Kconfig.modules:**14: Could not open '/home/dongtaotao/zephyrproject/zephyr/' (in 'osource "$(ZEPHYR_TRUSTED_FIRMWARE_M_KCONFIG)"') (EISDIR: Is a directory)**
**CMake Error at /home/dongtaotao/zephyrproject/zephyr/cmake/kconfig.cmake:265 (message):
  command failed with return code: 1**
Call Stack (most recent call first):
  /home/dongtaotao/zephyrproject/zephyr/cmake/app/boilerplate.cmake:536 (include)
  /home/dongtaotao/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/dongtaotao/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dongtaotao/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/dongtaotao/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja
/usr/bin/cmake --regenerate-during-build -S/home/dongtaotao/zephyrproject/zephyr/samples/bluetooth/hci_uart -B/home/dongtaotao/zephyrproject/zephyr/build
ninja: error: rebuilding 'build.ninja': subcommand failed
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/dongtaotao/zephyrproject/zephyr/build

**Additional context**
master branch works as expected

s-kelvin, 2021-02-22:
Running `west update` should fix this. 
If not, run `git clean -fd` after you've backed up all your local changes. Then run `west update`.

ioannisg, 2021-02-22:
Thanks @s-kelvin , agreed, this needs west update and local cleaning. 
Closing this, as I cannot reproduce on 2.5.0, neither on current master.
@taotaodong527 please, re-open, if needed.

closed 2021-02-22 by ioannisg

shreenibk, 2024-09-23:
@s-kelvin I an into something similar issue with different config variable, your solution worked. Thanks.