zephyrproject-rtos/zephyr #32511: Zephyr build fail with LLVM on Ubuntu for target ARC

YanzhouYu, 2021-02-20:
**Is your enhancement proposal related to a problem? Please describe.**
I am trying to build zephyr sample project 'hello_world' with ZEPHYR_TOOLCHAIN_VARIANT=llvm for board qemu_arc_em, qemu_arc_hs. But build failed with log <clang: error: unknown argument: '-mno-sdata'>. 

**Describe the solution you'd like**
Suppose zephyr can support build with llvm toolchain for target ARC related board.

**Additional context**
If I change target board to qemu_x86, it can build successfully. 
Could you please help confirm if ARC is supported for LLVM/clang compiler or oneApi ICX compiler currently?

nashif, 2021-02-20:
this is not supported yet, you can build using gcc or metaware.

ruuddw, 2021-02-22:
Hi @Yanzhou, as Anas mentioned, we do not support native ARC LLVM for Zephyr. Though there is initial, experimental ARC support for LLVM, the recommended and supported toolchains are GCC and MetaWare (which internally uses LLVM).
If this changes in the future, we can re-open this enhancement request - for now I close it.

closed 2021-02-22 by ruuddw