zephyrproject-rtos/zephyr #32515: zephyr/kernel/thread.c:382 failed

Dream-math, 2021-02-20:
Hi, my python code communicates with zephyr program based on qemu-riscv32 board, after several rounds, the zephyr will crash, and the python code will receive `zephyr/kernel/thread.c:382`( I guess this is the problem), their communication is based on FIFO file.

Here is the zephyr program writing message
```
int uart_rx_buf_read(struct uart_rx_buf_t* buf, uint8_t* data, size_t data_size_bytes) {                                                                                               
          unsigned int key = irq_lock();                                                                                                                                                       
          int bytes_read = ring_buf_get(&buf->buf, data, data_size_bytes);                                                                                                                     
          if (bytes_read)                                                                                                                                                                      
            TVMLogf("READ MESSAGE FROM HOST : %d", bytes_read);                                                                                                                                
          irq_unlock(key);                                                                                                                                                                     
          return bytes_read;                                                                                                                                                                   
  }        
```
and `zephyr/kernel/thread.c:382` code is
```
void z_check_stack_sentinel(void)
{
        uint32_t *stack;

        if ((_current->base.thread_state & _THREAD_DUMMY) != 0) {
                return;
        }

        stack = (uint32_t *)_current->stack_info.start;
        if (*stack != STACK_SENTINEL) {
                /* Restore it so further checks don't trigger this same error */
                *stack = STACK_SENTINEL;
                z_except_reason(K_ERR_STACK_CHK_FAIL);
        }
}
```
when I debug with single step, I found it crashed even not calling zephyr functions. 

my command to run zephyr program:
```
qemu-hack/qemu-system-riscv32 -s -S -nographic -machine virt -bios none -m 256 -net none -pidfile qemu.pid -chardev pipe,id=con,mux=on,path=/tmp/tmpl_8blijp/fifo -serial chardev:con -mon chardev=con,mode=readline -icount shift=6,align=off,sleep=off -rtc clock=vm -kernel /home/hye/tmp/tvm/tests/micro/qemu/test_zephyr-host-workspace/2021-02-20T16-09-11/build/runtime/zephyr/zephyr.elf
```
Any help would be appreciated, thanks!

Dream-math, 2021-02-22:
If I replace the qemu-riscv32 with qemu-x86, there is nothing wrong, it works well, so is this related to riscv ?

cwshu, 2021-02-22:
Hi @Dream-math,
Exception at line 382 occurs because the checking of `CONFIG_STACK_SENTINEL` is failed. `CONFIG_STACK_SENTINEL` is a SW-based method to check the stack overflow by checking the pollution of magic bytes at lowest address of stack.
It means this thread has stack overflow and you may need to enlarge the thread stack size. (or other stack overflow reason)

Dream-math, 2021-02-23:
@cwshu Thanks for your reply,  there are one place in my main.c related to memory size explicitly:
```
K_HEAP_DEFINE(tvm_memory_pool, 8000); /* 122880 = 1024 * 120 */

tvm_crt_error_t TVMPlatformMemoryAllocate(size_t num_bytes, DLContext ctx, void** out_ptr) {
  /* use k_heap_alloc to keep pace with K_HEAP_DEFINE */
  /* *out_ptr = k_mem_pool_malloc(&tvm_memory_pool, num_bytes); */
  *out_ptr = k_heap_alloc(&tvm_memory_pool, num_bytes, K_NO_WAIT);
  return (*out_ptr == NULL) ? kTvmErrorPlatformNoMemory : kTvmErrorNoError;
}
```
and the next place is running the program:
```
qemu-system-riscv32 -s -S -nographic -machine virt -bios none -m 256 
```
so I just need to increase these two size? I search on the internet, but didn't find too much, can you give me some advice about enlarging the thread stack size, how to find and set the stack size? Thanks!

cwshu, 2021-02-23:
@Dream-math You should find which thread has stack overflow at first (e.g. check `_current`). Then enlarging stack size in the `k_thread_create()` or `K_THREAD_DEFINE()` of this thread.
If it's stack overflow in main thread, you can set `CONFIG_MAIN_STACK_SIZE` to change stack size. (other system threads also have their own kconfig option)

Dream-math, 2021-02-23:
Thanks, it works!

carlescufi, 2021-02-23:
Closing since this seems solved now. Please reopen if that is not the case.

closed 2021-02-23 by carlescufi