zephyrproject-rtos/zephyr #32516: RFC: 1-Wire driver

thedjnK, 2021-02-20:
## Introduction

1-Wire is a common communication bus used by maxim/dallas parts for things like ID tokens, temperature sensors and other misc. devices which are common on many devices/sensor platforms. 1-wire client devices can be powered with a dedicated power line or can be parasitically powered from the 1-wire communication line itself, the protocol supports having multiple sensors on a single bus which makes it very useful for constrained embedded devices and platforms.

### Problem description

There is currently no support in zephyr for 1-wire, the aim of this RFC would be to come up with a plan of how a universal driver (i.e. bitbang) can be created and fit within the zephyr ecosystem and run on all supported devices with the tight timing requirements that the 1-wire protocol requires (us delays are required)
A good starting point for reading upon 1-wire would be this datasheet: https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf the timing and details on the 1-wire protocol is towards the end of the document.

### Proposed change/RFC

There was previously this draft for 1-wire support but it was not fixed up and has since been closed because assumingly the original author has abandoned it: https://github.com/zephyrproject-rtos/zephyr/pull/6875
Given that this change was last updated about two years ago and zephyr has moved on a lot since then, it would be good to see if all functionality that would be required for implementation of a 1-wire driver is available, or if not, then when (assuming it will) it will become available.

My questions are:
1. As mentioned in a linked comment in that pull request, nanosleep is now implemented in zephyr as per https://github.com/zephyrproject-rtos/zephyr/pull/25559 but it doesn't seem to give better than us accuracy - is this usable for 1-wire or would a faster/more accurate delay system be required?
2. The previous PR is nordic-centric because of how the timing system works on nordic devices, do we have a system/API in zephyr which can provide the required timings which is compatible with all modules and does not need device HAL-specific function calls?
3. What test code would be required for such a driver to ensure timing is correct and can be verified from CI?
4. Maxim offers an alternative implementation for 1-wire hosts using a UART which switches from 9600 baud to 115200 baud, detailed on https://www.maximintegrated.com/en/design/technical-documents/tutorials/2/214.html - if a UART was used instead of a GPIO (or as well as, having 2 drivers is not an issue) would this then resolve the issue of needing a cross-platform compatible accurate timing/delay sounce? And if this were to be implemented, how would things like baud speed be set given that the UART driver in the DTS requires a static baud rate (assuming similar implementation to modem drivers in that they are using the base UART peripheral) or would it require a separate DTS peripheral that does not sit on top of a UART peripheral?

This is my basic list of questions before thinking about how to possibly implement or structure the code which would depend on everyone's opinions. If anyone else has any questions about having 1-wire support I would be interested to hear those.

### Concerns and Unresolved Questions

1. If the delay timing system in zephyr is of a good enough accuracy to work with the 1-wire timing
2. If said delay system can be utilised on all supported boards with a general purpose API that is not tied to e.g. the specific HAL of one module
3. How to ensure timing requirements can be met assuming driver is implemented accross all supported platforms from CI (this also raises another question of ensuring code/drivers work on real hardware, it's all well and good to run through test cases in qemu but running the code on a real microcontroller might give a very different output than the simulation - are there any plans to have some sort of real hardware test platform which can be used to test components against all supported boards? I have 4 MCUs to test with which is a a bit short of the 250 or so supported modules in zephyr)

picsil, 2021-02-20:
I would love to see this implemented as my application relies extensively on having multiple masters on the board (each using its own GPIO pin).

de-nordic, 2021-02-22:
Figure 16 of the [DS18S20](https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf) best describes timing requirements.
I do not think that there would be problems with implementing the "master" side as there quite long, although there may be need to lock interrupts for "master" 1 bit as it consists of falling edge followed by rising edge that should occur in 1us and be at Vcc level at most in 15us.
The bit slots are long and should be between 60 and 120us which is a lot of time, and in case of 1 you can actually hold line in hi-level as long as you wish, the keeping it at 0 level will cause reset after 480us - this basically means that the time between bits in transfer may be very long
The reading might be a problem because master is supposed to pull line down, to mark bit start, and release it within 1us for "slave" to pull it down is 0 is to be transferred and then sample it within 15us from the pull-down. The "slave" device is not expected to hold the line down, for 0 bit, more than 15us, as the "master" device is.
This means that if you are using single pin for that, you will have
1) Set IO state down
2) Switch to out
3) Switch to in
4) wait for 15us - (time between start of step 2 to step 4) - (time to sample)

The timers, how long does it take to serve interrupt routines will depend on how the mcu is clocked. So it is not how accurately the sleep is but rather how fast can you do something after the sleep ends, whether you have to idle wait or can you jump out and do some stuff on faster devices to not waste time idling. 
Probably the timing could be relaxed if two gpio are used as there would be no direction switching.

We should probably do some measurement first across different boards, regarding how fast the pins can be served and how does the clocking, etc influence the timing.

I have done several mcu based 1-wire, bare metal,  "slave" devices for various purposes and the mcu I have been using at that time would take ~7us to just jump to and serve the interrupt not including line sample or gpio direction change, fortunately for me the timings for the "slave" to sample line are quite relaxed and are not expected to occur in first 15us from "master" pulling line down: you have like 30us to do stuff.

MarcianoPreciado, 2021-09-27:
Having implemented a 1-wire slave in Zephyr on an nRF52840, I think that a universal bit-banged driver will be incredibly difficult. There are, however, [application notes](https://www.maximintegrated.com/en/design/technical-documents/tutorials/2/214.html) on how to implement 1-wire masters using a UART. Further, I think it would be good for the community to have a single 1-wire master driver that could use bit-bang, UART, or a bridge device like the [DS2480B](https://www.maximintegrated.com/en/products/interface/controllers-expanders/DS2480B.html), all with the same interface.

thedjnK, 2021-09-28:
> Having implemented a 1-wire slave in Zephyr on an nRF52840, I think that a universal bit-banged driver will be incredibly difficult. There are, however, [application [...]

There's an open PR for 1-wire here https://github.com/zephyrproject-rtos/zephyr/pull/37663
I created a basic bit-bang 1-wire PoC driver on top of zephyr 2.5 some time ago, talking to an DS18B20 and it worked but as you say how universal it would have been between other micros I do not know

carlescufi, 2021-11-02:
@MarcianoPreciado and @thedjnK and @picsil could you please review the proposed implementation [here](#37663)?

lairdjm, 2021-11-03:
@carlescufi Doesn't suit the original requirement for me at least because it requires using a UART, which for the target devices I had in mind, do not have a spare UART. The mock up code I created wrote the data via a GPIO, i.e. called a function to output a byte of data which called a function to output a bit of data, because of the timing criticality of 1-wire, the bit output function needs to be in a critical section but the byte output part doesn't (to prevent missing important events such as radio events, etc.), it worked for nRF51 and nRF52832 when I was looking into it. I guess that PR could be expanded to have a GPIO driver but looks to have been abandoned for 3 months

MarcianoPreciado, 2021-11-03:
@lairdjm it may not have a bit banged implementation yet, but it is the most complete foundation for a zephyr 1-wire api I've seen. I don't think it would be too difficult for somebody else to make a bitbanged driver using his work.

str4t0m, 2021-11-04:
Hi,
I have left a comment in the PR #37663. Please let me know if I you would be interested to start the discussion of a 1-Wire API using this PR and if you already have any feedback especially on the API, but also on the serial driver implementation.

carlescufi, 2021-12-21:
> @carlescufi Doesn't suit the original requirement for me at least because it requires using a UART, which for the target devices I had in mind, do not have a spare UART. The mock up code I created [...]

Looks like @str4t0m has picked it up again recently and replied to your comment @lairdjm

PR #37663 (2022-07-05): 1-Wire API, with zephyr-serial driver and DS18B20 Sensor
This PR introduces an API for 1-Wire devices.
Additionally, the zephyr-serial driver implements this API using serial polling operations.
More optimized implementations could be added later, but a driver on top of serial polling should work on most platforms.
Basic API tests have been added to make sure the driver compiles without errors.
In order to demonstrate the API and bindings for 1-Wire, also a DS18b20 driver and a sample are added.
The API mostly matches the one proposed in #6875 with a few modifications and additions.

Closes #3 [... cut]
closed 2022-07-05 by carlescufi