zephyrproject-rtos/zephyr #32531: get_maintainer.py cannot parse MAINTAINERS.yml

aescolar, 2021-02-22:
**Describe the bug**
The script get_maintainer.py cannot parse MAINTAINERS.yml
It seems the syntax of MAINTAINERS.yml has diverged quite a bit from what get_maintainers can process.

**To Reproduce**
Steps to reproduce the behavior:
1. Try running get_maintainers against any file:
```
$ scripts/get_maintainer.py path CMakeLists.txt 
```

**Expected behavior**
get_maintainer not erroring out

**Impact**
Users cannot use script to find maintainers

**Logs and console output**
```$ scripts/get_maintainer.py path CMakeLists.txt 
scripts/get_maintainer.py: error: zephyr/MAINTAINERS.yml: glob pattern 'subsys/bluetooth/*' in 'files' in area 'Bluetooth' matches a directory, but has no trailing '/'
```

**Environment (please complete the following information):**
 - OS: Linux
 - Python 3.6.8

yashi, 2021-04-29:
how do you want to fix this?  It's nice for a casual user to check who's maintaining what.  I've checked the file and noticed that:
a) some files have been moved to other dirs
b) some glob patterns match other dirs

for (a), I can send a patch to MAINTAINERS.yml.  But for (b), I'm not sure what an acceptable fix is.  you can 1) fix MAINTAINERS.yml, or 2) ignore directory match when `/` is not present in a given pattern.

Here is a quick patch to convert the fatal errors to warnings:
```
@@ -419,6 +419,9 @@ def _check_maintainers(maints_path, yaml):
     def ferr(msg):
         _err("{}: {}".format(maints_path, msg))  # Prepend the filename
 
+    def warn(msg):
+        _warn("{}: {}".format(maints_path, msg)) # Prepend the filename
+
     if not isinstance(yaml, dict):
         ferr("empty or malformed YAML (not a dict)")
 
@@ -471,7 +474,7 @@ def _check_maintainers(maints_path, yaml):
                     if not glob_pattern.endswith("/"):
                         for path in paths:
                             if path.is_dir():
-                                ferr("glob pattern '{}' in '{}' in area '{}' "
+                                warn("glob pattern '{}' in '{}' in area '{}' "
                                      "matches a directory, but has no "
                                      "trailing '/'"
                                      .format(glob_pattern, files_key,
@@ -530,6 +533,9 @@ def _err(msg):
     raise MaintainersError(msg)
 
 
+def _warn(*args):
+    print(*args, file=sys.stderr)
+
 def _giterr(msg):
     raise GitError(msg)
```

Is this acceptable?

aescolar, 2021-04-29:
@yashi The issue (as far as I understand it) would be that the syntax of MAINTAINERS has diverged from the original file format, which is what get_maintainer.py expected. But the yml file is also parsed now to autolabel, add reviewers and auto assign PRs by a separate script which @nashif wrote (AFAIK).
So I guess the first question is, should MAINTAINERS syntax be rolled back to the original and ensured that the new script works with that, OR the syntax accepted as the new one, and get_maintainer.py updated.
I guess @nashif can make that call as he seems to be the one doing most around the scripting here nowadays.

mbolivar-nordic, 2021-08-11:
I'm quite surprised that this ticket was marked low priority.

mbolivar-nordic, 2021-09-15:
I'm using this ticket to track the process WG item for deprecating CODEOWNERS in favor of MAINTAINERS.yml. Maybe that's a mistake but it will do for now.

nashif, 2021-09-16:
> I'm using this ticket to track the process WG item for deprecating CODEOWNERS in favor of MAINTAINERS.yml. Maybe that's a mistake but it will do for now.

that is a bad idea, this is not related to CODEOWNERS at all, please create a new issue for tracking :)

PR #38583 (2021-09-16): MAINTAINER: path and glob fixes
Lots of things have moved and fix a few globs

Fixes #32531
commit d9250b1c (2021-09-16): MAINTAINER: path and glob fixes

Lots of things have moved and fix a few globs

Fixes #32531
closed 2021-09-16 by nashif