zephyrproject-rtos/zephyr #32536: Codec phy connection ASSERTION FAIL [event.curr.abort_cb]

a1ien, 2021-02-22:
I try test coded phy connection with bt_shell.

**To Reproduce**
Steps to reproduce the behavior:
1. cmake -B ble_shell -GNinja -DBOARD=nrf52840dongle_nrf52840 -DCONF_FILE=bt_shell_defconfig zephyr/samples/subsys/shell/shell_module
2. cmake --build ble_shell
3. now flash to two device
4. First device command 
```
bt init
bt adv-create conn-nscan ext-adv coded name
bt adv-start
```
5. Second device command 
```
bt init
bt scan on coded no-1m
bt scan off
bt connect <address> random coded
```

[bt_shell_defconfig.zip](https://github.com/zephyrproject-rtos/zephyr/files/6021340/bt_shell_defconfig.zip)

**Expected behavior**
Connect established

**Impact**
Coded phy connection not work. And get assert
`uart:~$ ASSERTION FAIL [event.curr.abort_cb] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:321`

**Logs and console output**
Logs from periph
```
*** Booting Zephyr OS build zephyr-v2.4.0-4440-gd1a0568e1123  ***
uart:~$ bt init
B�uetooth0:00:27.521,331] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.521,881] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.521,972] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,064] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,155] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,247] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,308] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,399] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,491] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,583] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,674] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,766] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,857] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,918] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.522,949] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:27.522,949] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:27.522,979] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.5 Build 99
[00:00:27.523,040] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,132] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,223] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,315] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,406] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,529] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:27.523,681] <inf> bt_hci_core: Identity: D8:16:A0:49:9F:71 (random)
[00:00:27.523,712] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:27.523,712] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
 initialized
uart:~$ bt adv-create conn-nscan ext-adv coded name
Created adv id: 0, adv: 0x20002590
uart:~$ [00:00:35.170,898] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:35.170,989] <dbg> bt_conn.bt_conn_prepare_events: 
bt adv-start
Advertiser[0] 0x20002590 set started
uart:~$ [00:00:41.267,852] <dbg> bt_conn.bt_conn_set_state: disconnected -> connect-adv
[00:00:41.267,883] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
[... 68 lines cut ...]
[00:02:38.600,921] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
[00:02:38.600,921] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
[00:02:38.600,952] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
[00:02:38.600,952] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
[00:02:38.600,952] <err> os: fpscr:  0x00034c95
[00:02:38.600,952] <err> os: Faulting instruction address (r15/pc): 0x0003301c
[00:02:38.600,982] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:02:38.600,982] <err> os: Fault during interrupt handling[00:00:37.600,982] <err> os: Current thread: 0x20001b40 (idle 00)
[00:02:38.734,466] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit d1a0568e1123

**Additional context**
In bt_shell_defconfig add 
```
CONFIG_BT_CTLR_GPIO_PA=y
CONFIG_BT_CTLR_GPIO_PA_PIN=17
CONFIG_BT_CTLR_GPIO_LNA=y
CONFIG_BT_CTLR_GPIO_LNA_PIN=19
```
for our board I don't think is options relevant to bug, but I have included a complete config example

carlescufi, 2021-02-22:
@a1ien could you please run `addr2line` on `0x0003301c` and give us the source file and line number?

a1ien, 2021-02-22:
@carlescufi it's assertion line zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:321

cvinayak, 2021-02-22:
@a1ien could you try with CONFIG_BT_CTLR_PRIVACY=n ?

cvinayak, 2021-02-22:
Hey, there is no implementation as central to connect to extended advertising.

@andrzej-kaczmarek could you please confirm?

a1ien, 2021-02-22:
But What then does this PR https://github.com/zephyrproject-rtos/zephyr/pull/28754?

andrzej-kaczmarek, 2021-02-22:
@cvinayak correct, it's in progress
@a1ien it only adds support for connectable extended advertising in peripheral role

a1ien, 2021-02-22:
ohh... it is sad

carlescufi, 2021-02-23:
Closing this since it is not implemented yet. @a1ien please check back in time for the 2.6 release to see if it's implemented.

closed 2021-02-23 by carlescufi