zephyrproject-rtos/zephyr #32537: Fatal error syscall_list.h

expploitt, 2021-02-22:
Hello, I'm trying to compile zenoh-pico as a external lib (such as the sample project) but when I compile I get the error:

```
~/zephyrproject/zephyr/include/syscall.h:11:10: fatal error: syscall_list.h: No such file or directory
   11 | #include <syscall_list.h
```

I don't know how to solutions this, I'm reviewing my code but It seems that is some configuration that.

Thanks!

de-nordic, 2021-02-22:
The file that is missing here is generated at compile time to $BUILD_DIR/zephyr/include/generated/syscall_list.h.
I am not sure what you are trying to do with the "external lib": do you want to link this library in a similar way to for example tinycbor or mcumgr, which are built as Zephyr modules (modules/lib/) or fatfs library (modules/fs/fatfs)?

expploitt, 2021-02-22:
I'm trying to do something like the sample named external_lib in the **zephyrproject/zephyr/samples/application_development/external_lib** example.

There is a folder with the lib code, and it's linked after compilation.

de-nordic, 2021-02-22:
Have you looked at the CMakeLists.txt within that sample?
I have just built that sample for nrf52840dk_nrf52840 and it has generated the missing header and compiled ok.

expploitt, 2021-02-22:
yeah, the example works fine. But I open the issue if someone knows why could be  the reason of the error. I have to review then the code.. I don't have others errors, only this.

de-nordic, 2021-02-22:
Look at the CMakeLists.txt this is basically the recipe that makes the sample build, it should contain all operations that are needed for includes to work properly.

expploitt, 2021-02-22:
My Makefile inside the zenoh-lib folder:

```
PREFIX ?= .
OBJ_DIR ?= $(PREFIX)/obj
OBJ_DIR2 ?= $(PREFIX)/obj/net
OBJ_DIR3 ?= $(PREFIX)/obj/net/unix

LIB_DIR ?= $(PREFIX)/lib

BUILD_FLAGS ?= -DZENOH_ZEPHYR=1 -DZENOH_DEBUG=2 #-DCONFIG_NETWORKING=y -DCONFIG_NET_SOCKETS=y -DCONFIG_PTHREAD_IPC=y -DCONFIG_NET_SOCKETS_POSIX_NAMES=y 
all:
	mkdir -p $(OBJ_DIR) $(LIB_DIR)
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/codec.c -o $(OBJ_DIR)/codec.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/collection.c -o $(OBJ_DIR)/collection.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/iobuf.c -o $(OBJ_DIR)/iobuf.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/utils.c -o $(OBJ_DIR)/utils.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/codec.c -o $(OBJ_DIR)/codec.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/internal.c -o $(OBJ_DIR)/internal.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/lease_task.c -o $(OBJ_DIR)/lease_task.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/msgcodec.c -o $(OBJ_DIR)/msgcodec.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/property.c -o $(OBJ_DIR)/property.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/rname.c -o $(OBJ_DIR)/rname.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/session.c -o $(OBJ_DIR)/session.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/unix/network.c -o $(OBJ_DIR)/network.o
	$(CC) -c $(CFLAGS) ${BUILD_FLAGS} -MD -Iinclude src/net/unix/system.c -o $(OBJ_DIR)/system.o
	#$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/codec.o $(OBJ_DIR)/collection.o $(OBJ_DIR)/iobuf.o $(OBJ_DIR)/utils.o $(OBJ_DIR)/codec.o $(OBJ_DIR)/internal.o $(OBJ_DIR)/lease_task.o $(OBJ_DIR)/msgcodec.o $(OBJ_DIR)/property.o $(OBJ_DIR)/rname.o $(OBJ_DIR)/session.o $(OBJ_DIR)/network.o $(OBJ_DIR)/system.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/codec.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/collection.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/iobuf.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/utils.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/codec.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/internal.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/lease_task.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/msgcodec.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/property.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/rname.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/session.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/network.o
	$(AR) -rcs $(LIB_DIR)/libzenohpicolib.a $(OBJ_DIR)/system.o

clean:
	rm -rf $(OBJ_DIR) $(LIB_DIR)

```

And this is my CMakeLists.txt:

```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.13.1)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(external_lib)

target_sources(app PRIVATE src/main.c)

# The external static library that we are linking with does not know
# how to build for this platform so we export all the flags used in
# this zephyr build to the external build system.
#
# Other external build systems may be self-contained enough that they
# do not need any build information from zephyr. Or they may be
# incompatible with certain zephyr options and need them to be
# filtered out.
zephyr_get_include_directories_for_lang_as_string(       C includes)
zephyr_get_system_include_directories_for_lang_as_string(C system_includes)
zephyr_get_compile_definitions_for_lang_as_string(       C definitions)
zephyr_get_compile_options_for_lang_as_string(           C options)

set(external_project_cflags
  "${includes} ${definitions} ${options} ${system_includes}"
  )

include(ExternalProject)

# Add an external project to be able download and build the third
# party library. In this case downloading is not necessary as it has
# been committed to the repository.
set(mylib_src_dir   ${CMAKE_CURRENT_SOURCE_DIR}/zenoh-pico)
set(mylib_build_dir ${CMAKE_CURRENT_BINARY_DIR}/zenoh-pico)

set(MYLIB_LIB_DIR     ${mylib_build_dir}/lib)
set(MYLIB_INCLUDE_DIR ${mylib_src_dir}/include)

if(CMAKE_GENERATOR STREQUAL "Unix Makefiles")
# https://www.gnu.org/software/make/manual/html_node/MAKE-Variable.html
set(submake "$(MAKE)")
else() # Obviously no MAKEFLAGS. Let's hope a "make" can be found somewhere.
set(submake "make")
endif()

ExternalProject_Add(
  mylib_project                 # Name for custom target
  PREFIX     ${mylib_build_dir} # Root dir for entire project
  SOURCE_DIR ${mylib_src_dir}
  BINARY_DIR ${mylib_src_dir} # This particular build system is invoked from the root
  CONFIGURE_COMMAND ""    # Skip configuring the project, e.g. with autoconf
  BUILD_COMMAND
  ${submake}
  PREFIX=${mylib_build_dir}
  CC=${CMAKE_C_COMPILER}
  AR=${CMAKE_AR}
  CFLAGS=${external_project_cflags}
  INSTALL_COMMAND ""      # This particular build system has no install command
  BUILD_BYPRODUCTS ${MYLIB_LIB_DIR}/libzenohpicolib.a
  )

# Create a wrapper CMake library that our app can link with
add_library(mylib_lib STATIC IMPORTED GLOBAL)

add_dependencies(
  mylib_lib
  mylib_project
  )
set_target_properties(mylib_lib PROPERTIES IMPORTED_LOCATION             ${MYLIB_LIB_DIR}/libzenohpicolib.a)
set_target_properties(mylib_lib PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${MYLIB_INCLUDE_DIR})

add_dependencies(
  mylib_lib
)

target_link_libraries(app PUBLIC mylib_lib)

``` 

Also I'm using this prj configuration:

```
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_PTHREAD_IPC=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
```

de-nordic, 2021-02-22:
Can you dump here build invocation with console dump?

expploitt, 2021-02-22:
```
╰─➤  west build -p auto -b nrf52840dk_nrf52840 samples/application_development/zenoh_lib
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/moises/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/moises/zephyrproject/zephyr/samples/application_development/zenoh_lib
-- Zephyr version: 2.5.99 (/home/moises/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/moises/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/home/moises/zephyr-sdk-0.12.3)
-- Found dtc: /home/moises/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/moises/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/moises/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/moises/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/moises/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/moises/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/moises/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/moises/zephyrproject/zephyr/samples/application_development/zenoh_lib/prj.conf'
Configuration saved to '/home/moises/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/moises/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/moises/zephyrproject/zephyr/build
-- west build: building application
[1/186] Preparing syscall dependency handling

[8/186] Performing build step for 'mylib_project'
FAILED: zenoh-pico/src/mylib_project-stamp/mylib_project-build zenoh-pico/lib/libzenohpicolib.a 
cd /home/moises/zephyrproject/zephyr/samples/application_development/zenoh_lib/zenoh-pico && make PREFIX=/home/moises/zephyrproject/zephyr/build/zenoh-pico CC=/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc AR=/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-ar "CFLAGS=-I/home/moises/zephyrproject/zephyr/include -I/home/moises/zephyrproject/zephyr/build/zephyr/include/generated -I/home/moises/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/moises/zephyrproject/zephyr/lib/libc/minimal/include -I/home/moises/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/moises/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/moises/zephyrproject/modules/hal/nordic/nrfx -I/home/moises/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/moises/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/moises/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -I/home/moises/zephyrproject/modules/debug/segger/rtt -I/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include -I/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include-fixed -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=zephyr-v2.5.0-358-gc0ee8c4a4326 -D__PROGRAM_START -DNRF52840_XXAA -Os -imacros /home/moises/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/moises/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/moises/zephyrproject/zephyr/samples/application_development/zenoh_lib=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/moises/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/moises/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -isystem/home/moises/zephyrproject/zephyr/lib/libc/minimal/include -isystem/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include -isystem/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/include-fixed" && /usr/bin/cmake -E touch /home/moises/zephyrproject/zephyr/build/zenoh-pico/src/mylib_project-stamp/mylib_project-build
mkdir -p /home/moises/zephyrproject/zephyr/build/zenoh-pico/obj /home/moises/zephyrproject/zephyr/build/zenoh-pico/lib
/home/moises/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -c -I/home/moises/zephyrproject/zephyr/include -I/home/moises/zephyrproject/zephyr/build/zephyr/include/generated -I/home/moises/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/moises/zephyrproject/zephyr/lib/libc/minimal/include -I/home/moises/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/moises/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/moises/zephyrproje [... cut ...]
                 from src/codec.c:17:
/home/moises/zephyrproject/zephyr/include/syscall.h:11:10: fatal error: syscall_list.h: No such file or directory
   11 | #include <syscall_list.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:117: all] Error 1
[9/186] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/moises/zephyrproject/zephyr/build

```

expploitt, 2021-02-22:
I don't know why, but if I execute the command two times, I get the error above. However, if I execute one more time then the software compiles. I'm going to test if runs well in the board.

de-nordic, 2021-02-22:
> 
> 
> I don't know why, but if I execute the command two times, I get the error above. However, if I execute one more time then the software compiles. I'm going to test if runs well in the board.

Can you check each time you run whether the zephyr/include/generated/syscall_list.h has been created after each build attempt?

de-nordic, 2021-02-22:
I have modified sample to include sys_call.h and indeed compilation fails. It seems that the sys_call.h is generated later than the library is build, so if library depends on syscall_list.h it will not build. Bringing back Bug label.
@tejlmand any thoughts?

tejlmand, 2021-02-23:
> I have modified sample to include sys_call.h and indeed compilation fails. It seems that the sys_call.h is generated later than the library is build, so if library depends on syscall_list.h it will not build. Bringing back Bug label.
> @tejlmand any thoughts?

Well, if you rely on `syscalls` then you must tell the build system that you rely on it, for example by depending on Zephyr interface.

By the looks of the error message, it seems it's the external project having this dependency, so you probably want to add:
```
ExternalProject_Add(
  mylib_project                 # Name for custom target
  ...
  DEPENDS zephyr_interface
  )
```

de-nordic, 2021-02-23:
Thanks @tejlmand , that works. @expploitt can you check whether the solution solves the problem on your side and close the issue?

expploitt, 2021-02-23:
Thanks @tejlmand !! It solves the problem. I close the issue.

closed 2021-02-23 by expploitt