zephyrproject-rtos/zephyr #32563: SECTION_DATA_PROLOGUE BUG

hongshui3000, 2021-02-23:
**Describe the bug**
file : linker-tool-gcc.h
```
#ifdef CONFIG_XIP
#define SECTION_DATA_PROLOGUE(name, options, align) \
	name options : ALIGN_WITH_INPUT align
#else
``

If CONFIG_XTENSA & CONFIG_XIP is configured, ALIGN_WITH_INPUT is not supported in XTENSA's xt-ld

Please check again

carlescufi, 2021-02-23:
@hongshui3000 could you perhaps send a Pull Request for this?

andyross, 2021-02-23:
What are you trying to do, exactly?  We don't support XIP on any existing xtensa platforms, and XIP as implemented definitely requires binutils-style alignment control.  If you need it for your app maybe there's a hybrid solution where you could build with the Xtensa toolchain and link with binutils?

hongshui3000, 2021-02-24:
> What are you trying to do, exactly? We don't support XIP on any existing xtensa platforms, and XIP as implemented definitely requires binutils-style alignment control. If you need it for your app maybe there's a hybrid solution where you could build with the Xtensa toolchain and link with binutils?

XIP mode is used in my application. I wrote the ld script myself and found this problem when I wanted to use the SECTION_DATA_PROLOGUE macro definition. So I gave up the SECTION_DATA_PROLOGUE macro when writing the ld script to avoid this problem. As you said, I did use the hybrid solution

hongshui3000, 2021-02-24:
> @hongshui3000 could you perhaps send a Pull Request for this?

My solution may not have much reference value for zephyr. I just rewrote my own ld script. So I am not going to submit these

hongshui3000, 2021-02-24:
By the way, there is no big problem with xip on Xtensa, except for linking scripts.

andyross, 2021-02-26:
So... this isn't really a bug.  I'm going to change it to an enhancement: "Add Xtensa xt-ld linker support to CONFIG_XIP" and assign it to @hongshui3000 in the hope that something gets contributed back. :)

But broadly, I agree.  Our whole linkage framework is sort of a mess and could stand with rework.  This is one glitch of many.

hongshui3000, 2021-03-04:
file：zephyr\soc\xtensa\intel_adsp\cavs_v15\linker.ld  
```
#define RAMABLE_REGION ram :ram_phdr
#define ROMABLE_REGION ram :ram_phdr ------------------------------->this
#include <linker/common-rom.ld>------------------------------------------>this
```
I feel bad about this way of changing the name to specify the characteristics
This is the beginning of code confusion

dcpleung, 2024-02-08:
No XIP on Xtensa support. So this is a moot point... closing for now. If needed, please re-open.

closed 2024-02-08 by dcpleung