zephyrproject-rtos/zephyr #32572: tests/kernel/timer/timer_api/test_timeout_abs fails on stm32 boards

FRASTM, 2021-02-23:
**Describe the bug**
after The  PR :  kernel/timeout: Fix off-by-one in absolute timeouts #32505
the test fails on many stm32 boards
```
START - test_timeout_abs                                                                
                                                                                        
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:725:e
                                                                                        
 FAIL - test_timeout_abs  
```                                                              

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l152re tests/kernel/timer/timer_api
2. west flash
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

nashif, 2021-02-23:
The failure in details:

```

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:725: test_timeout_abs: (cap_ticks + rem_ticks + 1 == exp_ticks) || (rem_ticks + cap2_ticks + 1 == exp_ticks) || (INEXACT_MS_CONVERT && (cap_ticks + rem_ticks == exp_ticks)) || (INEXACT_MS_CONVERT && (rem_ticks + cap2_ticks == exp_ticks)) is false

```

nashif, 2021-02-23:
This is failing on other boards:

```
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  1/16 sam_e70_xplained          tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 7.809s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/sam_e70_xplained/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  2/16 nrf52840dk_nrf52840       tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 11.026s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  3/16 nucleo_f746zg             tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 12.469s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f746zg/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  4/16 nucleo_f429zi             tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 14.157s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f429zi/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  5/16 reel_board                tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 15.312s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  6/16 nucleo_f207zg             tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Failed (device 16.009s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f207zg/tests/kernel/timer/timer_api/kernel.timer.tickless/handler.log
INFO    -  7/16 disco_l475_iot1           tests/kernel/timer/timer_api/kernel.timer.tickless PASSED (device 15.541s)
INFO    -  8/16 nrf52840dk_nrf52840       tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 24.401s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    -  9/16 sam_e70_xplained          tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 7.576s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/sam_e70_xplained/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    - 10/16 reel_board                tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 8.604s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    - 11/16 nucleo_f746zg             tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 11.689s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f746zg/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    - 12/16 nucleo_f429zi             tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 10.495s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f429zi/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    - 13/16 nucleo_f207zg             tests/kernel/timer/timer_api/kernel.timer          FAILED Failed (device 11.761s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/nucleo_f207zg/tests/kernel/timer/timer_api/kernel.timer/handler.log
INFO    - 14/16 mimxrt1050_evk            tests/kernel/timer/timer_api/kernel.timer.tickless FAILED Timeout (device 63.781s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/mimxrt1050_evk/tests/kernel/timer/timer_api/kernel.timer.tickless/build.log
INFO    - 15/16 disco_l475_iot1           tests/kernel/timer/timer_api/kernel.timer          PASSED (device 14.803s)
 INFO    - 16/16 mimxrt1050_evk            tests/kernel/timer/timer_api/kernel.timer          FAILED Timeout (device 63.587s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/mimxrt1050_evk/tests/kernel/timer/timer_api/kernel.timer/build.log

INFO    - 2 of 16 test configurations passed (12.50%), 14 failed, 0 skipped with 0 warnings in 169.30 seconds
INFO    - In total 224 test cases were executed, 0 skipped on 8 out of total 327 platforms (2.45%)
INFO    - 16 test configurations executed on platforms, 0 test configurations were only built.
```

hakehuang, 2021-02-24:
also fails on nxp platforms RT1xxx and RT600

andyross, 2021-02-25:
Potential fix up at #32683 .  There was indeed a bug in the test exposed by the recent absolute timeout fix on high tick rate systems (i.e. hardware, not CI).  I caught it happening on the adsp I was working on, so I'm fairly confident this is the issue.  If those with affected systems could test and report results, I'd be appreciative.

Though I will note that the fact that all those systems are systick-based makes me a tiny bit worried there may be a driver bug in play too.

closed 2021-02-26 by nashif

MaureenHelm, 2021-02-26:
> Potential fix up at #32683 . There was indeed a bug in the test exposed by the recent absolute timeout fix on high tick rate systems (i.e. hardware, not CI). I caught it happening on the adsp I was [...]

@hakehuang can you verify the fix on RT1xxx and RT600?

hakehuang, 2021-03-01:
> @hakehuang can you verify the fix on RT1xxx and RT600?
this fix works. @MaureenHelm 
`
*** Booting Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a  ***
Running test suite timer_api
===================================================================
START - test_time_conversions
 PASS - test_time_conversions
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period
===================================================================
START - test_timer_restart
 PASS - test_timer_restart
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync
===================================================================
START - test_timer_k_define
 PASS - test_timer_k_define
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data
===================================================================
START - test_timer_remaining
 PASS - test_timer_remaining
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs
===================================================================
START - test_sleep_abs
 PASS - test_sleep_abs
===================================================================
Test suite timer_api succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
`

MaureenHelm, 2021-03-01:
> this fix works for rt1xxx and rt6xx.

thanks @hakehuang !

PR #32683 (2021-02-26): tests/timer_api: Fix absolute timeout logic
The test_timeout_abs case had baked in similar mistakes to the
off-by-one in the absolute timer implementation.  FOR THE RECORD:

If you have an absolute timeout expiration set for a tick value "N",
and the current time returned by k_uptime_ticks() is "T", then the
time returned (at the same moment) by any of the *_remaining_ticks()
APIs must ALWAYS AND FOREVER BE EXACTLY "N - T" (also: "N - T > 0"
always, until the moment the kernel ISR hands off control to the first
timeout handler expiring at that tick).

The tick math is exact.  No slop is needed [... cut]
commit 63972d6d (2021-02-26): tests/timer_api: Fix absolute timeout logic

The test_timeout_abs case had baked in similar mistakes to the
off-by-one in the absolute timer implementation.  FOR THE RECORD:

If you have an absolute timeout expiration set for a tick value "N",
and the current time returned by k_uptime_ticks() is "T", then the
time returned (at the same moment) by any of the *_remaining_ticks()
APIs must ALWAYS AND FOREVER BE EXACTLY "N - T" (also: "N - T > 0"
always, until the moment the kernel ISR hands off control to the first
timeout handler expiring at that tick).

The tick math is exact.  No slop is neede [... cut]