zephyrproject-rtos/zephyr #32577: target_compile_features in CMakeLists.txt triggers an error

Thurnheer, 2021-02-23:
I am developing a framework. In the framework's CMakeLists.txt I've the following statement:
target_compile_features(myFramework PRIVATE cxx_std_17)

The application and the framework are developed using C++.

When I now run west build, I got the following error during cmake configuration:
target_compile_features no known features for C compiler "GNU" version 9.2.0

Using the toolchains from the zephyr SDK fails while trying to compile a simple test program.

I'm using the zephyr SDK for compilation, develop on Ubuntu LTS 20.04, use west version 0.8.0 and zephyr master.

Is there a way out of this?

tejlmand, 2021-02-24:
this happens because Zephyr has it own toolchain handling and compiler verification scheme, which means the normal CMake check is not performed and therefore the `target_compile_features` won't work.

If you want to build using a specific language standard, you can do so by specifying: `CONFIG_STD_CPP17=y` either in `prj.conf` or on command line `-DCONFIG_STD_CPP17=y`.

See more here for possible settings:
https://docs.zephyrproject.org/latest/reference/kconfig/choice_359.html#choice-359

Thurnheer, 2021-02-25:
The problem is that the framework is targeting windows, linux and zephyr, so at least for windows and linux we need the target_compile_features. It would be convenient to just set it in the CmakeLists file so every project using the framework has automatically the correct settings, or is it possible to somehow set it for zepyr in the cmake file?

tejlmand, 2021-02-25:
> The problem is that the framework is targeting windows, linux and zephyr, so at least for windows and linux we need the target_compile_features.

I see, and thanks for this additional info.
Although Zephyr itself has it's own toolchain handling, for sure this should not prevent users from using common CMake features in shared projects.
I will look into how this may be handled.

Note: there is a wish in community to also be able to use the common CMake toolchain functionality, but that is a bigger task to introduce this, as several compile options are handled through Kconfig and the toolchain infrastructure, so it must be ensure the two principles doesn't conflict.

carlescufi, 2021-05-19:
Changing to Enhancement to support CMake's `CMAKE_TOOLCHAIN_FILE`

yashi, 2021-09-25:
If the whole point is to workaround the error caused by `target_compile_features()`,  how about setting `CMAKE_C_COMPILE_FEATURES` based on `CSTD`?  Or, for c++ variant.

@Thurnheer, if it's just for you, you might want to `set(CMAKE_CXX_COMPILE_FEATURES cxx_std_17)` before going down to your framework.

tejlmand, 2021-09-27:
The fix to this seems to be the same as I proposed here:
https://github.com/zephyrproject-rtos/zephyr/issues/36558#issuecomment-914038682

but never got a reply on that message.
I would like to be sure it fixes the use-cases described and that they won't hit another obstacle afterwards.
(I prefer complete fixes, not partial fixes before sending a PR :wink: )

yashi, 2021-09-27:
Ah, I wasn't aware of the issue.  Thanks!
(I think we can close this as a duplicate.  The other has medium priority :grin:).

FWIW, I'm using the same workaround when I integrate a CMake based project as a Zephyr module.  So +1 for this.

tejlmand, 2021-09-30:
Fixed here: #39014

tejlmand, 2021-11-03:
Fixed here: #39014 and #39060

closed 2021-11-03 by tejlmand