zephyrproject-rtos/zephyr #32590: RFC: namespace to address Rule 21.2 violations under coding guidelines.

yerabolu, 2021-02-23:
## Introduction

**Coding guidelines Rule 21.2: A reserved identifier or macro name shall not be declared**.  
Almost all the violations for this rule are because of **identifier or macro name beginning with an underscore**. 

### Problem description
After triaging majority of the violations, most of them are related to macro name beginning with  underscore. Removing underscores will not work in all the scenarios as there could be clash with other identifiers/macro names and break the functionality. This could be a challenge to maintain and also address the similar issues if identified in the new code changes that will be coming in future. 

### Proposed change
Would like to propose a namespace to address the violations for the identifiers/macro name starting with underscores and follow proposed namespace to fix similar issues for changes coming in future while maintaining the readability.

- subsystem name
- filename

### Concerns and Unresolved Questions
Changes will trigger lot of changes in different files which are used by different boards. Testing could be challenge.
For ex: include/init.h - this file is concatenating entry names for device drivers.

andyross, 2021-02-24:
I say this every time the subject comes up, but it's not worth starting a fight over:

The **spirit** behind the rule in the C standard (and the variants elsewhere) that reserves identifiers beginning with an underscore is that the system has already used that namespace to define internal symbols it needs.  The rule allows the system to have a clean internal space that it knows legal application code can't collide with.

But... **We are the system.**[1]

The rule about reserved identifiers?  It's there for **our benefit**.  In a reasonable interpretation of this rule, we should be exploiting this feature and not trying to comply with it.

[1] Not us alone, obviously.  Toolchain and platform headers also tend to use reserved identifiers.  It's not like we'll never collide or that we don't have to be reasonable about things.  But IMHO this rule is silly when applied to an OS.

jharris-intel, 2021-02-24:
It's kind of like the amusing weaseling of:

1. There's an implementation defined minimum number of prefix characters to uniquely identify an identifier in the C99 spec, with no explicit range given. ("The number of significant characters in an identifier is implementation-defined." (6.4.2.2.5)
2. The C99 spec states "Any identifiers that differ in a significant character are different identifiers. If two identifiers differ only in nonsignificant characters, the behavior is undefined."
3. The C99 spec states "The implementation shall be able to translate and execute at least one program that contains at least one instance of every one of the following limits [...] 63 significant initial characters in an internal identifier or a macro name" (5.2.4.1.1)
4. There exists a C99 program that returns 42 and has no other visible side effects, that fits the requirements of 3.
5. The C99 spec requires both `__FILE__` (6.10.8.1) and __func__` (6.4.2.2) to be defined.

Therefore it is legal to compile all C99 programs to "return 42". (Why? Choose an impdef minimum number of significant characters <= 2, with the exception of the program identified in 4. Then every input program is either a) undefined behavior (because of 2 and 5), or b) our program, in which case it is legal to compile it to "return 42". All valid programs for our compiler can be compiled to "return 42" as a result.)

Does this mean that you should submit a patch to GCC that does so? Probably not.

*****

In general, the C99 standard (and C standards in general) have a very different spirit than the C++ standards. Notably, C code can be non-portable, as the C99 rationale document mentions.

*****

All told I am in agreement with you here @andyross.

yerabolu, 2021-03-08:
Yes, i am in agreement with above discussion.  @nashif : Any other thoughts on this ? If we are in agreement will close the RFC and include it in the deviations report.

ceolin, 2021-03-09:
> Yes, i am in agreement with above discussion. @nashif : Any other thoughts on this ? If we are in agreement will close the RFC and include it in the deviations report.

It not only about deviation IMHO. We should document it in Zephyr code guideline, the concerns were more about underscore usage and standard C library, but the guideline itself cover other things that @jharris-intel mentioned. We still should stick to identifier translation limits, not contains an invalid multibyte character or does not begin and end in the initial shift state, the initial character of an identifier is a universal character name designating a digit, ...

RobertoBagnara, 2021-03-22:
The point of Rule 21.2 is that invading the namespace of the implementation may result in undefined behavior.  The implementation is constituted by all the translation machinery (compiler, assembler, linker), the Standard Library and, for hosted implementation, the part of the operating system that is required to implement the Standard Library. Note that there is more to Rule 21.2 than the presence of leading underscores. For example, Zephyr uses the name "exp" for an integer variable.  That name is reserved by the implementation: it is the name of a function (<math.h>) and it is the name of a macro (<tgmath.h>).  You may say that you are not including those standard header files, but this is not enough: they may, e.g., be provided as intrinsics, but the bottom line is that you cannot make any assumption about what the implementation of the language does to anything you call "exp".  The least thing that may happen is that, in Zephyr, due to a trivial typo, what now is

    int exp = ...;
    ...
    if (exp != 0) {
       ...
    }

becomes

    int exq = ...;

    if (exp != 0) {
      // Now always executed.
      ...
    }

because the address of function exp() is never zero.

Said that, sometimes the rule has to be deviated, but this cannot be done blindly.  For example, we recently worked on a project using CMSIS (as "adopted code", a MISRA technical term), which does invade the implementation namespace. The deviation involved (mostly) automatic checks to ensure that the reserved identifiers and macro names used in CMSIS were in fact not used by the implementation (that specific implementation, that specific version of the implementation).

There is enormous latitude in the choice of naming conventions that do not invade what is reserved to the implementation so, whenever this is possible, complying to Rule 21.2 is the right course of action and the cheapest one.