zephyrproject-rtos/zephyr #32599: bbc_microbit build failure is blocking CI

mbolivar-nordic, 2021-02-24:
The `bluetooth.mesh.main.microbit` case in tests/bluetooth/mesh is failing and causing CI failures in other PRs.

I'm getting this error in CI in an unrelated PR (#31965):

```
/workdir/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:106:7: error: 'BT_CTLR_CONN_MAX' undeclared here (not in a function)
  106 |       BT_CTLR_CONN_MAX + \
      |       ^~~~~~~~~~~~~~~~
```

Details in https://buildkite.com/zephyr/zephyr/builds/21988#868a8486-8a62-4f2a-9274-f55581716664.

This reproduces for me on master with:

```
west build -p -b bbc_microbit -s tests/bluetooth/mesh -- -DCONF_FILE=$PWD/tests/bluetooth/mesh/microbit.conf
```

mbolivar-nordic, 2021-02-24:
@cvinayak bisection from v2.5.0 shows bee6aa325b4e3e11616d98af463dc581136e20dd as the first commit with this error, but before that I get RAM overflows.

mbolivar-nordic, 2021-02-24:
I got a hint from @cvinayak offline on how to fix and posted a PR. This restores the old RAM overflow behavior.

jhedberg, 2021-02-24:
There’s a fix for the RAM overflow in #32562

PR #32600 (2021-02-24): Bluetooth: controller: fix typo
A macro is misspelled.

Fixes: #32599
commit 31c1dc58 (2021-02-24): Bluetooth: controller: fix typo

A macro is misspelled.

Fixes: #32599
closed 2021-02-24 by nashif