zephyrproject-rtos/zephyr #32608: Revert practice of removing devicetree labels

pabigot, 2021-02-24:
In 2.5.0 we gained the ability to get device references at build time using `DEVICE_DT_GET()`, which eliminates the need to look up devices by labels, and so the `label` property in devicetree can be marked `required: false`.

In initial changes to this model such as #32172 these edits were accompanied by removal of the `label` attribute from existing devicetree files.

As pointed out in https://github.com/zephyrproject-rtos/zephyr/pull/32558#discussion_r581390302 and https://github.com/zephyrproject-rtos/zephyr/pull/32127#pullrequestreview-594066129 it remains useful to have a unique name associated with `dev->name` for both diagnostics and runtime device lookup (#32587).

This issue stands as a proposal to have policy to retain existing labels, and a task to restore the ones that have been removed since 2.5.0 was released.

galak, 2021-02-24:
Its been a long term goal to remove the need for `label` in the devicetree.  I disagree with having this be a requirement.  For diagnostic, I've recommend using node name if there is no `label` property.

pabigot, 2021-02-24:
That's not how I understand that goal.  For me it's been a long term goal to remove the use of `label` as the standard (sole) way of retrieving a device reference at runtime, but attempts to switch to another identifier like `zephyr,devid` never got support.

Use of a label for its documented purpose of "a human readable string describing a device" is still useful, in particular as an alternative to things like `[20003b8]` as a unique device identifier when there is no name.  Existing labels do satisfy that role.

galak, 2021-02-24:
> That's not how I understand that goal. For me it's been a long term goal to remove the use of `label` as the standard (sole) way of retrieving a device reference at runtime, but attempts to switch [...]

From a devicetree point of view, I think utilizing node name and/or dt path should be the means we utilize for "name".

pabigot, 2021-02-24:
> From a devicetree point of view, I think utilizing node name and/or dt path should be the means we utilize for "name".

node name is not globally unique, and dt path is both incomprehensible and often exceeds `Z_DEVICE_MAX_NAME_LEN`.

mbolivar-nordic, 2021-02-24:
There's a difference between removing the *need* for labels and removing the labels themselves. The labels are useful and I would oppose attempts to remove them from existing nodes that users interact with. I agree with @pabigot's analysis rejecting node names and paths as a suitable alternative.

galak, 2021-02-24:
> There's a difference between removing the _need_ for labels and removing the labels themselves. The labels are useful and I would oppose attempts to remove them from existing nodes that users interact with. I agree with @pabigot's analysis rejecting node names and paths as a suitable alternative.

What is meant by `users interact` with?  The only bindings so far to remove the required `label` are pinmux bindings.  Is the feeling that those should be restored?  Why does `label` need to be required?  If for a given device the node name is descriptive enough?

mbolivar-nordic, 2021-02-24:
> What is meant by "users interact" with?

I mean things like peripheral drivers, sensor drivers, etc. Basically the stuff we've got samples/ for and we print the names of.

> The only bindings so far to remove the required `label` are pinmux bindings. Is the feeling that those should be restored?

No, I'm not saying that the ones that are removed should be restored. I did say (on Slack) that I'd be -1 to remove any labels from nRF DTSIs, though.

> Why does `label` need to be required? If for a given device the node name is descriptive enough?

It is descriptive but not unique, which breaks device_get_binding() and device_from_name().

thedjnK, 2021-02-25:
I think removing labels is one very backwards step. Picking a board I have here up and looking at the DTS https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/bt510/bt510.dts gives me a very clear indication of the usefulness of labels, I can see led_1a is "Green LED" and led_1b is "Red LED" - how would I know which is which just looking at device IDs if there was no label and how would that user friendly? I'm with @pabigot on this, the label shouldn't need to be required but that doesn't mean they should be removed or can't be present

pabigot, 2021-02-25:
@thedjnK actually LEDs are about the only thing that were using labels as intended. Individual LEDs are not (usually) individual devices, so they aren't affected.

The question here is more about things like pinmux drivers: should they have labels so they can get names like `PINMUX1`?  If no pinmux drivers or users that need to identify the device by name, they probably don't, but that starts a slippery slope: I2C and GPIO drivers do need to have meaningful and convenient names so they can be accessed in the shell.

closed 2022-09-26 by gmarull